realsenseai/librealsense #6421: How to recover the GLSL setting of realsense-viewer

miku39s3, 2020-05-19:
| | |
|---|---|
|**librealsense**|2.33.4|
|**OS**|Raspberry Debian|

When enable 'use GLSL for Rending' and 'use GLSL for Processing' in the setting panel, the realsense-viewer can't be launched anymore. The error message is that 'GLSL 1.30 is not supported'. Due to the GUI is not availble anymore, I can't recover the setting, but I don't know where the configuration file is. I uninstalled and install again, the problem still exists.

MartyG-RealSense, 2020-05-19:
I hope that the link below will be useful in helping you to locate the Viewer's configuration file.

https://github.com/IntelRealSense/librealsense/issues/3821#issuecomment-485668383

dorodnic, 2020-05-19:
If you cannot access the UI,
```
rm ~/.realsense-config.json
```
This will reset everything to default

closed 2020-06-10 by MartyG-RealSense