realsenseai/librealsense #6424: D435i: Minimal IMU callback example doesn't deliver selected frame rate.

rafaelspring, 2020-05-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.11.06.250, 5.12.03.00 | 
| Operating System & Version |  Win10  | 
| SDK Version                            |  2.34.0 | 
| Language                            |  C++ | 

This is a follow-up to #6404 . There seem to be a lot of things going wrong all at once and I was able to isolate at least one of the issues.

Below is a minimal code example. This is roughly based on the rs-callback example.

The expected behavior is to receive 200 callbacks per second for the Gyro and 63 callbacks per second for the Accelerometer since these are the default stream profiles.

However on my D435i I only receive ~30 callbacks per second for both Gyro and Accelerometer.

I was able to reproduce this on firmwares 5.11.06.250 and 5.12.03.00. I haven't tried any other FW versions.

The issue seems to only affect the D435i. **Things work as expected on the L515!**

```cpp

// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <iostream>
#include <map>
#include <chrono>
#include <mutex>
#include <thread>
#include <string>

static std::vector<rs2::stream_profile> getIMUStreamProfiles(const rs2::context& ctx,
                                                             rs2::sensor* ssr = nullptr) {
  std::vector<rs2::stream_profile> rsIMUProfiles;

  auto allSensors = ctx.query_all_sensors();

  for (const rs2::sensor& sens : allSensors) {
    std::string sensorName = std::string(sens.get_info(RS2_CAMERA_INFO_NAME));

    if (sensorName.find("Motion Module") != std::string::npos) {
      // Let's just get the default profile(s) for now.
      auto profiles = sens.get_stream_profiles();
      for (const rs2::stream_profile& sp : profiles) {
        if (sp.is_default() && sp.stream_type() == RS2_STREAM_GYRO) { // <-- Same issue for ACCEL
          std::cout << "Stream Profile " << sp.stream_type() << ", format " << sp.format() << " at " << sp.fps() << " fps" << std::endl;
          rsIMUProfiles.push_back(sp);
        }
      }

      if (ssr) { *ssr = sens; }
    }
  }

  return rsIMUProfiles;
}

static rs2::device getFirstRSDevice() {
  rs2::context ctx;
  rs2::device firstDev;

  std::vector<rs2::device> devList(ctx.query_devices());
  for (const auto& dev : devList) {
    const std::string devName(dev.get_info(RS2_CAMERA_INFO_NAME));
    if (devName.find("ense D4") != std::string::npos ||
        devName.find("ense L5") != std::string::npos) {
      firstDev = dev;
      break;
    }
  }
  return firstDev;
}

int main(int argc, char * argv[]) try {
    //rs2::log_to_console(RS2_LOG_SEVERITY_ERROR);

    std::map<int, int> counters;
    std::mutex mutex;

    // Our IMU callback. Just increments a frame counter.
    auto callback = [&](const rs2::frame& frame) {
        // Uncomment to see that it's really the IMU calling back.
//      std::cout << "at " << int(frame.get_timestamp() * 1000 + 0.5) << " us: "
//                << frame.get_profile().stream_name() << " at " << frame.get_profile().fps() << " fps." << std::endl;

      std::lock_guard<std::mutex> lock(mutex);
      counters[frame.get_profile().unique_id()]++;
    };

    rs2::context ctx;
    rs2::sensor imuSsr;
    {
      // Start the IMU with the default Gyro or Accel profile, using the callback.
      auto imuProfiles = getIMUStreamProfiles(ctx, &imuSsr);
      imuSsr.open(imuProfiles);
      imuSsr.start(callback);
    }

    // Output the frame counter every second, then quit after 10 seconds.
    int i = 0;
    while (true)
    {
        std::this_thread::sleep_for(std::chrono::seconds(1));

        {
          std::lock_guard<std::mutex> lock(mutex);

          for (auto p : counters) {
              std::cout << "[" << p.first << "]: " << p.second << " frames received" << std::endl;
          }
        }

        if (i == 10) {
          imuSsr.stop();
          break;
        }

        ++i;
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

```

ev-mp, 2020-05-20:
@rafaelspring hello, I cannot identify any issue with the code and it should indeed generate callbacks at the prescribed rate. Is it possible that you're running in Debug and not Release  ? (cmake flag `-DCMAKE_BUILD_CONFIGURATION=release`) 

To further isolate the issue please run `rs-data-collect` tool with [IMU only configuration](https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect#config-file-format) for 10-15 sec and then post the resulted log for review.
Thanks.

rafaelspring, 2020-05-20:
@ev-mp 
I checked Release mode but same results.
Attached is the output of a release-build of `rs-data-collect` running for 15 secs. I also attached the config file I used.
[rsdc-config.txt](https://github.com/IntelRealSense/librealsense/files/4656808/rsdc-config.txt)
[rs-dc-log.txt](https://github.com/IntelRealSense/librealsense/files/4656810/rs-dc-log.txt)

rafaelspring, 2020-05-20:
@ev-mp 
And as I expected ( #6426 ) adding a depth stream makes the issue disappear (see attached config and log).
[rsdc-config.txt](https://github.com/IntelRealSense/librealsense/files/4656836/rsdc-config.txt)
[rs-dc-log.txt](https://github.com/IntelRealSense/librealsense/files/4656837/rs-dc-log.txt)

ev-mp, 2020-05-20:
@rafaelspring , I've checked the attached log and it indicates an actual issue:
 - The HW timestamps indicate that gyro data is transmitted in bursts of 5 frames (5*5=25msec) + ~100msec interval till the next burst.
- On the other hand the host timestamps report that the gyro readings are streamed with equal intervals 25msec  apart.
The actual rate in both cases is identical  ~33FPS. But if the HW timestamps are correct there is an actual issue with the IMU sensor HW or its FW on this D435i.
I don't recall encountering such exhibited behavior, so we'll need to further investigate it to reach some conclusions.

Can you please repeat that test as follows:
1. Streaming Accel only (and posting the log).
2. Streaming Gyro with the same configuration but on a different PC (Linux/Win)
3. In case you have an additional D435i please rerun the same test on the original PC.
Those steps will allow to isolate the issue.

rafaelspring, 2020-05-20:
> Streaming Accel only (and posting the log).

Attached.
[rsdc-config.txt](https://github.com/IntelRealSense/librealsense/files/4658153/rsdc-config.txt)
[rs-dc-log.txt](https://github.com/IntelRealSense/librealsense/files/4658155/rs-dc-log.txt)

Also interesting: The issue also goes away in my own code (everything works as expected) if I turn off global time: `motionModule.set_option(RS2_OPTION_GLOBAL_TIME_ENABLED, 0);`
The default seems to be global time turned on.

ev-mp, 2020-05-20:
@rafaelspring , I think that you might have just discovered the root cause
> The issue also goes away in my own code (everything works as expected) **if I turn off global time**:

We'll work on reproduction and if confirmed - it is an SDK that we'll need to fix.
In the meanwhile please use the W/A and disable the GlobalTime option before start streaming.

BTW, the additional Accel-only log is also consistent with the original inconsistency (hm....)

doronhi, 2021-05-19:
It has been a year. I could not reproduce it using the latest librealsense2 version 2.45 and firmware 05.12.13.50
Added [log2.xlsx](https://github.com/IntelRealSense/librealsense/files/6507394/log2.xlsx) - Produced by a 3 minutes run of rs-data-collect with IMU only configuration and converted from log2.csv
In log2.xlsx we can see that a Gyro frame arrives every 4.98-5.06 milliseconds and an Accel frame every 15.97-16.23 milliseconds. Nothing like the bursts described in the original issue.

RealSenseSupport, 2021-05-25:
@rafaelspring Did you get chance to try with the latest librealsense and Firmware version as @doronhi suggested? Looking forward to your update. Thanks!

rafaelspring, 2021-05-25:
@RealSenseSupport @doronhi We simply turned off global timing (see previous replies) and it's not been an issue since. I haven't tried to reproduce the issue but will trust @doronhi that it's fixed now. @ev-mp is there any PR / commit hash we can associate with this?

RealSenseSupport, 2021-07-13:
@rafaelspring Thanks for you trust! As the issue was created more than one year ago and there're so many code changes since then. I'm afraid it's not that easy to locate which PR/commit related to this issue. If you don't think this is issue for you, can we close it now? Looking forward to your reply. Thanks!

rafaelspring, 2021-07-13:
@RealSenseSupport Sure, feel free to close this.

RealSenseSupport, 2021-07-13:
@rafaelspring Thanks for your reply! Close the issue accordingly.

closed 2021-07-13 by RealSenseSupport