realsenseai/librealsense #6425: rs2::config::disable_stream() and disable_all_streams() have no effect when using callback.

rafaelspring, 2020-05-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415, D435, D435i, L515 | 
| Firmware Version                   | any | 
| Operating System & Version |  Win10 | 
| SDK Version                            |  2.34.0 | 
| Language                            |  C++ | 

Another follow-up to #6404 , another issue that I've been able to isolate.

This one is very easy to reproduce. The example below is almost exactly the rs-callback example with the only difference being the pipeline being configured using an `rs2::config` object.

Expected behavior is that the disabled streams would not deliver frames through the callback.
Actual behavior is that they still do (it looks as if the rs2::config object used on the pipeline has no effect).

```cpp
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <iostream>
#include <map>
#include <chrono>
#include <mutex>
#include <thread>

// The callback example demonstrates asynchronous usage of the pipeline
int main(int argc, char * argv[]) try
{
    //rs2::log_to_console(RS2_LOG_SEVERITY_ERROR);

    std::map<int, int> counters;
    std::map<int, std::string> stream_names;
    std::mutex mutex;

    // Define frame callback
    // The callback is executed on a sensor thread and can be called simultaneously from multiple sensors
    // Therefore any modification to common memory should be done under lock
    auto callback = [&](const rs2::frame& frame)
    {
        std::lock_guard<std::mutex> lock(mutex);
        if (rs2::frameset fs = frame.as<rs2::frameset>())
        {
            // With callbacks, all synchronized stream will arrive in a single frameset
            for (const rs2::frame& f : fs)
                counters[f.get_profile().unique_id()]++;
        }
        else
        {
            // Stream that bypass synchronization (such as IMU) will produce single frames
            counters[frame.get_profile().unique_id()]++;
        }
    };

    // Declare RealSense pipeline, encapsulating the actual device and sensors.
    rs2::pipeline pipe;

    // CONFIG HERE:
    rs2::config cfg;
    cfg.disable_all_streams();   // <-- Instead of this call ...
//    cfg.disable_stream(RS2_STREAM_COLOR);   // we can also do this ...
//    cfg.disable_stream(RS2_STREAM_DEPTH);   // or this ...
//    cfg.disable_stream(RS2_STREAM_INFRARED);  // or this...
    // .. or all of them, the effect is the same (no effect).

    // Start streaming through the callback with default recommended configuration
    // The default video configuration contains Depth and Color streams
    // If a device is capable to stream IMU data, both Gyro and Accelerometer are enabled by default

    rs2::pipeline_profile profiles = pipe.start(cfg, callback);

    // Collect the enabled streams names
    for (auto p : profiles.get_streams())
        stream_names[p.unique_id()] = p.stream_name();

    std::cout << "RealSense callback sample" << std::endl << std::endl;

    while (true)
    {
        std::this_thread::sleep_for(std::chrono::seconds(1));

        std::lock_guard<std::mutex> lock(mutex);

        for (auto p : counters)
        {
            std::cout << stream_names[p.first] << "[" << p.first << "]: " << p.second << " [frames] || ";
        }
        std::cout << std::endl;
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

```

MojamojaK, 2020-05-19:
I think this is how its supposed to work, because calling pipe.start() without any parameters would open default streams. I think it's safe to say that passing a config with everything disabled would make it behave similarly.
Besides, what's the point of starting nothing?

rafaelspring, 2020-05-19:
@MojamojaK 
I don't think it is how it's supposed to work. Consider the following code:

```cpp
    rs2::config cfg;  // Expectation is that this carries the def. config.
    cfg.disable_stream(RS2_STREAM_COLOR); // Def. config. minus color.
    rs2::pipeline_profile profiles = pipe.start(cfg, callback);

    // But we are still going to receive color frames...
```

MojamojaK, 2020-05-19:
**Okay, with that I totally agree.**

A workaround I could think of right now would be 
1. to discard unwanted frames.
ex. ) When you only want pose frames
```C++
auto callback = [&](const rs2::frame& frame)
    {
        if (frame.get_profile().stream_type() != rs2_stream::RS2_STREAM_POSE) return;
        // ... do whatever with pose frame here
    };
```
I understand. This can be somewhat inefficient.

2. to use frame_queue/syncer and custom callbacks per sensor
https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0
Although it is the standard way, I personally never use pipelines and always use frame_queue/syncer because I think you can have more control over how librealsense uses CPU/memory/time/sensors although this does introduce a bit of complexity to your code.

rafaelspring, 2020-05-19:
@MojamojaK Thanks for the link on frame buffering. Now that I read it I remember having used `syncer` in the past but then completely forgot about it. Perhaps I need to reconsider it.

RealSenseCustomerSupport, 2020-05-28:
----------------------------------------------

@rafaelspring Thanks for pointing this out! This issue is very similar to https://github.com/IntelRealSense/librealsense/issues/3919 We have reported this to our engineering team. Any update will let you know. Thanks!

RealSenseSupport, 2021-04-09:
@rafaelspring The issue fixed with the latest SDK https://github.com/IntelRealSense/librealsense/releases/tag/v2.44.0. Please have a try and let us know if any issue. Thanks!

RealSenseSupport, 2021-04-27:
@rafaelspring We didn't hear from you for weeks and will close this ticket. Please feel free to create new one if you still have any questions or issues. Thanks!

closed 2021-04-27 by RealSenseSupport