realsenseai/librealsense #6428: Where is the source of  Depth Map Improvements for Stereo-based Depth Cameras on Drones

patrickpoirier51, 2020-05-20:
Hello,

On this white paper
https://dev.intelrealsense.com/docs/depth-map-improvements-for-stereo-based-depth-cameras-on-drones#section-references

They make reference to the source code:
_Our highly optimized algorithms for post-processing are released as open source under https://github.com/IntelRealSense/librealsense._

Where is the code ?
The only post-processing code I see is here:
https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing

And I cannot relate this to the  paper.

MartyG-RealSense, 2020-05-20:
Files for the various filter types can be found in the **proc** folder of Librealsense.

https://github.com/IntelRealSense/librealsense/tree/master/src/proc

ev-mp, 2020-05-20:
@patrickpoirier51 , the demo you mention is one of the examples that utilizes post-processing filters.
To see the source code follow the link provided by @MartyG-RealSense.
But before delving into the code I would suggest to go through [its documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md)

patrickpoirier51, 2020-05-20:
Thanks for quick response 
The white paper refers to different techniques based on openCV , maybe @dorodnic can answer as he is one of the authors

dorodnic, 2020-05-20:
Hi @patrickpoirier51 
I placed the implementation under [wrappers/opencv/depth-filter](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/depth-filter)
Sorry its not easy to find, feel free to ask any question about the code

patrickpoirier51, 2020-05-20:
@dorodnic  ah cool thanks
Maybe you could a direct reference on the white paper ?

Regards

patrickpoirier51, 2020-05-20:
@dorodnic  I do have a question:
Did you implemented this technique to the flight controler stack for onboard navigation ?
If so, how did you extract information to sent to Flight Controller?

Regards

MartyG-RealSense, 2020-06-05:
Tagging @dorodnic as a reminder to kindly provide input for the two questions above from @patrickpoirier51  - thanks!

dorodnic, 2020-06-21:
Hi @patrickpoirier51 
Thanks @MartyG-RealSense , I apologise for late response. 
I've updated the link in the white-paper.
Unfortunately, I cannot answer your second question - my contribution to this project was limited to camera software.

MartyG-RealSense, 2020-07-05:
Case closed due to no further comments received.  Thanks everyone!

closed 2020-07-05 by MartyG-RealSense