realsenseai/librealsense #6430: D435i memory leak problem

cdb0y511, 2020-05-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (5.12.03) | 
| Operating System & Version |   {Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { l 2.<33>.<1> }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi, I am suffer from D435i memory leak problem when I use D435i to capture depth continuously.

Frames is use cv::Mat to restore depth and RGB frames. 
Even I delete Frames, release cv::Mat .the video buffer seems still in the memory.
I wonder if the frameSet from pipe.wait_for_frames() or rs2::frame filtered is released automatic and properly . I find release frame method in the frameSet object, it seems release at end of function. Do I need to release the frameSet manually? and How?
Thx.

part of my code:
  void RealsenseInputSource::simpleFrames(rs2::pipeline pipe, Frames &frames) {
        bool alignedColor = config.alignedColor;
        //bool alignedInfrared = config.alignedInfrared;
        rs2::align alignToColor(RS2_STREAM_COLOR);
        rs2::align alignToDepth(RS2_STREAM_DEPTH);
        //rs2::align alignToInfrared(RS2_STREAM_INFRARED);

        auto depth_stream_file = pipe.get_active_profile().get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();

        int width = depth_stream_file.get_intrinsics().width;
        int height = depth_stream_file.get_intrinsics().height;

        auto frameSet = pipe.wait_for_frames();
        if (alignedColor) {
            frameSet = alignToColor.process(frameSet);
        }
//        else if(alignedInfrared){
//            frameSet = alignToInfrared.process(frameSet);
//        }
        else {
            frameSet = alignToDepth.process(frameSet);
        }
        auto color = frameSet.get_color_frame();
        Mat rgbFrame_Copy;
        Mat(Size(width, height), CV_8UC3, (void *) color.get_data(), Mat::AUTO_STEP).copyTo(rgbFrame_Copy);
        //cv::cvtColor(rgbFrame_Copy, frames.rgbFrame_, cv::COLOR_BGR2RGB);
        //rgbFrame_Copy.copyTo(*frames.rgbFrame_);
        frames.rgbFrame_ = boost::make_shared<cv::Mat>(rgbFrame_Copy);
        // use SDK spatial_filter temporal_filter
        rs2::spatial_filter spat_filter;
        rs2::temporal_filter temp_filter;
        rs2::threshold_filter thresh_filter(0.15,1.7);
        rs2::frame filtered = frameSet.get_depth_frame();
        filtered = thresh_filter.process(filtered);
        filtered = spat_filter.process(filtered);
        filtered = temp_filter.process(filtered);

        frames.internalParams.depthScale = rs2::sensor_from_frame(filtered)->get_option(RS2_OPTION_DEPTH_UNITS);
        //cout<<depthImage.at<u_short>(0,0)<<endl;
        //depthImage.copyTo(*frames.depthFrame_);
        int size = filtered.get_data_size();
        void* tmp2 = (void*)filtered.get_data();
        frames.depthFrame_ =  boost::shared_ptr<Mat> (new Mat(Size(width, height), CV_16UC1, tmp2, Mat::AUTO_STEP));
}

cdb0y511, 2020-05-20:
and in ~Frame()
depthFrame_  -> release();
rgbFrame_  -> release();
rgbFrame_  -> reset();
depthFrame_  -> reset();

MartyG-RealSense, 2020-05-20:
Hi @cdb0y511   With cv::mat, you can forcibly clear its memory with the **deallocate()** instruction, though you should not try to use the data of cv::mat after the deallocation.

The link below discusses the differences between **deallocate()** and **release()**.

https://stackoverflow.com/questions/38328063/what-does-cvmats-deallocate-method-do

cdb0y511, 2020-05-20:
@MartyG-RealSense 
Thanks, but it seems memory leak  still exists ,even I use  deallocate() instead of release().
Do rs2::frameSet or rs2::frame need manually release?

MartyG-RealSense, 2020-05-20:
Resources used by the pipeline can be released by stopping the pipeline with **Stop()** according to the documentation.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html#a142ee1adb798f9a03f86eb90895dd8a5

cdb0y511, 2020-05-23:
> Resources used by the pipeline can be released by stopping the pipeline with **Stop()** according to the documentation.
> 
> https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html#a142ee1adb798f9a03f86eb90895dd8a5

I don't think stop and pipeline is a good ideal. Cause it always takes a few frames to enable auto exposure . The realsense-view never suffer from memory leak. Any advice ?

MartyG-RealSense, 2020-05-23:
I did another check of your script and noticed that you seem to be doing alignment first and then post-processing.  The recommended order of operations is to do post-processing first and then alignment afterwards.  This can help to avoid distortion problems such as aliasing (jagged lines).

Also, if you are experiencing slowdown then this may be a consequence of applying alignment and three post-processing filters, which are demanding processing operations that consume CPU resources.  Offloading CPU work onto the graphics GPU can improve performance.

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

If you are releasing and deallocating cv::mat and still have a resource drain, I wonder if it might be instead coming from the aligning and filtering.

https://github.com/IntelRealSense/librealsense/issues/2115#issuecomment-406873087

cdb0y511, 2020-05-24:
> I did another check of your script and noticed that you seem to be doing alignment first and then post-processing. The recommended order of operations is to do post-processing first and then alignment afterwards. This can help to avoid distortion problems such as aliasing (jagged lines).
> 
> Also, if you are experiencing slowdown then this may be a consequence of applying alignment and three post-processing filters, which are demanding processing operations that consume CPU resources. Offloading CPU work onto the graphics GPU can improve performance.
> 
> [#4905 (comment)](https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888)
> 
> If you are releasing and deallocating cv::mat and still have a resource drain, I wonder if it might be instead coming from the aligning and filtering.
> 
> [#2115 (comment)](https://github.com/IntelRealSense/librealsense/issues/2115#issuecomment-406873087)
second one solved my problem. thanks 
Any other object should not create inside frame stream or  just initialize once?
I aware all filter type should not create inside the frame stream.
I wish to have a notice in the document.  It  may help many developers.

closed 2020-05-24 by cdb0y511

MartyG-RealSense, 2020-05-24:
Great news that your problem was solved!

Could you create a new case please just for your documentation request.  Then I can label it so that it can be tracked by RealSense team members responsible for documentation changes.  Thanks.    :)

cdb0y511, 2020-05-28:
> Great news that your problem was solved!
> 
> Could you create a new case please just for your documentation request. Then I can label it and keep the case open so that it can be tracked by RealSense team members responsible for documentation changes. Thanks. :)

Sorry for late reply.
I have opened a new case
Documentation request for memory leak guidance #6471

MartyG-RealSense, 2020-05-28:
I have labelled your feature request so it can be tracked.  Thanks!