realsenseai/librealsense #6431: Compression and Decompression in Unity

fajarnugroho93, 2020-05-20:
Hi all,

I have seen that there are LZ4 and libjpeg-turbo implementations from the ethernet branch in the 2.34.0 branch for ethernet networking purposes. In Unity, I could send byte arrays via TCP from one program to another. My goal is to compress capture data from the RealSense, transmit it, then decompress it, all in Unity. The question is, which compress-decompress functions should I call and how to call them? 

I have seen JpegCompression.cpp and Lz4Compression.cpp, but I could not call them in Unity. Are these currently not included in the c# wrapper? How to check this?

Is my goal possible to be done?

I hope the questions are not too dumb, I am not that good with c++

MartyG-RealSense, 2020-05-22:
Apologies for the delay in responding to your question.  I have been considering ways in which your request for depth data compression might be able to be efficiently met.

In SDK 2.32.1 a new compressed depth format called **Z16H** was introduced, along with a processing block for decoding it.

[image]

Do you have the option to select **Z16H** format for your depth stream in Unity in SDK 2.34.0, please?  I have not currently got the Unity RealSense wrapper installed, so cannot check myself if it is available in Unity yet.

fajarnugroho93, 2020-05-23:
No worries! Thank you for responding. Yes, Z16 is the default format option for depth in Unity.
[image]

How do I proceed with this?

MartyG-RealSense, 2020-05-23:
There is more information on Z16H format in the link below:

https://github.com/IntelRealSense/librealsense/pull/5564

Though Z16H format is in the SDK, it also requires support in the firmware driver.  I have not got official confirmation on whether it is in the firmware yet - the official release notes for the firmware do not mention it yet.

@dorodnic @ev-mp  Do you have any knowledge about the current status of firmware support for Z16H that you can share publicly, please?

fajarnugroho93, 2020-05-23:
Oops, I just realized that you meant Z16H not Z16. But yet, Z16H is also available.

[image]

MartyG-RealSense, 2020-05-23:
Great, good to know that Z16H is selectable.  Are you able to use it to stream depth in Unity please instead of Z16?

fajarnugroho93, 2020-05-24:
Ah, apparently there is this error when I tried to use Z16H:

`ExternalException: rs2_pipeline_start_with_config(pipe:000001D047ED71C0, config:000001D047ED7E80)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <d373078ba8884c7fb564d7ab8d7595f0>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <d373078ba8884c7fb564d7ab8d7595f0>:0)
RsDevice.OnEnable () (at C:/Thesis/RS-Mirror/Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)`

https://github.com/IntelRealSense/librealsense/blob/bc64e52c9e128aa82cc9f25a14283bbe2bcf8828/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsDevice.cs#L68-L73

MartyG-RealSense, 2020-05-24:
Yes, it may be that although the SDK supports Z16H, it still needs firmware support to work completely.  This has been the case with other features such as camera hardware sync, where it was necessary to wait for both the SDK and firmware components of the feature to be complete before they could be used.  It seems as though Z16H could be an ideal solution for your project needs though.

fajarnugroho93, 2020-05-25:
I see. Okay then, I will just wait for the firmware update for it.

Thank you!

MartyG-RealSense, 2020-06-05:
Hi @fajarnugroho93  Is it okay to close this case please whilst you are waiting for a firmware that supports Z16H?   Thanks!

fajarnugroho93, 2020-06-05:
Okay, I'm closing it now.

closed 2020-06-05 by fajarnugroho93

MartyG-RealSense, 2020-06-05:
@fajarnugroho93   Thanks very much!