realsenseai/librealsense #6432: error: Connection to server failed:  - -10057

Tomblarom, 2020-05-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Raspbian GNU/Linux 10 (buster)  | 
| Kernel Version (Linux Only)    |  4.19.97-v7l+ | 
| Platform                                 | Raspberry Pi 4 - 4GB  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 
 
### Issue Description
I installed the librealsense SDK using this guide: [acrobotic / Ai_Demos_RPi - Raspberry Pi 4 and Intel RealSense D435](https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435), decided to use the RPi4 as backend instead, followed this guide: [librealsense/doc/libuvc_installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md) and now I'm getting this error on connect:

`[RsRTSPClient::getStreams] error: Connection to server failed:  - -10057`

```
pi@scanner:~ $ rs-enumerate-devices
Device info:
    Name                          :     Intel RealSense D435
    Serial Number                 :     ***
    Firmware Version              :     05.12.03.00
    Recommended Firmware Version  :     05.12.03.00
    Physical Port                 :     2-1-4
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B07
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     D400
    Asic Serial Number            :     ***
    Firmware Update Id            :     ***
```

Wild guess: Camera is locked somehow, since I installed the default SDK beforehand.

dorodnic, 2020-05-20:
Hi @Tomblarom 
The camera is ok. 
Could you please explain what you are trying to do? You are getting the error "[RsRTSPClient::getStreams] error: Connection to server failed: - -10057" when you do what?

MartyG-RealSense, 2020-05-20:
Hi @Dorodnic it looks as though @Tomblarom is using the old backend install guide before RSUSB was introduced.

Hi @Tomblarom   A new UVC backend method was introduced in SDK version 2.30.0 in November 2019 (the linked-to guide was last updated in February 2019).  

Under the **RSUSB** backend method introduced in SDK 2.30.0, the steps (which require an internet connection for the build process) are:

1.  Go to the Librealsense root directory and create a **Build** folder, and then change to that created folder using the command below:

**mkdir build && cd build**

2.  Whilst in the Build folder, run the CMake command below to install Librealsense without dependency on Linux version, kernel version or patches:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

dorodnic, 2020-05-20:
Yes, but I don't think this is what causing the issue. RsRTSPClient error seems related to 
[network camera](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras) capabilities we introduced recently. 
Given output from `rs-enumerate-devices` I believe the Viewer should work on the Pi as well.

Tomblarom, 2020-05-20:
Yes there was this guide, which referred to the libuvc. Thought it's the official way to go: [Open-Source Ethernet Networking for Intel® RealSense™ Depth Cameras](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras) Basically the first link, when you google "d435 network".

So what's the right way for making your RealSense Camera available to the network, so you can use it remote with all SDK commands?

dorodnic, 2020-05-20:
Great, yes, this is what I assumed. 
I think the camera is working on your RPi. 
In order to have it available on the network you need to build and run `rs-server` (this step is not in the guide you linked but should be in the white-paper)
However before you do, I wanted to point your attention to - [section-2-3-preparing-the-sd-card](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card)
You can just download a ready-to-go SD-card image for the Pi with everything configured.
Once you boot into it, you can still absolutely tinker with it and modify it as you please, but its a good starting point.

Tomblarom, 2020-05-20:
I'm no fan of ready-to-go packages. It may work perfectly, but as soon as you want to modify things, it gets tricky, since it's not documented what exactly has been modified. Regardless of that, am I able to access all SDK commands, exactly like if the camera is plugging in directly?

dorodnic, 2020-05-20:
Not all. All is in the document - [section-3-6-software-limitations](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations)
You can also look into [section-3-2-building-from-source](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-2-building-from-source) for replicating the results yourself.

Tomblarom, 2020-05-20:
So there is no way to stream 1920x1080 in RGB mode?

dorodnic, 2020-05-20:
That is correct. Gigabit Ethernet is still 5 times slower than USB3, and low-power Arm boards do often struggle with Full-HD cameras (Raspberry Pi 4 looks very good, but I have encountered issues on similar boards) 
It can be done, if you are building yourself, you can just add 1920x1080 resolution here - [tools/rs-server/RsServer.cpp:68](https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-server/RsServer.cpp#L68)
I didn't have sufficient test data to recommend it at the time of writing the guide.

Tomblarom, 2020-05-25:
Thanks for this hint. For now I'm going to process the camera image directly on the RPi3, maybe going back to this in the future.

MartyG-RealSense, 2020-06-05:
@Tomblarom   Do you require further assistance on this case please, or can it be closed?  You are welcome to create a a new issue in the future if you go back to this.  Thanks!

MartyG-RealSense, 2020-06-14:
Closed due to no further comments received.

closed 2020-06-14 by MartyG-RealSense