realsenseai/librealsense #6433: RS Viewer showing "Frame Metadata Disabled" and other Metadata questions

rafaelspring, 2020-05-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D4xx | 
| Firmware Version                   | 5.12.03.00  | 
| Operating System & Version | Win10 | 
| SDK Version                            |  2.34.0  | 

Sometimes, not always, the RS Viewer shows in the top right corner a little pop-up saying "Frame Metadata Disabled" and asking me to enable it.

I have some questions about this (I already read https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md but it's very technical and doesn't answer my questions):

- What determines whether this shows up in the Viewer or not? Why does it show up sometimes but not always? I couldn't locate the respective parts in the RS Viewer source code.
- What are the benefits of enabling metadata? Will that unlock specific camera capabilities? What are the downsides of not enabling metadata? Is frame metadata primarily useful for development / debugging or also for release?
- I remember reading somewhere that on some D4xx models the ability to sync Depth and Color depends on the availability on frame metadata. Is that still correct? If yes does "frame metadata disabled" mean there will be problems with sync?

MartyG-RealSense, 2020-05-20:
1.  I can't say if this is a definitive cause of the 'Frame Metadata disabled' message appearing, but for me personally it usually occurs when I unplug the camera from the USB port and insert it into a different USB port.

2.  Metadata can be useful for certain functions such as getting the real value of the current exposure or retrieving timestamp information about camera frames.  You can read the frame metadata documentation in the link below to see whether it is likely to be relevant to your particular project.

3.  If you are syncing color and depth on the same camera (i.e not doing multiple-camera hardware sync), matching of frame timestamps may be an important factor.

https://github.com/IntelRealSense/librealsense/issues/1548#issuecomment-389070550

https://dev.intelrealsense.com/docs/frame-metadata

dorodnic, 2020-05-20:
Hi @rafaelspring 
You are right about the document being too technical, but it is a very technical subject. 
I'll try to answer your questions - 
To be able to use RealSense as a regular web-cam our cameras are implementing standard video streaming protocol (UVC). First version of this protocol was limited only to video streaming - you were unable to send any extra data with frame pixels. And unfortunately, this is the version most operating systems implement to this day. UVC 1.5 spec introduced the concept of frame metadata, but on Windows it was implemented as an opt-in feature. You need to add special registry key per camera to enable it. 
Hence the first answer - if registry key was added, everything will work. Else, the Viewer will suggest to handle it for you. The SDK doesn't do it unless you allow it, because it requires elevated privilege. 
Main information sent via metadata channel is the hardware timestamp. If you know when each frame was taken, and not just when it arrived, you can better account for latency and do better job of synchronizing between streams. It's all handled automatically in the SDK.
In addition, as @MartyG-RealSense mentioned we send more useful data with each frame, like actual exposure and gain.

rafaelspring, 2020-05-20:
@dorodnic @MartyG-RealSense  Thanks a lot! And since we're on the subject, more questions (sorry for being a nag here but we need to have depth/color/IMU sync and timestamps **completely nailed** for our application):

- Does this mean that for every individual camera that uses UVC and **does not** have the registry key, depth/color sync (within that individual camera) will be challenging?
- Is it possible to use a different protocol than UVC where the frame metatata is a non-issue (i.e. always available)?
- If `frame::get_frame_timestamp_domain` reports "Hardware Clock", can I assume that frame metadata is enabled?
- How does build option ENFORCE_METADATA affect things? Is this a requirement to get any sort of useful metadata over UVC?

dorodnic, 2020-05-20:
This can be addressed system-wide by installing a driver. I preferred to offer a method in the SDK that would not require custom drivers. 
If you install [Windows® 10 UWP Driver for Intel® RealSense™ D400 Product Family](https://www.intel.com/content/www/us/en/support/articles/000026698/emerging-technologies/intel-realsense-technology.html) it will enable metadata for RealSense devices. 
Basically, the firmware will always be UVC, but you don't have to use Windows inbox driver. 
We provide another custom driver for Win7, and you could install it on Win10, circumventing metadata issue, but I'm not sure if the IMU will work well with that driver. 

> If frame::get_frame_timestamp_domain() reports "Hardware Clock", can I assume that frame metadata is enabled?

Exactly

> How does build option ENFORCE_METADATA affect things? Is this a requirement to get any sort of useful metadata over UVC?

This flag is more useful to SDK developers. I recommend using the binary installer instead for Windows. 
In short, this flag says: "if I'm not compiling metadata correctly, please give me a compiler error and not just warning"

MartyG-RealSense, 2020-05-20:
@dorodnic  Thanks very much for replying to the latest questions.   I was just typing a response and had pretty much the same for the latter two questions but your answers to the first two were better than mine.  :)

rafaelspring, 2020-05-20:
@dorodnic Thanks again. Here's the bad news: Our app is classic Win32, not UWP. I assume that means the driver isn't going to be useful to us?
But here's the good news: We ship a realsense2.dll with our app so we're free to do whatever we like on the SDK side.

So does that mean we're back to registry modifications to enable frame metadata in our app?
Any best practices that you can share would be highly appreciated.
Our goal is to ensure we have HW time stamps in our app after our app is installed, without the user having to configure anything manually on his end.

rafaelspring, 2020-05-23:
@dorodnic can you share more info on this?

dorodnic, 2020-05-24:
Hi @rafaelspring 
Sorry for the delay. 
You can still redistribute and use the driver - it enables some UWP functionality, but also handles metadata for Win32 applications. 
Enabling metadata from code is not part of SDK API, but you can copy the code freely into your application / installer. Everything you need is here - [/common/metadata-helper.cpp:282](https://github.com/IntelRealSense/librealsense/blob/master/common/metadata-helper.cpp#L282)

rafaelspring, 2020-05-24:
@dorodnic Thanks! So just to be clear, would we have to use both the driver **and** the metadata-helper, or only one of them?

dorodnic, 2020-06-02:
Sorry for the delay
You can use either one. 
On Android no action is required. Similar to Win7, Android and Mac OS are using our own userspace UVC driver.

rafaelspring, 2020-06-02:
@dorodnic Thanks!

rafaelspring, 2020-07-06:
@dorodnic 
I just noticed 3 things:

1. [This code](https://github.com/IntelRealSense/librealsense/blob/master/common/metadata-helper.cpp#L213) looks broken. [ShellExecuteEx](https://docs.microsoft.com/en-us/windows/win32/api/shellapi/nf-shellapi-shellexecuteexw) returns a bool, not an error code.

2. In the [same function](https://github.com/IntelRealSense/librealsense/blob/master/common/metadata-helper.cpp#L238) not all control paths return a value.

3. The driver you linked [(Windows® 10 UWP Driver for Intel® RealSense™ D400 Product Family)](https://www.intel.com/content/www/us/en/support/articles/000026698/emerging-technologies/intel-realsense-technology.html) seems to be only for D400 family. Would it also enable frame metadata for the L515 or is there a different driver?

rafaelspring, 2020-07-06:
@dorodnic 

4. For some reason this call to [ShellExecuteEx](https://github.com/IntelRealSense/librealsense/blob/master/common/metadata-helper.cpp#L213) does not re-start the RealSense Viewer. However if we use the same code in our app (even after fixing 1) our app re-starts, which makes sense given that it's a new invokation of the executable. What does the RS Viewer do differently to avoid the re-start?

rafaelspring, 2020-07-07:
@dorodnic 
W.r.t. 4), by looking into RS-Viewer's `CMakeLists.txt` I just found [windows-app-bootstrap.cpp](https://github.com/IntelRealSense/librealsense/blob/master/common/windows-app-bootstrap.cpp) which explains everything. Not knowing this file it's basically impossible to understand the whole Admin-elevation process of the RS Viewer.

dorodnic, 2020-07-09:
Hi @rafaelspring 
Sorry, I should have linked that file as well. The viewer is being restarted during permission elevation. I'll look into control flow of the function. As for UWP driver for L515 I think there is no clear statement at this moment. If you raise it as a requirement its more likely to get prioritised.

RealSenseSupport, 2020-07-16:
Hi @rafaelspring 

As @dorodnic states the UWP for L515 support would be a new requirement and would need to be raised that way.  I believe you have direct access to RealSense support, if you submit that request via that direct support that would help.

Outside of the UWP, is there anything else on this thread needed?
*If we don’t hear from you in 7 days, this issue will be closed

rafaelspring, 2020-07-20:
@RealSenseSupport @dorodnic Filed a ticket with RealSense support on this. Closed the issue but may re-open if issue persists after new UWP driver release or in case there are more questions on on-demand Metadata code.

closed 2020-07-20 by rafaelspring