realsenseai/librealsense #6436: Ubuntu 20.04 release

dheera, 2020-05-21:
Hi,
I was wondering what the timeline will be for a 20.04 release of this driver?

MartyG-RealSense, 2020-05-21:
Hi @dheera   Whilst I do not have the information that you are seeking, a check of the public 'Commits' addition list for the next SDK version (probably named **2.35.0**) does not indicate a mention of Ubuntu 20.04, suggesting that it will be introduced at an unknown future date after the next version. 

https://github.com/IntelRealSense/librealsense/compare/v2.34.0...development

Wallbraker, 2020-06-03:
Any update on this?

Since Ubuntu 20.04 uses Python 3.8 #6296 will need to be fixed first.

MartyG-RealSense, 2020-06-03:
@Wallbraker   No further news yet about Python 3.8 supporting Pyrealsense2 installation with pip instead of having to build the wrapper manually.  Sorry.

dheera, 2020-06-08:
@Wallbraker @MartyG-RealSense Even if we could skip Python 3.8 and support only C++ for now that would be amazingly useful.

MartyG-RealSense, 2020-06-14:
@dheera  @Wallbraker   I have filed an official feature request with Intel for a pyrealsense2 wheel file for Python 3.8 so that pip install for pyrealsense2 can be used with Python 3.8 instead of compiling the wrapper from source code.

https://github.com/IntelRealSense/librealsense/issues/6296#issuecomment-643743979

Only a small number of cases have been received so far about using librealsense on Ubuntu 20.04.  At least one user has successfully installed it on 20.04 using the **backend** installation method, with minimal performance issues experienced.  This may be the best method to use for 20.04 until official support arrives in the SDK.

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

So as I do not yet have a date for official 20.04 support to be released, and the backend installation method seems to work with 20.04 in the meantime, I will close this case.  Thanks for your inputs!

closed 2020-06-14 by MartyG-RealSense

vkuehn, 2020-06-16:
it's sad you have closed it. 20.04 is the latest longtime version it's like supporting window 8 but not 10...

MartyG-RealSense, 2020-06-16:
@vkuehn   I understand your concern, but there is no practical reason to keep it open.  I have filed an official feature request with Intel for the creation of a Python 3.8 wheel file, and it is highly likely that the RealSense developer team will add official Ubuntu 20.04 support at a future date.

caelinsutch, 2020-07-01:
What are the full steps for backend installation? After running these commands, what else do we have to do?

MartyG-RealSense, 2020-07-02:
There are instructions for building from source with the backend method (which requires an internet connection and the downloading and extracting of a zip file of the SDK source code) in the link below.  

https://github.com/IntelRealSense/librealsense/issues/6689#issuecomment-648758973

If you wish to build the Pyrealsense2 wrapper at the same time as installing the SDK (which may work better for you on Ubuntu 20.04 than building the wrapper from pip or source code) then please use the modified CMake instruction below:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

caelinsutch, 2020-07-02:
@MartyG-RealSense Only worked after running `sudo make clean` before running `sudo make install`

MartyG-RealSense, 2020-07-02:
Thanks very much for the update @caelinsutch !

caelinsutch, 2020-07-02:
@MartyG-RealSense One question, even after building/installing succesffully, when trying to build the ROS2 Foxy wrapper, it doesn't find this package. Looking around the file system, there's no cmake files for librealsense2. Is that expected? How would I go about getting a cmake file

MartyG-RealSense, 2020-07-02:
@caelinsutch   I do not have any knowledge of the Foxy wrapper, so it is difficult to advise.  I would guess that the **sudo apt-get install ros-foxy-librealsense2** instruction is a script to install Librealsense from the official master branch.  Since you have already built Librealsense using CMake, you can probably skip this step.

The libssl-dev and libglfw3-dev files described in step 3 of the instructions are usually installed when compiling librealsense from source using the usual non-backend method. 

Since you have already built librealsense through the backend method (which installs all the dependencies over the internet), you can probably skip this stage too.

I would therefore progress straight to installing **ros2_intel_realsense** from source.

https://github.com/intel/ros2_intel_realsense/tree/refactor#install-ros2_intel_realsense-from-source

***************

I have never hard of the Foxy wrapper until now though, so the above advice - skipping straight to installing ros2_intel_realsense - is speculation on my part rather than certainty, so I apologise in advance if it does not work.

caelinsutch, 2020-07-02:
@MartyG-RealSense I cannot believe I was that dense :/, I completely missed that part in the readme. Worked perfectly after installing `ros2_intel_realsense` from source via the instructions, I wasn't on the right branch when building from source before. Thank you so much for the help, I really appreciate it!

MartyG-RealSense, 2020-07-03:
@caelinsutch   I'm very pleased to hear that you succeeded.  :)

benbongalon, 2020-07-05:
I'm leaving a note for those who are stuck on the **"config.h: No such file"** compile error even after `sudo make clean && sudo make install`. As suggested in #5518, you need to install the USB programming library development files which contains the missing config.h file.

$ sudo apt-get install libusb-1.0-0-dev

Then rerun cmake and make install. Here is a complete recipe that worked for me:

$ sudo apt-get install libusb-1.0-0-dev

$ git clone https://github.com/IntelRealSense/librealsense.git
$ cd librealsense/
$ mkdir build && cd build
$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
$ sudo make uninstall && make clean && make && sudo make install

MartyG-RealSense, 2020-07-05:
Thanks very much @benbongalon   :)

unigent-robotics, 2020-08-07:
Building it from source on 20.04 LTS worked for me no problem

MartyG-RealSense, 2020-08-07:
Thanks very much for confirmation of what worked for you in your particular case @unigent-software

AndreV84, 2020-08-28:
rules do not seem to build/ install correctly on 20.04; the udev rules;
which would be proper way to get not sudo user to use built binaries?

MartyG-RealSense, 2020-08-29:
@AndreV84   The alternative to building from source is to build from packages using these instructions:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

AndreV84, 2020-08-29:
@MartyG-RealSense Thank you for following up!
shall for ubuntu 20.04 the xenial / bionic repo branch be used? else?
e.g. sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u

MartyG-RealSense, 2020-08-29:
@AndreV84  There was a RealSense user recently who was able to install the Bionic distribution of librealsense on 20.04 but had some problems when using RViz in ROS, as Biionic is meant for 18.04.

https://github.com/IntelRealSense/realsense-ros/issues/1345

You are less likely to encounter problems on 20.04 if you build librealsense from source over an internet connection using the **RSUSB backend** method.

https://github.com/IntelRealSense/librealsense/issues/6436#issuecomment-643754251

masynthetic, 2020-09-10:
@MartyG-RealSense I am similarly trying to use pyrealsense2 with 20.04 and python 3.8 and I have seen a few of your comments reccomending the RSUSB backend method but can't find any examples or more documentation regarding this, could you point me in the right direction?

AndreV84, 2020-09-10:
@masynthetic 
it should be something passed to cmake as in the line below
```
RUN wget https://github.com/IntelRealSense/librealsense/archive/v2.38.0.tar.gz && tar -xvf v2.38.0.tar.gz && cd librealsense-2.38.0/ && mkdir build && cd build &&  cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true -DBUILD_GRAPHICAL_EXAMPLES=true && make -j8 && make install

```
source  https://github.com/AndreV84/jetson-containers/blob/master/ros1_realsense/Dockerfile
However, i tmight also look likew the below
```
cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DBUILD_GRAPHICAL_EXAMPLES=true -DCMAKE_CUDA_ARCHITECTURES=72  -DCMAKE_PREFIX_PATH=~/apriltag
```

masynthetic, 2020-09-11:
Thank you mine almost worked but I am getting this error 

```[ 11%] Performing build step for 'libusb'
Scanning dependencies of target usb
[  9%] Building C object CMakeFiles/usb.dir/libusb/core.c.o
/home/mason/librealsense-2.38.0/build/third-party/libusb/libusb/core.c:23:10: fatal error: config.h: No such file or directory
   23 | #include "config.h"
      |          ^~~~~~~~~~
compilation terminated.```

and I cant seem to find the librealsense2.so and pyrealsense2.so files so it doesn't seem to have worked _enough_

Do you have any suggestions for this, and did you have to use the method of copying those two files into the folder of your script?

AndreV84, 2020-09-11:
obviously libusb apt packages are missing
specifically their -dev version

masynthetic, 2020-09-11:
> obviously libusb apt packages are missed

You're right that was easy. Installed libusb1 dev and it made it past that.

thanks for your help! 

Unfortunately now seems like my little laptop just doesn't have enough memory to compile this.

```
killed signal terminated program cc1plus
```

AndreV84, 2020-09-11:
try without 
make -j6 flags,
just make

AndreV84, 2020-09-11:
librealsense2.so and pyrealsense2.so 
these two are created via renaming of corresponding output .so files unless they are presented after the building as is

masynthetic, 2020-09-11:
Thanks again. I've made it through the install, was able to satisfy the path issue with [this](https://github.com/IntelRealSense/meta-intel-realsense/issues/20)

Now getting
```
WARNING [default] Could not open device failed to set power state
```
:(

AndreV84, 2020-09-12:
try running with sudi;
for the power state resolution

masynthetic, 2020-09-12:
Running realsense-viewer with sudo gets me depth frames (YAY!) but no RGB, however the viewer did give me an error I didn't find anywhere else. My apologies if this is becoming a separate issue, it seems similar to https://github.com/IntelRealSense/librealsense/issues/5828 but not exactly. 

```
Description: RealSense UDEV-Rules are missing!
UDEV-Rules permissions configuration 
 for RealSense devices.

(forgot to copy this part of the message but it said that you can run with sudo (not recommended for some reason) or install the UDEV rules through apt but I wasn't able to find or install through apt so I went with the 3rd option command below ) 
$ sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger
```
this got rid of the error but still no RGB frames. I noticed that both in windows and Ubuntu the camera (in rs viewer) shows as a USB2.1 device despite using USB 3 capable ports and the cable, in windows it is still able to get both frames at reasonable resolution and rate but in Ubuntu no combination of rate, resolution, or order of enable gets me an RGB frame

MartyG-RealSense, 2020-11-18:
I just wanted to let commenters on this discussion know that support for Kernel 5.4 and Ubuntu 20 have been added in the new SDK **2.40.0** release.

AndreV84, 2020-11-18:
does its https://github.com/IntelRealSense/librealsense/releases/download/v2.40.0/Intel.RealSense.unitypackage supportunity 2019 also l515 / d455?

MartyG-RealSense, 2020-11-18:
@AndreV84  The 2.40.0 release notes do not mention D455 and L515 but do contain the Unity updates below:

**#7625 - [Unity] Adapting wrapper to Unity 2018-2020
Deleted GUI Layer from some scenes
Added AR Background Resitrictions for prevent using old classes in Unity 2020)**

AndreV84, 2021-01-22:
udev rules still won't work on 20.04? so won't run without sudo? else?

MartyG-RealSense, 2021-01-22:
@AndreV84   Does it help if you reload the udev rules and then reboot the computer?

https://github.com/IntelRealSense/librealsense/issues/3244#issuecomment-467883911

AndreV84, 2021-01-22:
@MartyG-RealSense 
thanks;
it worked after we re-downloaded, re- make the binaaaries also substituted old binaries with new resulting binaries;
thus resolved

MartyG-RealSense, 2021-01-22:
@AndreV84  Thanks, that's great to hear!

AndreV84, 2021-01-28:
@MartyG-RealSense 
could you suggest a way if GPU/CUDA support works after building the realsense sdk with the parameters below, please?
```
cmake -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true -DBUILD_WITH_CUDA:bool=true -DCMAKE_CUDA_ARCHITECTURES=72 -DCMAKE_CUDA_COMPILER_ENV_VAR=/usr/local/cuda-10.2/bin/nvcc -DCMAKE_CUDA_COMPILER=/usr/local/cuda-10.2/bin/nvcc -DOpenGL_GL_PREFERENCE=GLVND  ..

AndreV84, 2021-01-28:
as we have only remote shell access to the device 20.04 ubuntu with d435
I could see that executing ros2 realsense consumes 1-4% CPU also 1-4%GPU
are there any other headless methods to monitor the CUDA/ GPU use?
Thanks
https://github.com/IntelRealSense/librealsense/issues/6302#issuecomment-619516818

AndreV84, 2021-01-28:
-However, I've noticed that running headless rs-depth causes 0%GPU use

MartyG-RealSense, 2021-01-28:
Hi @AndreV84   I have not seen before any of the CUDA related flags that you are using, except for **-DBUILD_WITH_CUDA**, so I cannot comment on the use of those.

If you are building librealsense for a headless system then I recommend  setting**-DBUILD_GRAPHICAL_EXAMPLES** to False instead of true.  

The CMake **Build Configuration** documentation states about this flag:  "Controls the-subset of examples and tools dependent on OpenGL.  Disabling this option will reduce the set of tools and examples to rs-save-to-file, rs-terminal, rs-enumerate-devices and rs-fw-logger.  Recommended for headless systems without the graphic subsystem".

https://dev.intelrealsense.com/docs/build-configuration

In regard to performance profiling, the SDK's C++ **rs-data-collect** tool may suit your needs.

https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect

AndreV84, 2021-01-29:
@MartyG-RealSense 
Thank you for following up.
Howver, using the rs-data collet with the config below
```
DEPTH,640,480,30,Z16,0
INFRARED,640,480,30,Y8,1
INFRARED,640,480,30,Y8,2
COLOR,640,480,30,RGB8,0
```
results in only one core of 8 [CPU] to be loaded 2-11%
also GPU use is indicated like 0-3%

MartyG-RealSense, 2021-01-29:
@AndreV84   I cannot think of another way to profile CUDA's use of the GPU with a librealsense tool.  Conceivably if you are using a Jetson as the headless device that is being accessed then you could read the CUDA statistics with the **NVidia System Profiler** tool if you are not doing so already.

https://developer.nvidia.com/embedded/nvidia-system-profiler

AndreV84, 2021-05-23:
@MartyG-RealSense 
Could you extend on how we could implement 
realsense camera on jetson - but realsense viewer playing the rgb stream on other network device, please?
Does realsense-server needs to be installed in the jetson with camera or on the playing network computer?
does realsense-net go to the unit with camera or to the playing device?
reference:
https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-7-testing-the-camera

MartyG-RealSense, 2021-05-23:
The ethernet open-source networking white paper states that whilst they are using Raspberry Pi, the guide can be adapted woth minor modifications for other compute boards.

At the time of writing this, **rs-server** supports RGB at 640x480 (6, 15 or 30 FPS) or 424x240 (6, 15, 30 or 60 FPS).

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

In regard to your questions about where **rs-server** and **realsense2-net** should be placed: I should emphasize that I have not built this project myself, but I believe that they go on the compute board (the device with the camera attached) rather than the host because Intel provide a pre-made SD card image in the paper with the files pre-installed.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card

The paper also provides confirmation that **rs-server** runs on the compute device.

[image]

AndreV84, 2021-09-25:
@MartyG-RealSense 
How to add the metaframe thing on x86_64?
```
Unsupported kernel version 5.11.0-36-generic . The Focal patches are maintained for Ubuntu LTS with kernel 5.4 only
```
that is the output of the patch

MartyG-RealSense, 2021-09-25:
@AndreV84   A response has been posted to your question at https://github.com/IntelRealSense/realsense-ros/issues/1604#issuecomment-927139643

AndreV84, 2021-11-12:
@MartyG-RealSense 
if there is l515 camera failing
is the warranty still applicable ?

MartyG-RealSense, 2021-11-12:
@AndreV84   If the camera was purchased more than 30 days ago then whether it can be returned may depend on the particular problem that it has.

If you are having a problem with an L515, please create a new issue about it so that the problem can be analyzed for a potential cause.  Thanks!

AndreV84, 2021-11-16:
@MartyG-RealSense 
This
https://software.intel.com/iot/edgesoftwarehub/download/home/industrialinsights/
 doesn't integrate with realsense ecosystem, does it?

MartyG-RealSense, 2021-11-16:
@AndreV84   I did not find RealSense references in relation to Edge Insights for Industrial, though **Intel Edge Insights For Vision** can be paired with RealSense.

https://www.intel.com/content/www/us/en/developer/topic-technology/edge-5g/edge-solutions/vision-recipes.html?s=Newest

https://www.intel.la/content/dam/www/public/us/en/documents/solution-briefs/esh-aaeon-up-autonomous-mobile-robots-solution-brief.pdf

mintng, 2021-11-30:
@MartyG-RealSense  Hi Marty, I did your method like build the source code using the RSUSB backend. But I do not know what to do next. I mean, how to run the realsense-viewer after 'make'? I can not use roslaunch as well. Thanks for your help!

MartyG-RealSense, 2021-11-30:
If the build has successfully completed and you included the build terms **-DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true** in your CMake build instruction then Linux users should be able to find realsense-viewer (and executables of the other tools and examples) in the usr/local/bin folder.

In regard to ROS, have you built the RealSense ROS wrapper from source code after having completed the build of the librealsense SDK (as the ROS wrapper is not included in the librealsense build by default).

AndreV84, 2021-12-15:
@MartyG-RealSense , howdy
Is RMA possible for realsense unit that failed within 1yr after purchase? What is the procedure to apply for RMA within 1yr? Could you extend, please? Thanks

MartyG-RealSense, 2021-12-15:
@AndreV84  In general, a case describing the problem would be created on this forum would be created by the customer so that we can go through a diagnostic process and establish whether the camera is definitely unrecoverable.   

A camera should ideally be returned to the retailer that it was purchased from within 30 days of purchase, and a retailer may refuse to accept a return if it is later than 30 days even if Intel recommend RMA in a support case.

AndreV84, 2021-12-18:
@MartyG-RealSense so I shall create github issues topic? To start approaching 'broken unit [l515 ] thing
? Thanks

MartyG-RealSense, 2021-12-18:
@AndreV84  You are welcome to create a new issue for the purposes of diagnosing your L515.  Because of the apparent amount of time (months) that you were using the L515 before it had a problem though, there is likely no chance of being able to return it for a replacement now.

[9 more replies]