realsenseai/librealsense #6437: How to get accurate ply file using SDK2.0?

LW-SCU, 2020-05-21:
| Required Info                         |How to get Accurate ply file using SDK2.0?  |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i| 
| Firmware Version                   |05.12.03.00 | 
| Operating System & Version |    Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0-96-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.34.0                          | 
| Language                            |  C++                         | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

I successfully got point cloud using sdk,but the z value of point cloud  is an **integer millimeter**. However, I got point cloud using realsense-viewer tool. the z value is a **float millimeter**.
How do I get accurate ply file using SDK2.0?
[image]
**z vaule = -585mm by sdk**
[image]
**z vaule = -127.441 by realsense-viewer**
This is my code:
```c++
int main(){ 
    pipeline pipe; 
    colorizer c; 
    config cfg; 
    cfg.enable_stream(RS2_STREAM_COLOR,1920,1080,RS2_FORMAT_BGR8,30); 
    cfg.enable_stream(RS2_STREAM_DEPTH,480,270,RS2_FORMAT_Z16,30); 
    pipe.start(cfg); 
    // Declare pointcloud object, for calculating pointclouds and texture mappings 
    rs2::pointcloud pc; 
    // We want the points object to be persistent so we can display the last cloud when a frame drops 
    rs2::points points; 
    int i=0; 
    while (i++ <30) pipe.wait_for_frames(); 
    char k = 0; 
    while (k!='q'){ 
        frameset frameset = pipe.wait_for_frames(); 
        auto colorFrame = frameset.get_color_frame(); 
        Mat color(colorFrame.get_height(),colorFrame.get_width(),CV_8UC3, 
        (void*)colorFrame.get_data(),Mat::AUTO_STEP);

        depth_frame depthFrame = frameset.get_depth_frame(); 
        Mat depth(depthFrame.get_height(),depthFrame.get_width(),CV_16UC1, 
        (void*)depthFrame.get_data(),Mat::AUTO_STEP); 
        auto imdepthFrame = depthFrame.apply_filter(c);

        Mat imdepth(depthFrame.get_height(),depthFrame.get_width(),CV_8UC3, 
        (void*)imdepthFrame.get_data(),Mat::AUTO_STEP);

        Mat resizeColor,resizeDepth; 
        resize(color,resizeColor,Size(960,540)); 
        resize(imdepth,resizeDepth,Size(960,540)); 
        imshow("color",resizeColor); 
        imshow("depth",resizeDepth); 
        k = waitKey(20); 
        if(k=='s'){ 
            pc.map_to(colorFrame); 
            points = pc.calculate(depthFrame); 
            points.export_to_ply("1.ply", colorFrame); 
        }
    } 
    pipe.stop(); 
    return 0; 
}
```

MartyG-RealSense, 2020-05-21:
In most examples I have seen, a float value is achieved by deliberately defining a float type variable to store the coordinate value in.

The link below has an example C++ point cloud script contributed by a RealSense community member where they store their XYZ in a float:

https://forums.intel.com/s/question/0D70P0000069GjESAU

LW-SCU, 2020-05-21:
@MartyG-RealSense Thank your answer, but the link is not my solution.
I want to get the point cloud saved by realsense-viewer using SDK, and the Z value in its point cloud is decimal in millimeter scale. Like the Z value in the second picture above.

MartyG-RealSense, 2020-05-21:
Apologies if I misunderstood.  When you showed an overlay of the coordinates, I thought that you were aiming to create a C++ script that would generate a text read-out of the coordinates as a float value in millimeters instead of a rounded-up integer value.

Given that depth measurements are going to have an error factor in them anyway unless you have the camera very close to the observed object (since error increases as the observed object's distance from the camera increases), it may not make much difference to the accuracy of the results if they are rounded up or down by up to 0.5 mm.  You may get more accurate results by other means.  A few examples of such methods include:

- Using the D435's optimal depth resolution of 848x480
- Using a 'Visual Preset' such as "High Accuracy", or the "ShortRange" preset if you are sensing from close range.
-  Applying post-processing filters.  A range of filters are active by default when using the Viewer, but when writing your own program you have to enable post-processing filters manually

Steven-ZY, 2020-05-26:
@MartyG-RealSense @LW-SCU hello sir, Excuse me. Where can I get the code about obtaining the x, y, z value of the point cloud without rgb information? And why is the z value negative?

LW-SCU, 2020-05-26:
@Steven-ZY 
Get point cloud using save_to_ply Class,example
```c++
#include <librealsense2/rs.hpp>
#include <librealsense2/hpp/rs_export.hpp>
using namespace rs2;
int main(){
    pipeline pipe;
    config cfg;
    rs2::align align_tocolor(RS2_STREAM_COLOR);
    cfg.enable_stream(RS2_STREAM_COLOR,1920,1080,RS2_FORMAT_RGB8,30);
    cfg.enable_stream(RS2_STREAM_DEPTH,480,270,RS2_FORMAT_Z16,30);
    rs2::pipeline_profile profile = pipe.start();
    auto sensor = profile.get_device().first<rs2::depth_sensor>();
    sensor.set_option(RS2_OPTION_VISUAL_PRESET, 
    RS2_RS400_VISUAL_PRESET_MEDIUM_DENSITY);
    while (i++ <30) pipe.wait_for_frames();
    frameset frameset = pipe.wait_for_frames();
    frameset = align_tocolor.process(frameset);
    save_to_ply save("1.ply");
    save.set_option(save_to_ply::OPTION_PLY_BINARY, false);
    save.set_option(save_to_ply::OPTION_PLY_MESH, false);
    save.set_option(save_to_ply::OPTION_PLY_NORMALS, false);
    save.set_option(save_to_ply::OPTION_IGNORE_COLOR, false);
    save.process(frameset);
    return 0;
}
```

Steven-ZY, 2020-05-26:
@MartyG-RealSense Thank you so much. But I can't successfully run it. Here is the error. Could you give me some advice? 
[image]  
By the way, 1) How can I learn to use these classes in the library "liberalsense"? 2) I obtain the point cloud from the "Intel RealSense Viewer.exe", but all the z values of the point cloud are negative. Is it related to the way of building the coordinate system?

LW-SCU, 2020-05-26:
@Steven-ZY 
你应该看得懂中文哈，这个代码我在ubuntu下运行是没有问题的。你把debug编译换成realse试下。
- 我也是刚开始学这个库，它的代码还是挺复杂了，我也是在边看文档，边看源码在解决问题。
- 对于深度值为负的话，应该是坐标系的问题，对我们使用应该没有什么大的影响。
一起加油学习realsense吧

Steven-ZY, 2020-05-26:
@LW-SCU 感谢答疑！我刚刚好像回复错人了。
1)release下面也还是不行
2)我看Intel RealSense Viewer显示的时候Z值是为正的，但是直接在Intel RealSense Viewer保存得到的Z为负，查到有说是librealsense故意将z变为﹣z，为了保存的.ply在一些点云处理软件中可以更方便被查看。 但如果说想按照真实的+z保存该怎么做呢？

LW-SCU, 2020-05-26:
@Steven-ZY 
- 你那个错误应该是运行时的错误，编译应该是通过了的，具体是哪里有问题就需要你去一步一步进行调试一下，看哪行代码异常。
- 对于z的值应该直接取反就可以了，可以当作z的坐标轴反向，我是这样理解的。

closed 2020-05-27 by LW-SCU

wuhaoqi-hub, 2020-06-26:
我在windows下运行显示使用了可能未初始化的本地指针变量“texture_data”，不知道什么问题

LW-SCU, 2020-06-26:
@wuhaoqi-hub 你好，
我没有明白你的问题，我建议你新建一个issue，把相关代码和错误贴上，然后给我留言，如果我尽量帮你解决问题，谢谢