realsenseai/librealsense #6440: [Realsense Viewer] [Multicam] Unable to select cameras (index >0)

MojamojaK, 2020-05-21:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.12.3 (Doesn't matter) | 
| Operating System & Version |   Windows 10  | 
| Platform                                 | PC  |
| SDK Version                            |  Tested & Symptoms Confirmed 2.29.0 ~ 2.34.1 | 

### Issue Description
Able to select a single camera on top of the list ↓↓↓
[image]

Unable to select the ones after that ↓↓↓
[image]

MartyG-RealSense, 2020-05-21:
@dorodnic   I would think that it might be an IMGUI related issue if IMGUI is responsible for the user interface elements?

MartyG-RealSense, 2020-06-14:
@MojamojaK    Do you still experience this UI issue in the newest SDK version 2.35.2 please?

MojamojaK, 2020-06-14:
Just tested on the 2.35.2 viewer
Probelm persists

MartyG-RealSense, 2020-06-14:
It looks like a user-interface bug.  As there are not many other RealSense users who may be adding 20+ cameras into the RealSense Viewer, it is a difficult problem to recreate to prove that it is a bug.  I will seek advice from Intel about it.

MojamojaK, 2020-06-14:
This also happens with 2 cameras connected btw.

MartyG-RealSense, 2020-06-14:
Thanks, I will update my internal report.

MartyG-RealSense, 2020-07-26:
As we have not had any other reports from RealSense users of this particular user-interface effect occurring and have not been able to replicate it, the case will now be closed.  Thank you very much for your report @MojamojaK

closed 2020-07-26 by MartyG-RealSense