realsenseai/librealsense #6441: Realsense D435I error: FrameSync not supported

skychenjian, 2020-05-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435I        | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version | Win 10         | 
| Platform                                 | PC              |
| SDK Version                            |  2.19.0.570  | 
[image]

### Issue Description:
I faced an error when running Elsticfuion.exe which was generated by VS2015:
[RS2] ERROR at FILE D:\Elastic\librealsense\wrappers\openni2\src\Rs2Driver.cpp LINE 258 FUNC oni::driver::Rs2Driver::enableFrameSync
        FrameSync not supported.

This error is the same as (#5001 (comment)) mentioned. 
How to fix it ?

MartyG-RealSense, 2020-05-22:
I have explored your problem very carefully.  There is not much information available on this subject, so I had to make some theories about the cause of frame sync being unavailable.

As a first step, could you text please whether **metadata** support is enabled in Windows.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows

skychenjian, 2020-05-22:
> I have explored your problem very carefully. There is not much information available on this subject, so I had to make some theories about the cause of frame sync being unavailable.
>
> As a first [...]

Hello, Thank you so much for your support.
I checked as you mentioned, it seems metadata support is enabled in Windows because I checked the item **ENFORCE_METADATA** in the Cmake configuration. Configuring done, and Generating done.
[image]
What can I do next step ?

MartyG-RealSense, 2020-05-22:
I had another look at the problem.  It is difficult to diagnose though because there are few references about the **FrameSync Not Supported** issue, and it was originally developed by a non-Intel team called the Palo Alto Group.

https://github.com/PaloAlto-Group/RealSense-OpenNI2-Wrapper

I wonder if it would be possible to try Palo Alto's older version of the OpenNI2 wrapper in the link above to see if the frame sync error still occurs with that version.

skychenjian, 2020-05-25:
I tried, but this error was still happened.
Let me clarified my case as below:
1. Complied and generated "Librealsense" by VS2017. Copied "rs2drverd.dll and realsense2d.dll" to OPENNI2 folder. Operteing "NiViewer", D435I can be detected ;
2. But after generated Elasticfusion acc. the link: 
[https://github.com/mp3guy/ElasticFusion](url)
This error "FrameSync Not Supported" will be happened;
[image]

I doubt if it's problem from Librealsense ? or a problem from Elasticfusion?

MartyG-RealSense, 2020-05-25:
Given that ElasticFusion's last code update was in July 2018, before the D435i model was even launched in late 2018, it is probable that the program was not tested with D435i.  This opens the possibility of an incompatibility with D435i that is not present in the D435 model.

If you need to do Visual SLAM (VSLAM) with a D435i, I wonder if an option for you may be to switch to the modern **OpenVSLAM** software, which seems to be very flexible with the cameras that it will work with.

https://support.intelrealsense.com/hc/en-us/community/posts/360038549493-Highlighting-the-OpenVSLAM-open-source-Visual-SLAM-platform

skychenjian, 2020-05-25:
It's bad news for me. :(
But you mentioned OpenVSLAM, then it means D435i can be compatibility with this OpenVSLAM software, right ? It's totally new for me again.

MartyG-RealSense, 2020-05-25:
I do not know of anybody who has used OpenVSLAM with RealSense, though the instructions say that " the system can deal with various type of camera models, such as perspective, fisheye, and equirectangular.  If needed, users can implement extra camera models with ease".

If you want a SLAM system proven to work with D435i, Intel have this guide that uses ROS:

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

skychenjian, 2020-05-26:
It looks another way. Let me have a try. 
In parallel, I also can try again to compile ElasticFusion in Ubuntu system. 
Anyway, Thank you so much for your support and suggestion.

closed 2020-05-26 by skychenjian