realsenseai/librealsense #6442: View from D435i camera stutters on code examples from github repo

hunterman1, 2020-05-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Android}                                            | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { targetSdkVersion 27 and targetSdkVersion 28 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  { Smartphone  }                          | 

I'm observing that the view from the camera is stuttering. This could be observed on Android 9(Samsung Galaxy A50) and on Android 10(Samsung Galaxy A51). I'm using examples from the github source code..

I tried to install also application from Google Play store(https://play.google.com/store/apps/details?id=com.intel.realsense.camera&pageId=none). On Samsung Galaxy A50 everything works as expected - view from the camera is good, without any stutters or delays.

Is it possible to get source code for this application available on your Google Play store? Maybe some changes were implemented there which allows to run this application without problems.
I'm also attaching .zip file with working (but stuttering) application. It was implemented by "Capture" example available on your github repo.
[Test.zip](https://github.com/IntelRealSense/librealsense/files/4667457/Test.zip)

I can also send you an example movie how this application behaves. Here, I can upload only 10MB so, I can send you it via email/Google drive.

This issue is related with the previous one registered under this link: https://github.com/IntelRealSense/librealsense/issues/6406 
For more info you can contact me directly via email: rrobert.kawecki@gmail.com (double R in the beginning)

Could you please help me to resolve this problem? I'm interested in buying few new RS cameras for my project but I need to know if your Android API is working as I would expect.

Kind regards,
Robert

MartyG-RealSense, 2020-05-22:
Hi @hunterman1   Although I have provided advice on your other case https://github.com/IntelRealSense/librealsense/issues/6406 I have assigned myself to this case too in order to provide a single point of contact for both cases and offer any new insights that I can provide as the subject is researched further.  

RealSense team members are very welcome to contribute advice on this case.

hunterman1, 2020-05-24:
Of course I've added some loggers, like textView.setText("Start test1 " + counter); to observe "where" is the application code while my device is connected to the camera. Unfortunately, without these loggers image view is also stuttering.

If you have some questions please ask me here or via email. I'll try to answer asap.

Kind regards,
Robert Kawecki

MartyG-RealSense, 2020-05-24:
@hunterman1   Thanks a lot.  I will try to make sure that your question gets seen by a Android wrapper developer.

hunterman1, 2020-05-24:
I also implemented an example attached below. Here camera is connects to my android device, sometimes not. Besides this I'm observing delay (more than 1s) and the image also stutters.

[Capture.zip](https://github.com/IntelRealSense/librealsense/files/4673612/Capture.zip)

hunterman1, 2020-05-24:
I also tried with the native version of mobile app. I implemented an example basic on the code available on github. When I used com.intel.realsense:librealsense:2.34.0@aar I could see from the printout on textview that a device is connected. But I still think that the problem would appear so I'm waiting for your response before I'll take time to implement something using native code.

Regards,
Robert

hunterman1, 2020-05-24:
This is an example of developed by my working native example app.
[Test_native3.zip](https://github.com/IntelRealSense/librealsense/files/4673634/Test_native3.zip)

MartyG-RealSense, 2020-06-05:
Hi @hunterman1   Do you require further assistance with this case please or is it okay to close it?   Thanks!

MartyG-RealSense, 2020-06-14:
Case closed due to no further comments received.

closed 2020-06-14 by MartyG-RealSense

hunterman1, 2020-08-02:
> Hi @hunterman1 Do you require further assistance with this case please or is it okay to close it? Thanks!

I didn't receive help.... Second ticket was closed, and this one also - without any useful comment... I managed to solve this problem, but the documentation, how to use your D435i camera within Android SDK is very poor. It's very hard to do simple things here...

/Robert

MartyG-RealSense, 2020-08-03:
I'm very pleased that you managed to find a solution.  

The RealSense team's ability to provide support is dependent on an active conversation being maintained, so it is important for RealSense users to post comments on their cases, even if it is just to say that their problem is not yet solved and they require further assistance.  Otherwise a case is likely to be considered dormant and closed (but still findable with web searches) if no further comments are received after a couple of weeks since the last communication.