realsenseai/librealsense #6443: Get pointcloud vertices in GPU memory?

tom-garford, 2020-05-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (2.34) | 
| Operating System & Version |   {Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson   |
| SDK Version                            |  { 2.34}                          | 
| Language                            |  {C++}                          | 

### Issue Description

Hi All,

How do I get a pointer to vertices from a rs2::points object computed on the GPU (similar to rs-gl example) without copying the rs2::points object back to the CPU (vertices remain in GPU memory). I am keen to use the vertices computed in the pointcloud in other cuda kernels. 

I know .get_vertices() will do this for me but copy back to the CPU. I managed to find references to <rs2::gl::gpu_points_frame> in src/gl for the pointcloud_points::process_frame source code but it refers to a librealsense::frame_interface type which I can't find a reference to anywhere. I imagine I need the _vertex_id to access correct texture but would be gladly corrected. 

Any pointers (no pun intended) or a simpler method would be much appreciated!

https://github.com/IntelRealSense/librealsense/blob/f361e56f8a2fb5093a1d663293421c92dd432bba/src/gl/pc-shader.cpp

MartyG-RealSense, 2020-05-22:
Hi @tom-garford   Are you currently making use of CUDA on your Nano to offload work from the GPU to the GPU?  As Nano is an Nvidia product, it supports using the Librealsense SDK with CUDA.

In regard to documentation about **librealsense::frame_interface**, have you already seen the information in the link below, please?

http://docs.ros.org/kinetic/api/librealsense2/html/classlibrealsense_1_1frame__interface.html

dorodnic, 2020-05-24:
If this data was calculated using `realsense2-gl` module, the frame should be convertible to `gl::frame`:
```
if (auto gl_frame = frame.as<rs2::gl::gpu_frame>()){
    auto tex_id = gl_frame.get_texture_id();
```
Then, look into converting OpenGL texture to CUDA object. I found several recommendations on Google, but not confident enough to recommend.

tom-garford, 2020-05-24:
Hi @dorodnic  and @MartyG-RealSense 

Thanks both for your replies. Interesting reading the frame_interface docs, although as @dorodnic  has pointed out, I think I was barking up the wrong tree! 

Yes, if I call the vertex_tex_id I should be able to get the correct texture and hopefully convert to a cuda object. I shall have a go and report back. 

Many thanks, especially as you should be enjoying your weekends!

tom-garford, 2020-05-28:
Hi @dorodnic @MartyG-RealSense 

I have had another go at this and make no apologies for going off the beaten track.

What I am aiming for is a pointer to the float3 XYZ pointcloud coordinates stored in the rs2::points object in GPU memory after being processed in the kernels in src/cuda/cuda-pointcloud.cu. (equivalent to calling .get_vertices() on the CPU) 

Reading cuda-pointcloud.cu, I am confused, as this shouldn't be possible as the processed points appear to be copied back to host and the memory freed (cudaFree). However, this would be illogical as examples like rs-gl, etc. wouldn't be possible (points are obviously not copied back to CPU every time with GPU processing). 

```
void rscuda::deproject_depth_cuda(float * points, const rs2_intrinsics & intrin, const uint16_t * depth, float depth_scale)
{
    int count = intrin.height * intrin.width;
    int numBlocks = count / RS2_CUDA_THREADS_PER_BLOCK;
    
    float *dev_points = 0;	
    uint16_t *dev_depth = 0;
    rs2_intrinsics* dev_intrin = 0;
    cudaError_t result;

    result = cudaMalloc(&dev_points, count * sizeof(float) * 3);
    assert(result == cudaSuccess);
    result = cudaMalloc(&dev_depth, count * sizeof(uint16_t));
    assert(result == cudaSuccess);
    result = cudaMalloc(&dev_intrin, sizeof(rs2_intrinsics));
    assert(result == cudaSuccess);
    
    result = cudaMemcpy(dev_depth, depth, count * sizeof(uint16_t), cudaMemcpyHostToDevice);
    assert(result == cudaSuccess); 
    result = cudaMemcpy(dev_intrin, &intrin, sizeof(rs2_intrinsics), cudaMemcpyHostToDevice);
    assert(result == cudaSuccess); 
    
    kernel_deproject_depth_cuda<<<numBlocks, RS2_CUDA_THREADS_PER_BLOCK>>>(dev_points, dev_intrin, dev_depth, depth_scale); 

    result = cudaMemcpy(points, dev_points, count * sizeof(float) * 3, cudaMemcpyDeviceToHost);
    assert(result == cudaSuccess);

    cudaFree(dev_points);
    cudaFree(dev_depth);
    cudaFree(dev_intrin);
}
```
From my understanding (and a little guess work, I would be pleased if you could fill me in) it seems that the processed point coordinates are still stored in an openGL texture on the GPU and accessed in /src/gl/pc_shader.cpp where the "processing_pointcloud" function is defined using ".get()" function inside a "perform_gl_action" callback.

```
 rs2::frame pointcloud_renderer::process_frame(const rs2::frame_source& src, const rs2::frame& f)
        {
            //scoped_timer t("pointcloud_renderer");
            if (auto points = f.as<rs2::points>())
            {
                perform_gl_action([&]()
                {
                    //scoped_timer t("pointcloud_renderer.gl");

                    GLint curr_tex;
                    glGetIntegerv(GL_TEXTURE_BINDING_2D, &curr_tex);

                    auto vf_profile = f.get_profile().as<video_stream_profile>();
                    int width = vf_profile.width();
                    int height = vf_profile.height();
                    
                    if (glsl_enabled())
                    {
                        if (_width != width || _height != height)
                        {
                            obj_mesh mesh = make_grid(width, height);
                            _model = vao::create(mesh);

                            _width = width;
                            _height = height;
                        }
[... 45 lines cut ...]
                            glActiveTexture(GL_TEXTURE0 + _shader->texture_slot());

                            _shader->end();
                        }
                    }
```
I have tried replicating this code inside my own script but seem to get into dependency problems after including src/gl/synthetic-stream-gl.h and others in order to use to define "perform_gl_action" and "glsl_enabled" functions.  Using the frame_interface* type I guess lets you cast to the gpu_addon type shown in http://docs.ros.org/kinetic/api/librealsense2/html/classlibrealsense_1_1frame__interface.html.  I was wondering if this could instead be converted to float3: 

```
            if (auto gpuPoints = pointcloud_points.as<rs2::points>())
            {
                perform_gl_action([&]()
                {
                    float3* pointcloud = (float3*)gpuPoints.get();
                });
            }
```

If I just use ".get()" on a rs2::points object, I am able to cast to float3* and access it, however I get segmentation faults trying to use in a cuda function (even after doing init_glsl_processing/ init_glsl_rendering) and an empty buffer on the CPU side (meaningless, unchanging values). 

Any advice would be much appreciated. My current feeling is to copy the kernel converting depth values to X,Y,Z coordinates and implement my own function without openGL textures involved. 

Kind regards, 

Tom.

tom-garford, 2020-05-29:
Ok, after further thought (and realising that rs2::frame is just a more general container for frame information compared to rs2::points) I have got a pointer to a openGL texture ID for my rs2::points object. As @dorodnic mentions earlier, I think I need to look into Cuda/ openGL interoperability described in the Cuda Runtime API and go from there.

MartyG-RealSense, 2020-05-29:
@tom-garford   Thanks very much for the update.   Good luck!

tom-garford, 2020-06-02:
More Progress. I have now used the Cuda-OpenGL interop functions in Cuda Runtime API to convert texture IDs to cuda device pointers, however, only seem able to uncover two textures 24 and 32 bytes in length. Printing them to the screen does not seem to give any useful information (" 0 1 2 2 3 0" and " 01 1 1 1 0 0 0"). I think one of them is a frame reference but can't be sure. 
```
            if (auto gpuPointcloud  = pointcloud_points.as<rs2::gl::gpu_frame>())
            {
                if (pointcloud_points) pc_renderer.process(pointcloud_points);
                points_id = gpuPointcloud.get_texture_id(0);
                if (initial == true)
                {
                    gpuErrchk(cudaGraphicsGLRegisterBuffer(&glPoints_cuda,points_id,cudaGraphicsRegisterFlagsNone));
                }
                gpuErrchk(cudaGraphicsMapResources(1,&glPoints_cuda,0));
                gpuErrchk(cudaGraphicsResourceGetMappedPointer((void **)&gpuPoints, &num_bytes, glPoints_cuda));
                std::cout << num_bytes << std::endl;
                gpuErrchk(cudaGraphicsUnmapResources(1, &glPoints_cuda, 0));
                cudaFree(gpuPoints);
                initial=false;
            }
```
I have also used the gl::pointcloud_renderer.process_frame() function but receive no change in available textures. Are there any Intel Engineers I could speak further about this? I can see references to vertex_texture vertex_texture_id as detailed above, so they must be present. 

Is there any more support available for commercial customers? 

@dorodnic @MartyG-RealSense  @ev-mp

MartyG-RealSense, 2020-06-02:
Hi @tom-garford   Intel RealSense Non disclosure Agreement (NDA) customers can access a customer support portal called Zendesk.  My understanding though is that the granting of Intel NDA status depends on meeting a 'Return on Investment' expectation (a benefit to Intel from a privileged information partnership).

An example of an **economic** type of Return on Investment might be an intention to purchase a volume of more than 100,000 units of camera.

closed 2020-06-08 by tom-garford