realsenseai/librealsense #6444: Incomplete video frame detected - D435i on ubuntu 16.04

FrankCreen, 2020-05-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |    Linux (Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-101-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.34 | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-05-22:
Hi @FrankCreen   

Could you tell us please what computer / computing device you are using (PC, Mac, Android, Raspberry Pi, etc).

Also, are you using the official short USB cable that was supplied with the camera or a USB cable of your own choice?

If you are able to use the RealSense Viewer program, would it be possible to tell us the 'ASIC Temperature' and 'Projector Temperature' values whilst the camera is streaming?  These can be found under the "Controls" section of the Viewer's options side-panel.

FrankCreen, 2020-05-24:
> Hi @FrankCreen
>
> Could you tell us please what computer / computing device you are using (PC, Mac, Android, Raspberry Pi, etc).
>
> Also, are you using the official short USB cable that was supplied [...]

Yes! Trun on the usb 3.0 port,problem solved!

MartyG-RealSense, 2020-05-24:
Can you confirm please if your problem has been solved?  Thanks!

FrankCreen, 2020-05-24:
> Can you confirm please if your problem has been solved? Thanks!

yes! the problem has been solved.

MartyG-RealSense, 2020-05-24:
Great news!  I will close this case then.   Thanks so much - I'm very pleased for your success   :)

closed 2020-05-24 by MartyG-RealSense

xiaoyuansheng, 2020-06-15:
hey 
I have the same question about "Incomplete video frame detected"
Can you tell me how to solved?

MartyG-RealSense, 2020-06-15:
Hi @xiaoyuansheng  Can you provide information about these questions please:

Could you tell us please what computer / computing device you are using (PC, Mac, Android, Raspberry Pi, etc).

Also, are you using the official short USB cable that was supplied with the camera or a USB cable of your own choice?

If you are able to use the RealSense Viewer program, would it be possible to tell us the 'ASIC Temperature' and 'Projector Temperature' values whilst the camera is streaming? These can be found under the "Controls" section of the Viewer's options side-panel.

xiaoyuansheng, 2020-06-15:
I use ubuntu via VMware on PC and using official short USB cable.
The 'ASIC Temperature is thirty-seven' and 'Projector Temperature is thirty-four'.

MartyG-RealSense, 2020-06-15:
The ASIC and Projector temperatures are within acceptable tolerances.

The librealsrnse documentation recommends VMWare Workstation Player in order for librealsense to work properly with USB.  Are you using this VMWare program please?

xiaoyuansheng, 2020-06-15:
ya I use VMware Workstation 15 Player

MartyG-RealSense, 2020-06-15:
The RealSense SDK Manager has said in the past about this error: "The partial frames issue usually indicate that the uvcvideo driver does not get enough USB buffers from the USB subsystem.  This can be either because the system is overloaded (possibly thermal-throttled), CPU utilisation is near 100%, or the USB controller is not handling the requests in time".

DukeCheung, 2021-04-17:
Hello, I have the same question about "Incomplete video frame detected" with D455. I use the Ubuntu 16.04 via VBox and the USB version is 3.2, camera's IMU component cannot work in the VM. Can you tell me how to solve it? 
Thank you!

MartyG-RealSense, 2021-04-17:
Hi @DukeCheung   As commented on your post on https://github.com/IntelRealSense/librealsense/issues/8851 - Intel recommend using **VMWare Workstation Player** instead of Oracle VirtualBox due to USB 3 compatibility.

Christie-zhz, 2021-06-12:
Hello, I met the same error "Incomplete video frame detected"
I use ubuntu via VMware  workstation on PC and using USB cable 3.0.
The 'ASIC Temperature is  forty-seven' and 'Projector Temperature is forty-three'.
And my system is not overloaded.
could you help me to deal  it? thank you very much

MartyG-RealSense, 2021-06-12:
Hi @Christie-zhz   Above 42 degrees, that is a temperature range where image problems might start to appear (with 60 degrees being a safety shut-off point for the projector's laser).  The internal camera temperature can be affected by the external air temperature of the location that the camera is being used in, or a technical issue such as a bad USB cable or a USB port glitch.

An easy test for whether a USB issue might be causing the camera to rapidly overheat is to enable the streams and then after a few minutes put your hand on the outer camera casing (if you are using a model with an outer casing such as D415, D435/i and D455) and see whether the casing is already hot to the touch.

If the location's air temperature is relatively cool then I would expect an ASIC temperature of late 20's to early 30's if one or two streams are enabled.

Christie-zhz, 2021-06-12:
Excluding the influence of the USB port on the equipment temperature, the outside air temperature at the location where the camera is used is really too high. After changing the environment, the' ASIC temperature is thirty-seven' and' projector temperature is thirty-four, but it is still the error "Incomplete video frame detected"

MartyG-RealSense, 2021-06-12:
I recalled a case in the recent past of **Incomplete video frame detected** when using VMWare Workstation Player with a D455 (which like the D435i has an IMU component).

https://github.com/IntelRealSense/librealsense/issues/8851

In that particular case, the chances of the error occurring were reduced by:

-  Using a lower stream resolution.
-  Only enabling two streams at a time instead of three (such as only enabling IMU and RGB instead of Depth, RGB and IMU).

How many streams do you have enabled at the same time please?

Christie-zhz, 2021-06-13:
Thank you for your answer, I started two streams at the same time, RGB and Depth, using a resolution of 640*480, do I need to upgrade the firmware?

MartyG-RealSense, 2021-06-13:
@Christie-zhz  Which firmware version is your camera currently using please?  And which librealsense SDK version is being used?

Christie-zhz, 2021-06-13:
Firmware Version              : 	05.12.14.50
 Recommended Firmware Version  : 	05.12.14.50
May no longer need to upgrade
librealsense  SDK 2.0
No error is reported when using the camera under Windows system

MartyG-RealSense, 2021-06-13:
Thanks very much for the information.  The details provided at the top of this case list that the librealsense SDK version being used is 2.34.  Is that still true, please?  If it is then firmware 5.12.14.50 would not be a suitable firmware to use.  If firmware 5.12.8.200 or newer is used then SDK 2.39.0 or newer should be used due to internal changes in the firmware.

Also, I would not recommend using SDK 2.34.0 because it had an issue for some users where it generated timing errors continuously.