realsenseai/librealsense #6445: Cannot start pipeline of T265 after using command context.query_all_sensors() or context.query_devices()

MoBaT, 2020-05-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (0.2.0.926) | 
| Operating System & Version |   { Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  ( 5.3.0-51-generic )                                         | 
| Platform                                 | x86_64 Desktop  |
| SDK Version                            |  { v2.34.0 }                          | 
| Language                            |  { C++ }                          | 

### Issue Description
When calling the function `rs2::context().query_devices()` or `rs2::context().query_sensors()`, I am unable to start the T265 pipeline.

```
rs2::context  ctx;
for (auto &&dev : ctx.query_devices())
        {
            rs2::pipeline pipe(ctx);
            rs2::config cfg;
            cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
            pipe.start(cfg);
        }
```

jfa1, 2020-05-25:
I too have the same problem running SDK v2.34.0 but on Windows 10 (firmware 0.2.0.926). An unhandled exception is thrown using the above code.

I came across this error first trying to run the `rs-multicam` example which uses the same method to start pipelines as above. I have been able to replicate this issue on two separate Windows computers. It is worth noting that realsense-viewer works fine, without any problems streaming from the T265.

Currently, the only way I can get the T265 to work is by enabling the streams desired without querying devices beforehand:

```
rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
cfg.enable_stream(RS2_STREAM_FISHEYE, 1);
cfg.enable_stream(RS2_STREAM_FISHEYE, 2);
pipe.start(cfg);
```

This is the same method as per `rs-ar-basic`. Next, I will try to find a previous SDK version which works with `ctx.query_devices()`, and if that does not work, reverting the T265 firmware to a previous version.

joshuafavetti, 2020-05-25:
After going back and compiling the previous releases of the SDK, the most recent version that still works with `rs-multicam` and `ctx.query_devices()` is v2.31.0.

jfa1, 2020-05-25:
After going back and compiling the previous releases of the SDK, the most recent release that still works with `rs-multicam` and `ctx.query_devices()` is v2.31.0. 

Each SDK was built using Windows 10 (1903), CMake GUI 3.17.0 and Visual Studio 2019. The CMake configuration was left as default EXCEPT for checking `ENFORCE_METADATA`.

MoBaT, 2020-06-09:
@ev-mp, Any update on this?

MoBaT, 2020-06-30:
@ev-mp ping on this issue.

dorodnic, 2020-07-09:
Last release included update to `rs-multicam` showing how to use `query_devices` together with `pipeline`

closed 2020-08-05 by dorodnic