realsenseai/librealsense #6446: T265 and L515 Rig

JBBee, 2020-05-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            |  2.34.0                          | 
| Language                            |  C#                          | 
| Segment			|  AR                         | 

### Issue Description
I would eventually like to mate a T265 and an L515 together on a mount/rig.  Are there any plans to design such a mount we can 3D print?

Thanks.

rafaelspring, 2020-05-25:
+1 Same here.

mpottinger, 2020-05-25:
Both cameras have the same type of mounting screw on the bottom, so it should fit anyway. Don't see why it wouldn't. However even if that works, you still need the information on the relative poses between the cameras, which would be different. It would be nice to have an out of the box solution. I wish they would just make a product line that has both tracking and depth integrated into the same device and is plug and play.

AndreV84, 2020-07-23:
+1 for 3d mount for t265 with L515

RealSenseSupport, 2021-03-18:
Thank you for highlighting the need for a mount that connects T265 with L515. We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265.

closed 2021-03-18 by RealSenseSupport