realsenseai/librealsense #6447: Unable to access frame and sensor timestamps

Inffzy99, 2020-05-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.0 | 
| Language                            | C++ | 
| Segment			| AR | 

### Issue Description
I am having trouble with getting Kinect Fusion to work with Nvidia GPU, so I want to check if the timestamps are correct. From the 4 types of time related RS2_FRAME_METADATA, I am able to get `RS2_FRAME_METADATA_TIME_OF_ARRIVAL` and `RS2_FRAME_METADATA_BACKEND_TIMESTAMP` but not `RS2_FRAME_METADATA_FRAME_TIMESTAMP` or `RS2_FRAME_METADATA_SENSOR_TIMESTAMP`. Also, `get_frame_timestamp_domain()` outputs System Time and the outputs from `get_timestamp()` are all the same. I have two D435 cameras and both are experiencing the same condition. I'm not sure what is wrong.

MojamojaK, 2020-05-23:
You have to enable metadata
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows

MojamojaK, 2020-05-23:
Have you enabled metadata?
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows

rafaelspring, 2020-05-23:
Turn off global time: `sensor.set_option(RS2_OPTION_GLOBAL_TIME_ENABLED, 0);`

closed 2020-07-09 by dorodnic