realsenseai/librealsense #6449: No module named pyrealsense2 when using python wrapper

miroslavradojevic, 2020-05-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   {Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  5.3.0-53 | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0 }  | 
| Language                            |  { python }  | 
| Segment			|  { Robot }   | 

### Issue Description
I am trying to work with Realsense D435i camera on Ubuntu 18.04, using python wrapper. Upon calling 
`import pyrealsense2 as rs `
in the script, I get error message: **import pyrealsense2 as rs**

I have installed SDK 2.0 following the [guidelines](https://www.intelrealsense.com/sdk-2/).
SDK is installed using packages (it was not built from source). 
Also I called 
`pip install pyrealsense2`
as stated in the [Installation](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python)

Can anyone provide any hints why the import is not working? I did scan previous similar issues on PC/RaspberryPIs and assumed it would not be necessary to build SDK from source. According to the documentation, calling `pip install pyrealsense2` should be sufficient.

MartyG-RealSense, 2020-05-23:
Hi @miroslavradojevic   I recently installed Pyrealsense2 on Windows and wrote my own manual for the process.  I have not done it on Linux but will try to provide some useful guidance.

Firstly, do you have the **pip** package installation system installed in Ubuntu please?

https://itsfoss.com/install-pip-ubuntu/

miroslavradojevic, 2020-05-23:
Yes, pip is installed on Ubuntu, it is the command I use very often. I have it paired with python3.
I switched to windows 10 where SDK is also installed and I get following after trying to insta pyrealsense2 package:
>pip install pyrealsense2
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2

As a reference,
>pip -V
pip 20.1.1 from c:\python38\lib\site-packages\pip (python 3.8)
since I got python installed in c:\python38

I could not get the python wrapper library on windows 10 either. I got SDK installed in 
C:\Program Files (x86)\Intel RealSense SDK 2.0 
I can see the libraries (pyrealsense2.pyd, pyrealsense.dll) in:
C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64
Should I maybe add \bin\x64\ folder to the PATH?

MartyG-RealSense, 2020-05-23:
PyPi is supported up to Python 3.7 currently.  Python 3.8 users should use 3.7 or compile from source.

https://github.com/IntelRealSense/librealsense/issues/5777#issuecomment-582480988

miroslavradojevic, 2020-05-23:
Thanks for the help, reverting Python to 3.7.6 worked. I will double check which version it was in Ubuntu, wrapper seems to be running in windows 10.

MartyG-RealSense, 2020-05-23:
I'm very glad you found a version that worked for you!

I'm off shift for the day now as it is almost midnight in my time zone.   If you need further assistance, I will pick up new comments when I return in 7 hours from the time of writing this and be happy to continue the support conversation.   Good luck!

miroslavradojevic, 2020-05-23:
No worries, it is not urgent, would be great to clarify what happens in linux, but this will do it for now.

MartyG-RealSense, 2020-05-24:
I'm back.  Thanks for your patience.

I checked various sources of information about the pip installation process for Ubuntu.  They recommended this instruction when installing on Python 3:

**pip3 install pyrealsense2**

miroslavradojevic, 2020-05-28:
Hello, I tried installing pyrealsense2 wrapper on Ubuntu 18.04 using command:
`pip3 install pyrealsense2`
Collecting pyrealsense2  Could not find a version that satisfies the requirement pyrealsense2 (from versions: )
No matching distribution found for pyrealsense2

Ubuntu 18.04 runs on Jetson Nano computer, and I did previously install SDK 2.0 (it is possible to call realsense-viewer). 
Both pip (for python 2 and 3 ) are installed:
```
pip -V
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)
pip3 -V
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.6)

```
What I also noticed is that when istalling SDK 2.0, command:
`sudo apt-get install librealsense2-dkms`
failed
**E: Unable to locate package librealsense2-dkms**

It is till not possible for me to get the wrapper running on Ubuntu - and I need it running there because of the hardware.

MartyG-RealSense, 2020-05-28:
On another case, involving an attempt to use pip on Raspberry Pi, the RealSense SDK Manager said that pip would not work on Pi because it has an Arm processor and there is not a suitable package for it.

https://github.com/IntelRealSense/librealsense/issues/6460#issuecomment-634103430

Are you trying to install on a Pi with pip, please?

miroslavradojevic, 2020-05-31:
I can confirm that the installation of the SDK 2.0 went fine on Ubuntu 18.04, and the wrapper is also installed and functional which should conclude this thread.

closed 2020-05-31 by miroslavradojevic

MartyG-RealSense, 2020-05-31:
@miroslavradojevic   Great to know that you had a successful install on Ubuntu.  Thanks for the update!

GOBish, 2020-06-28:
I am using an nvidia xavier nx, and all of the following fail:
pip install pyrealsense2
pip3 install pyrealsense2
```
Collecting pyrealsense2 Could not find a version that satisfies the requirement pyrealsense2 (from versions: )
No matching distribution found for pyrealsense2
```
and
sudo apt-get install librealsense2-dkms
`E: Unable to locate package librealsense2-dkms`

MartyG-RealSense, 2020-06-28:
@GOBish   pip install will not work with Jetson devices unfortunately as they use **Arm** processors, which are incompatible with the pyrealsense2 pip packages.    Jetson users should build the pyrealsense2 wrapper from source.

GOBish, 2020-06-28:
Understood, thanks for the reply. I built from the Linux installation page using sudo apt-get and I was under the impression that that process should set up the wrappers and not need to be built further. Am I wrong?

MartyG-RealSense, 2020-06-28:
The PyPi pip packages are separate from the main SDK files and are built for x86 (32-bit) and x64 (64-bit) processors.  Arm processors are not 'x' architecture.

If you are building Librealsense from source with CMake then you can build the Pyrealsense2 wrapper at the same time by including **-DBUILD_PYTHON_BINDINGS:bool=true** in your CMake build statement.

An Xavier user recently succeeded in building the Pyrealsense2 wrapper with this statement:

**-DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND=ON -DPYTHON_EXECUTABLE=...**

GOBish, 2020-06-28:
Thanks Marty.
I am also trying the flag --DBUILD_WITH_CUDA:bool=true but I get the error 

```
-- Info: Building with CUDA requires CMake v3.8+
-- The CUDA compiler identification is unknown
CMake Error at CMake/cuda_config.cmake:3 (project):
  No CMAKE_CUDA_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CUDACXX" or the CMake cache entry CMAKE_CUDA_COMPILER to the full
  path to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMake/global_config.cmake:65 (include)
  CMakeLists.txt:44 (global_set_flags)

```
cmake --version returns 3.10.12

GOBish, 2020-06-28:
I think CUDA error is found by adding this to bashrc:

```
# cuda 10.2
export CUDA_HOME=/usr/local/cuda
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda/lib64:/usr/local/cuda/extras/CUPTI/lib64
export PATH=$PATH:$CUDA_HOME/bin
```

MartyG-RealSense, 2020-06-28:
@GOBish   If you are using the latest 2.35.2 SDK, it has a bug in it where Librealsense does not build if CUDA is set to True when building for Jetson.  A fix for this is in the link below:

https://github.com/IntelRealSense/librealsense/issues/6573#issuecomment-643725732

A proper fix has been merged into the SDK development.

https://github.com/IntelRealSense/librealsense/pull/6581

GOBish, 2020-06-28:
OK, I am still trying, I get the following error now:

```
$ cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND=ON -DPYTHON_EXECUTABLE=... -DBUILD_WITH_CUDA:bool=true 
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.35.2
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- CUDA_LIBRARIES: /usr/local/cuda/include /usr/local/cuda/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/aarch64-linux-gnu/librt.so;/usr/local/cuda/lib64/libcusparse.so;/usr/lib/aarch64-linux-gnu/libcublas.so
-- using RS2_USE_LIBUVC_BACKEND
CMake Error at wrappers/python/third_party/pybind11/tools/FindPythonLibsNew.cmake:95 (message):
  Python config failure:

Call Stack (most recent call first):
  wrappers/python/third_party/pybind11/tools/pybind11Tools.cmake:16 (find_package)
  wrappers/python/third_party/pybind11/CMakeLists.txt:33 (include)

-- Configuring incomplete, errors occurred!
```

MartyG-RealSense, 2020-06-28:
Does it make any difference if you leave out **-DPYTHON_EXECUTABLE=...**

I have not personally seen this flag used in pyrealsense2 building with CMake before that particular Xavier case.

GOBish, 2020-06-28:
taking that line out or using  -DPYTHON_EXECUTABLE=/usr/bin/python makes the build complete successfully.

MartyG-RealSense, 2020-06-28:
@GOBish   Great news!

GOBish, 2020-06-28:
Thank you very much for the excellent and quick help!

GOBish, 2020-07-14:
Just as an update, with the new version of the SDK 2.36.0 (released July 9,2020), the PYTHONPATH must now be
`export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python2.7/pyrealsense2`

GOBish, 2020-07-14:
Also, I have never been able to get the library to work with python3. The files never get created in the python3 folder even if I specify  -DPYTHON_EXECUTABLE=/usr/bin/python3  (which is what "which python3" points to). Using  -DPYTHON_EXECUTABLE=/usr/bin/python always creates the python libraries in the correct python 2 folder (which I point to in the post above is now different)
Has anyone been able to get it to work with python 3?

MartyG-RealSense, 2020-07-15:
@GOBish  The Python wrapper definitely can work with Python 3.  The wrapper's documentation says that **-DPYTHON_EXECUTABLE** is used if you have both Python 2 and 3 installed on the same computer.  To quote:

**Note: To force compilation with a specific version on a system with both Python 2 and Python 3 installed, add the following flag to CMake command: -DPYTHON_EXECUTABLE=[full path to the exact python executable]**

One RealSense user's example of successfully specifying a full path to Python 3 on their particular machine was:

**-DPYTHON_EXECUTABLE= /usr/bin/python3.5**

Can you check what the actual folder name of your Python installation is at the usr/bin location please?

GOBish, 2020-07-15:
Thanks as always for the quick responses, @MartyG-RealSense .
Nope, I tried -DPYTHON_EXECUTABLE= /usr/bin/python3 and -DPYTHON_EXECUTABLE= /usr/bin/python3.6    (my python version), and the packages still do not get installed. 
There is no pyrealsense2 folder created  in /usr/local/lib/python3.6  created like there was in /usr/local/lib/python2.7 when I installed last.
SDK 2.36.0 installs fine in the python2.7 folder when -DPYTHON_EXECUTABLE is left out.

MartyG-RealSense, 2020-07-15:
I read back through your part of this discussion.  To summarise, my understanding is that:

- You are using Nvidia Jetson Xavier NX, and so installing pyrealsense2 with pip is not an option due to the Arm processor incompatibility with the pyrealsense2 pip packages.
-  You were able to build pyrealsense2 successfully from source for Python 2.7 by leaving out the **DPYTHON_EXECUTABLE** instruction from the CMake statement or using -DPYTHON_EXECUTABLE=/usr/bin/python
- But you cannot get CMake to build for Python 3.6 when using DPYTHON_EXECUTABLE

Is that accurate, please?

GOBish, 2020-07-15:
The build always does complete successfully whether I use -DPYTHON_EXECUTABLE= /usr/bin/python , -DPYTHON_EXECUTABLE= /usr/bin/python3 , -DPYTHON_EXECUTABLE= /usr/bin/python3.6 or just leave out -DPYTHON_EXECUTABLE totally.

But the pyrealsense2 files always get put only in the python2.7 folder, and I have not been able to successfully import the library into python3.

MartyG-RealSense, 2020-07-15:
Assuming that you have already successfully built Librealsense, the CMake statement only needs to include the Python wrapper references.  Can you confirm please that the CMake statement you are using does not include RSUSB, and just looks like this:

cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE= /usr/bin/python3.6

GOBish, 2020-07-15:
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND=ON -DPYTHON_EXECUTABLE=/usr/bin/python3 -DBUILD_WITH_CUDA:bool=true

GOBish, 2020-07-15:
I just tried building leaving out the other flags, no change

MartyG-RealSense, 2020-07-15:
Ok, now you have just done that build instruction, what happens if you use the next two commands in the instructions for building the wrapper from source:

**make -j4
sudo make install**

GOBish, 2020-07-15:
yes, always completed the whole process, no change.

MartyG-RealSense, 2020-07-15:
Apologies for asking this, but are you certain that pyrealsense2 has not been built in the Python 3.6 folder.  It would naturally already be in the Python 2.7 folder because you have already built pyrealsense2 for that Python version.

GOBish, 2020-07-15:
yes, I check the folder every time. No apologies needed, I have for sure overlooked the basics plenty of times in the past.

MartyG-RealSense, 2020-07-15:
I have been through the case thoroughly and cannot see anything obviously wrong with how you have done things.

As suggested by the wrapper instructions, did you install Python 3 using this method:

**sudo apt-get install python3-dev**

GOBish, 2020-07-15:
python2 and python3 is installed with the disk image for the xavier jetpack 4.4

MartyG-RealSense, 2020-07-15:
Yes, I thought that it would be.  I just wondered whether there was something in the pre-installed version of Python 3 that might be incompatible, compared with the one installed with **sudo apt-get install python3-dev**

GOBish, 2020-07-15:
"python3-dev is already the newest version"

MartyG-RealSense, 2020-07-15:
I think every possibility has already been covered for getting Python 3 pyrealsense2 working using building from source.  I do not know of anything else to try, unfortunately.

GOBish, 2020-07-15:
ok, thanks for trying. I'll keep at it and update with any results I can achieve.

MartyG-RealSense, 2020-07-15:
I appreciate your patience.   I'm sorry I couldn't be of more help.

GOBish, 2020-07-16:
I tried "make clean" and then did the full build again using DPYTHON_EXECUTABLE=/usr/bin/python3.6, no change

GOBish, 2020-07-26:
Can you think of any way to change the config? When I do sudo make install after specifying the python3.6 folder, the output still shows:

```
Install the project...
.
.
.
-- Up-to-date: /usr/local/lib/python2.7/pyrealsense2/pybackend2.so.2.36.0
-- Up-to-date: /usr/local/lib/python2.7/pyrealsense2/pybackend2.so.2
-- Up-to-date: /usr/local/lib/python2.7/pyrealsense2/pybackend2.so
-- Up-to-date: /usr/local/lib/python2.7/pyrealsense2/pyrealsense2.so.2.36.0
-- Up-to-date: /usr/local/lib/python2.7/pyrealsense2/pyrealsense2.so.2.36
-- Up-to-date: /usr/local/lib/python2.7/pyrealsense2/pyrealsense2.so
 
```

MartyG-RealSense, 2020-07-26:
I was working on another Pyrealsense2 installation case today that was successful in the end.  The RealSense user in that case posted their own method that worked for them.

https://github.com/IntelRealSense/librealsense/issues/6922#issuecomment-663976346

GOBish, 2020-07-26:
Thanks, Marty - I saw that thread and I think that case seems different than mine.

I looked at the cmake_install.cmake file in the librealsense/build/wrappers/python folder and all of the install directories there are set to /lib/python2.7/pyrealsense....

I tried manually changing those to python3.6, and while it did change the install location of the libraries to the right place, python3 could still not import pyrealsense2, still giving the error undefined symbol _Py_ZeroStruct

MartyG-RealSense, 2020-07-26:
@GOBish  I am not certain why you are having so much difficulty with the install path on your Xavier NX when installing the wrapper from source.  

You have probably seen the Xavier pyrealsense2 case below, but I will link to it just in case you haven't.

https://github.com/IntelRealSense/librealsense/issues/2431#issuecomment-432118867

GOBish, 2020-07-26:
yeah, super frustrating because I have some other programs that only run with python 3 and I need to integrate realsense into those projects.

MartyG-RealSense, 2020-07-26:
Is there any possibility of uninstalling Python 2.7 and so preventing Pyrealsense2 from being installed to it, leaving Python 3 as the only choice?

GOBish, 2020-07-26:
just removed all python2 and dependencies and it still installed it into a python2.7 folder

MartyG-RealSense, 2020-07-26:
At this point it may be best to start a new case of your own and describe your problem so that someone else can look at it with a fresh perspective.  I can only think that there is something already installed on the NX that is causing a conflict with installation, since pyrealsense2 is only supposed to use Python 2 if it is not possible to use Python 3.

GOBish, 2020-07-31:
Hi @MartyG-RealSense - I was finally able to solve this. See issue https://github.com/IntelRealSense/librealsense/issues/6980 .

MartyG-RealSense, 2020-07-31:
@GOBish   Awesome, thanks for the update on your success!