realsenseai/librealsense #6451: How can I Remove sdk 2.34 ?

Eslamrady42, 2020-05-24:
Kindly please I want to remove all SDK 2.34 with everything with it , then reinstall SDK 2.33 .

MartyG-RealSense, 2020-05-24:
If you installed using a Debian distribution then you can remove all Librealsense related packages with this command:

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

A RealSense support team member from Intel Support (not the GitHub RealSense team) said that the folders remain though after the packages are removed with the command above.  They thought that starting with a fresh install was preferable if possible.
********************
If you installed 2.34.0 by compiling from source on Linux ... removing a previous installation and installing a new one is not honestly something I have done myself yet (I am primarily a Windows user).  I would think (not certain) that following the standard process for compiling the SDK from source, with a small change to the instructions, may accomplish what you want to do.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

The part that I would change is that instead of downloading the zip file of the latest stable version from the master branch (master.zip), I would instead download the source-code zip file from the list of files in the Assets list at the base of the release notes for 2.33.1.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1

[image]

I would then speculate (again, not with certainty) that the command later in the instructions for installing Librealsense from source would remove the existing 2.34.0 version and install 2.33.1 using the unzipped source code for it that was downloaded .

[image]

MartyG-RealSense, 2020-06-05:
Hi @Eslamrady42   Do you require further assistance with this case please or can it be closed?   Thanks!

Eslamrady42, 2020-06-05:
I’m sorry to answering you late, because I didn’t have enough time to work on this but I’ll return back to work tomorrow and I’ll continue the case with you .
Thank you so much for your support.

MartyG-RealSense, 2020-06-05:
Thanks very much for the update!   I will be happy to resume the case with you when you are ready.  Good luck!

Eslamrady42, 2020-06-11:
Hi @MartyG-RealSense I'm back and I'll start at the steps you said but I see it's the same as I used before !!

MartyG-RealSense, 2020-06-11:
Welcome back.  I read the case again and see that there is no information about the installation method that you use.  Do you install using source code or with a Debian package, please?

Eslamrady42, 2020-06-11:
I installed it as 2.352 IDK why !!

MartyG-RealSense, 2020-06-11:
Okay, that is fine.  Are you using Windows or Linux?

Eslamrady42, 2020-06-11:
Linux Ubuntu18

MartyG-RealSense, 2020-06-11:
Ok.  And have you tried using the camera with any programs such as the RealSense Viewer now that you have installed 2.35.2 (which is the newest SDK version that was released yesterday)?

Eslamrady42, 2020-06-11:
Yes, it's 2.35.2 and I used it on realsense-viewer and worked well, but now I'll try it with ROS.

MartyG-RealSense, 2020-06-11:
Okay, I look forward to hearing about your ROS results.

Eslamrady42, 2020-06-11:
I tried and this the result, I ignored the error and continued with Rviz but I can't see any results on it after I added all parameters.
BTW, I see the same if try one camera with it's commend or both cameras.
 

eslam@root:~$ roslaunch realsense2_camera rs_d400_and_t265.launch 
... logging to /home/eslam/.ros/log/564f5db0-ac18-11ea-a1e1-64006afb8a06/roslaunch-root-25374.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://root:35401/

SUMMARY
========

PARAMETERS
 * /d400/realsense2_camera/accel_fps: 0
 * /d400/realsense2_camera/accel_frame_id: d400_accel_frame
 * /d400/realsense2_camera/accel_optical_frame_id: d400_accel_optica...
 * /d400/realsense2_camera/align_depth: True
 * /d400/realsense2_camera/aligned_depth_to_color_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/allow_no_texture_points: False
 * /d400/realsense2_camera/base_frame_id: d400_link
 * /d400/realsense2_camera/calib_odom_file: 
 * /d400/realsense2_camera/clip_distance: -2.0
 * /d400/realsense2_camera/color_fps: 30
 * /d400/realsense2_camera/color_frame_id: d400_color_frame
 * /d400/realsense2_camera/color_height: 480
 * /d400/realsense2_camera/color_optical_frame_id: d400_color_optica...
 * /d400/realsense2_camera/color_width: 640
 * /d400/realsense2_camera/depth_fps: 30
 * /d400/realsense2_camera/depth_frame_id: d400_depth_frame
 * /d400/realsense2_camera/depth_height: 480
 * /d400/realsense2_camera/depth_optical_frame_id: d400_depth_optica...
 * /d400/realsense2_camera/depth_width: 640
 * /d400/realsense2_camera/device_type: d4.5
 * /d400/realsense2_camera/enable_accel: True
 * /d400/realsense2_camera/enable_color: True
 * /d400/realsense2_camera/enable_depth: True
 * /d400/realsense2_camera/enable_fisheye1: True
 * /d400/realsense2_camera/enable_fisheye2: True
 * /d400/realsense2_camera/enable_fisheye: True
 * /d400/realsense2_camera/enable_gyro: True
 * /d400/realsense2_camera/enable_infra1: True
 * /d400/realsense2_camera/enable_infra2: True
 * /d400/realsense2_camera/enable_pointcloud: False
 * /d400/realsense2_camera/enable_sync: False
 * /d400/realsense2_camera/filters: pointcloud
 * /d400/realsense2_camera/fisheye1_frame_id: d400_fisheye1_frame
 * /d400/realsense2_camera/fisheye1_optical_frame_id: d400_fisheye1_opt...
 * /d400/realsense2_camera/fisheye2_frame_id: d400_fisheye2_frame
 * /d400/realsense2_camera/fisheye2_optical_frame_id: d400_fisheye2_opt...
 * /d400/realsense2_camera/fisheye_fps: 30
 * /d400/realsense2_camera/fisheye_frame_id: d400_fisheye_frame
 * /d400/realsense2_camera/fisheye_height: 0
 * /d400/realsense2_camera/fisheye_optical_frame_id: d400_fisheye_opti...
 * /d400/realsense2_camera/fisheye_width: 0
 * /d400/realsense2_camera/gyro_fps: 0
 * /d400/realsense2_camera/gyro_frame_id: d400_gyro_frame
 * /d400/realsense2_camera/gyro_optical_frame_id: d400_gyro_optical...
 * /d400/realsense2_camera/imu_optical_frame_id: d400_imu_optical_...
 * /d400/realsense2_camera/infra1_frame_id: d400_infra1_frame
 * /d400/realsense2_camera/infra1_optical_frame_id: d400_infra1_optic...
 * /d400/realsense2_camera/infra2_frame_id: d400_infra2_frame
 * /d400/realsense2_camera/infra2_optical_frame_id: d400_infra2_optic...
[... 189 lines cut ...]
[ INFO] [1591903162.402755616]: HectorSM p_map_update_distance_threshold_: 0.400000 
[ INFO] [1591903162.402774775]: HectorSM p_map_update_angle_threshold_: 0.060000
[ INFO] [1591903162.402791203]: HectorSM p_laser_z_min_value_: -1.000000
[ INFO] [1591903162.402804054]: HectorSM p_laser_z_max_value_: 1.000000
^C[hector_geotiff_node-13] killing on exit
[hector_trajectory_server-12] killing on exit
[rviz-11] killing on exit
[hector_mapping-10] killing on exit
[base_link_to_d400_and_t265-9] killing on exit
[odom_to_base_link-8] killing on exit
[map_to_odom-7] killing on exit
[t265_to_d400-6] killing on exit
[d400/realsense2_camera_manager-4] killing on exit
[t265/realsense2_camera_manager-2] killing on exit
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 122 in /tmp/binarydeb/ros-melodic-class-loader-0.4.1/src/class_loader.cpp
[hector_trajectory_server-12] escalating to SIGTERM
[rviz-11] escalating to SIGTERM
[hector_mapping-10] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
eslam@root:~$

MartyG-RealSense, 2020-06-11:
If you are not using ROS wrapper version 2.2.13 (the most recent available right now), please try using it with SDK 2.35.2.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.13

Eslamrady42, 2020-06-11:
I did it now and I have another error 
[image]

eslam@root:~/catkin_ws$ roslaunch realsense2_camera rs_d400_and_t265.launch 
... logging to /home/eslam/.ros/log/027a1e10-ac1c-11ea-a1e1-64006afb8a06/roslaunch-root-28853.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://root:44449/

SUMMARY
========

PARAMETERS
 * /d400/realsense2_camera/accel_fps: 0
 * /d400/realsense2_camera/accel_frame_id: d400_accel_frame
 * /d400/realsense2_camera/accel_optical_frame_id: d400_accel_optica...
 * /d400/realsense2_camera/align_depth: True
 * /d400/realsense2_camera/aligned_depth_to_color_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/allow_no_texture_points: False
 * /d400/realsense2_camera/base_frame_id: d400_link
 * /d400/realsense2_camera/calib_odom_file: 
 * /d400/realsense2_camera/clip_distance: -2.0
 * /d400/realsense2_camera/color_fps: 30
 * /d400/realsense2_camera/color_frame_id: d400_color_frame
 * /d400/realsense2_camera/color_height: 480
 * /d400/realsense2_camera/color_optical_frame_id: d400_color_optica...
 * /d400/realsense2_camera/color_width: 640
 * /d400/realsense2_camera/depth_fps: 30
 * /d400/realsense2_camera/depth_frame_id: d400_depth_frame
 * /d400/realsense2_camera/depth_height: 480
 * /d400/realsense2_camera/depth_optical_frame_id: d400_depth_optica...
 * /d400/realsense2_camera/depth_width: 640
 * /d400/realsense2_camera/device_type: d4.5
 * /d400/realsense2_camera/enable_accel: True
 * /d400/realsense2_camera/enable_color: True
 * /d400/realsense2_camera/enable_depth: True
 * /d400/realsense2_camera/enable_fisheye1: True
 * /d400/realsense2_camera/enable_fisheye2: True
 * /d400/realsense2_camera/enable_fisheye: True
 * /d400/realsense2_camera/enable_gyro: True
 * /d400/realsense2_camera/enable_infra1: True
 * /d400/realsense2_camera/enable_infra2: True
 * /d400/realsense2_camera/enable_pointcloud: False
 * /d400/realsense2_camera/enable_sync: False
 * /d400/realsense2_camera/filters: pointcloud
 * /d400/realsense2_camera/fisheye1_frame_id: d400_fisheye1_frame
 * /d400/realsense2_camera/fisheye1_optical_frame_id: d400_fisheye1_opt...
 * /d400/realsense2_camera/fisheye2_frame_id: d400_fisheye2_frame
 * /d400/realsense2_camera/fisheye2_optical_frame_id: d400_fisheye2_opt...
 * /d400/realsense2_camera/fisheye_fps: 30
 * /d400/realsense2_camera/fisheye_frame_id: d400_fisheye_frame
 * /d400/realsense2_camera/fisheye_height: 0
 * /d400/realsense2_camera/fisheye_optical_frame_id: d400_fisheye_opti...
 * /d400/realsense2_camera/fisheye_width: 0
 * /d400/realsense2_camera/gyro_fps: 0
 * /d400/realsense2_camera/gyro_frame_id: d400_gyro_frame
 * /d400/realsense2_camera/gyro_optical_frame_id: d400_gyro_optical...
 * /d400/realsense2_camera/imu_optical_frame_id: d400_imu_optical_...
 * /d400/realsense2_camera/infra1_frame_id: d400_infra1_frame
 * /d400/realsense2_camera/infra1_optical_frame_id: d400_infra1_optic...
 * /d400/realsense2_camera/infra2_frame_id: d400_infra2_frame
 * /d400/realsense2_camera/infra2_optical_frame_id: d400_infra2_optic...
 * /d400/realsense2_camera/infra_fps: 30
[... 202 lines cut ...]
[ INFO] [1591904719.851098956]: Expected frequency for infra2 = 30.00000
[ INFO] [1591904719.867589837]: Expected frequency for color = 30.00000
[ INFO] [1591904719.883186334]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1591904719.901422281]: setupStreams...
[ INFO] [1591904719.902003453]: insert Depth to Stereo Module
[ INFO] [1591904719.902078289]: insert Color to RGB Camera
[ INFO] [1591904719.902115035]: insert Infrared to Stereo Module
[ INFO] [1591904719.902144832]: insert Infrared to Stereo Module
[ INFO] [1591904719.927236083]: SELECTED BASE:Depth, 0
[ INFO] [1591904719.959465387]: RealSense Node Is Up!
[d400/realsense2_camera_manager-4] process has died [pid 28883, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/eslam/.ros/log/027a1e10-ac1c-11ea-a1e1-64006afb8a06/d400-realsense2_camera_manager-4.log].
log file: /home/eslam/.ros/log/027a1e10-ac1c-11ea-a1e1-64006afb8a06/d400-realsense2_camera_manager-4*.log
[d400/realsense2_camera-5] process has finished cleanly
log file: /home/eslam/.ros/log/027a1e10-ac1c-11ea-a1e1-64006afb8a06/d400-realsense2_camera-5*.log
[t265/realsense2_camera-3] process has finished cleanly
log file: /home/eslam/.ros/log/027a1e10-ac1c-11ea-a1e1-64006afb8a06/t265-realsense2_camera-3*.log
^C[t265_to_d400-6] killing on exit
[t265/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[rosout-1] escalating to SIGTERM
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
eslam@root:~/catkin_ws$

MartyG-RealSense, 2020-06-11:
What happens if you only have the D435 plugged in and use a single-camera launch instruction like **roslaunch realsense2_camera rs_camera.launch**

Eslamrady42, 2020-06-11:
this error :
[image]

MartyG-RealSense, 2020-06-11:
I have not experienced the ROS error **corrupted size vs prev_size** before.  Research suggests that it is related to memory corruption.

Does the error occur if you run the launch instruction again a second time after the launch crashes?

Eslamrady42, 2020-06-11:
It’s didn’t crash but I can’t access any thing from the parameters of rviz or the camera itself.

MartyG-RealSense, 2020-06-11:
You would not get any IMU topics from a D435 as it does not have an IMU.  Can you receive anything in RViz from **camera/infra1/image_rect_raw**

Eslamrady42, 2020-06-11:
All I can see **noimages**

MartyG-RealSense, 2020-06-11:
Please see if the link below provided useful guidance for getting output from RViz:

https://github.com/IntelRealSense/realsense-ros/issues/302#issuecomment-362202818

Eslamrady42, 2020-06-11:
I did it in every time I used rviz.

MartyG-RealSense, 2020-06-11:
I am not in work time now as it is night time where I am.  I will come back to your case when my next work day starts in 9 hours from now.  I apologise for the wait until then and hope you can make progress.

Eslamrady42, 2020-06-11:
No problem I’ll wait for you tomorrow with another errors ^^
Thank you

Eslamrady42, 2020-06-12:
I tried all of the launch files and I see the same error as this : https://github.com/IntelRealSense/librealsense/issues/6451#issuecomment-642906573 
so I think the problem is from your side not from ROS.

MartyG-RealSense, 2020-06-12:
I recalled that there was a case 9 days ago about problems regarding using ROS with SDK 2.35.0.  @doronhi the ROS wrapper developer has seen it.

In that case, the RealSense user got a working SDK and ROS wrapper combination by using SDK 2.33.1 with firmware 12.3.0.  Presumably, the ROS wrapper version is 2.2.13, since that is one that is compatible with SDK 3.33.1.

https://github.com/IntelRealSense/realsense-ros/issues/1191  

If ROS is critical to your work and only being able to use Librealsense is not sufficient, then the options available are: 

(1)  To wait until there is an updated ROS wrapper that is matched with SDK 2.35.0; or 
(2) Start a completely fresh installation of Ubuntu 18, with no previous Librealsense / ROS wrapper installs, and install SDK 2.33.1, ROS wrapper 2.2.13 and camera firmware 5.12.3.

Eslamrady42, 2020-06-12:
First of all thank you for your efforts, I deleted the SDK with the libraries 3 times until now, then I installed SDK 2.33.1 in each time but what happened is the same result all I can see is the newest SDK version!!!, all I need now from you to tell me how can I remove everything for the fourth time and reinstall SDK 2.33.1. Also I Don’t know how can you release a new version of the SDK and the previous one have errors !!!!

MartyG-RealSense, 2020-06-12:
The Librealsense SDK and the ROS wrapper are not developed in sync, and so updates to the ROS wrapper may follow behind the schedule of SDK updates.  

The ideal situation is to have a specific ROS wrapper version to match with a specific SDK version (e.g SDK 2.33.1 / ROS wrapper 2.2.13).  It was found though that ROS wrapper 2.2.13 also worked with SDK 2.34.0, so there was not a need for a dedicated wrapper version for 2.34.0 in that case.

It is possible that as the latest SDK versions move further away from 2.33.1, a new ROS wrapper version may become needed to take account of changes made in the recent Librealsense SDKs (of which there were a lot of changes in SDK 2.35.0).  I am on the Support team though and not involved in Development, so I cannot comment on release plans.

As I said earlier, the best way to ensure a clean build as a stable base for a new SDK / wrapper installation would be to wipe Ubuntu 18 and reinstall it from the beginning, so that there is no hidden left-overs from previous installations that could cause conflicts with new installations.  I appreciate that this may not be a good option if you have a lot of stuff on the computer that you do not want to lose.

Since you were able to install SDK 2.35.2, I am guessing that you built the SDK from source code and not a Debian package.

Let's try and clean up your previous Librealsense installations by going to the build folder then and running this command from within it:

**sudo make uninstall && make clean**

Once the build folder is clean, build SDK 2.33.1 using the "backend" method, which requires an internet connection but has no patching and is not dependent on Linux versions or kernel versions.
Instructions for installing 2.33.1 with this method are in the link below:

https://github.com/IntelRealSense/realsense-ros/issues/1220#issuecomment-636814580

MartyG-RealSense, 2020-06-14:
Doronhi the RealSense ROS developer has mentioned that a new **realsense2_camera** version is arriving in the coming week to match with SDK 2.35.2.

https://github.com/IntelRealSense/realsense-ros/issues/1246#issuecomment-643814516

Eslamrady42, 2020-06-15:
I'm glad to hear that https://github.com/IntelRealSense/librealsense/issues/6451#issuecomment-643821532 
I've reached SDK 2.33.1 finally then I tried commends to deal with ROS and the camera :
**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

**rosrun tf static_transform_publisher 0 0 0 0 0 0 0 map camera_depth_optical_frame 100**

**rosrun rviz rviz**

Add -> by topic -> depth -> /color -> /points -> pointCloud2 -> ok
   
Add -> by topic -> (camera: depth, color or infra) -> /image_raw -> Image -> ok

this the steps I used but unfortunately not working with new errors.

the source that I used : 
https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

and this one to download the latest version of **realsense-ros-development** :   
https://github.com/IntelRealSense/realsense-ros

and this the error I found : 

eslam@root:~$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/eslam/.ros/log/cbdf6f3e-aed9-11ea-a6d8-64006afb8a06/roslaunch-root-5054.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://root:46749/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
[... 108 lines cut ...]
 15/06 09:29:01,389 WARNING [140333187290880] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 15/06 09:29:01,455 WARNING [140333187290880] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 15/06 09:29:01,477 ERROR [140332364764928] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 15/06 09:29:03,551 WARNING [140333212468992] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 11, 
[ WARN] [1592206144.325190189]: Hardware Notification:USB SCP overflow,1.59221e+12,Error,Hardware Error
 15/06 09:29:04,834 WARNING [140332382787328] (sync.cpp:171) Device destroyed
 15/06 09:29:04,834 WARNING [140332382787328] (sync.cpp:171) Device destroyed
 15/06 09:29:04,834 WARNING [140332382787328] (sync.cpp:171) Device destroyed
 15/06 09:29:04,834 WARNING [140332382787328] (sync.cpp:171) Device destroyed
 15/06 09:29:04,835 WARNING [140332382787328] (sync.cpp:171) Device destroyed
 15/06 09:29:04,835 WARNING [140332382787328] (sync.cpp:171) Device destroyed
 15/06 09:29:04,835 WARNING [140332382787328] (sync.cpp:171) Device destroyed
 15/06 09:29:04,836 WARNING [140332382787328] (sync.cpp:171) Device destroyed
 15/06 09:29:04,836 WARNING [140332382787328] (sync.cpp:171) Device destroyed
 15/06 09:29:04,836 WARNING [140332382787328] (sync.cpp:171) Device destroyed
[camera/realsense2_camera_manager-2] process has died [pid 5083, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/eslam/.ros/log/cbdf6f3e-aed9-11ea-a6d8-64006afb8a06/camera-realsense2_camera_manager-2.log].
log file: /home/eslam/.ros/log/cbdf6f3e-aed9-11ea-a6d8-64006afb8a06/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/eslam/.ros/log/cbdf6f3e-aed9-11ea-a6d8-64006afb8a06/camera-realsense2_camera-3*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
eslam@root:~$

MartyG-RealSense, 2020-06-15:
When debugging, I focus on the first error that occurs in the log.  This is because once one error occurs, it can create a "cascade" of errors following afterwards that make it appear as though there are multiple problems.  When in fact, if you fix that first error then the others can just disappear.

Continuously generating **control_transfer returned** errors (the first error that appears in the log) suggests that there are serious communication problems between the camera and the USB port.  

Do you have a different USB port that you can try the camera in, please?

Also, are you using the official short USB cable that came with the camera or are you using your own longer cable?

Eslamrady42, 2020-06-15:
I tried **USB-3.2**  **USB-3.1** **USB-2**
Yes, the official short USB cable that came with the camera.

Eslamrady42, 2020-06-15:
eslam@root:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/eslam/.ros/log/75dda930-af32-11ea-9f8d-64006afb8a06/roslaunch-root-13854.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://root:34247/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
[... 67 lines cut ...]
[ INFO] [1592244221.351604798]: Done Setting Dynamic reconfig parameters.
[ INFO] [1592244221.393245314]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1592244221.393493866]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1592244221.393687449]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1592244221.419633691]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1592244221.419777742]: setupPublishers...
[ INFO] [1592244221.422208733]: Expected frequency for depth = 30.00000
[ INFO] [1592244221.449691789]: Expected frequency for infra1 = 30.00000
[ INFO] [1592244221.464875649]: Expected frequency for infra2 = 30.00000
[ INFO] [1592244221.484215625]: Expected frequency for color = 30.00000
[ INFO] [1592244221.503536510]: setupStreams...
[ INFO] [1592244221.595397598]: insert Depth to Stereo Module
[ INFO] [1592244221.595531446]: insert Color to RGB Camera
[ INFO] [1592244221.595562209]: insert Infrared to Stereo Module
[ INFO] [1592244221.595585664]: insert Infrared to Stereo Module
[ INFO] [1592244221.846425396]: SELECTED BASE:Depth, 0
[ INFO] [1592244221.896421181]: RealSense Node Is Up!
 15/06 20:03:41,961 WARNING [140655993014016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 15/06 20:03:42,022 WARNING [140655993014016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
[camera/realsense2_camera_manager-2] process has died [pid 13882, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/eslam/.ros/log/75dda930-af32-11ea-9f8d-64006afb8a06/camera-realsense2_camera_manager-2.log].
log file: /home/eslam/.ros/log/75dda930-af32-11ea-9f8d-64006afb8a06/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/eslam/.ros/log/75dda930-af32-11ea-9f8d-64006afb8a06/camera-realsense2_camera-3*.log

MartyG-RealSense, 2020-06-15:
Do you have the option of using a mains electricity powered USB hub with the camera instead of plugging the camera directly into the computer or into a hub not powered by a mains electricity wall socket?

RealSense cameras are very sensitive to the state of the USB port.  If the problem is related to the USB ports of the computer not being supplied with power stably enough for good communication with the camera then a mains powered USB hub may help to make the control_transfer returned errors disappear.

MartyG-RealSense, 2020-06-15:
USB ports that are built into the computer draw their power from the computer's power supply components and so are vulnerable to any instabilities in that supplied power.  USB ports that are supplied power by the mains electricity socket have greater stability of supplied power.

Eslamrady42, 2020-06-15:
So what is the solution in this case ?

MartyG-RealSense, 2020-06-16:
Intel have successfully tested the AmazonBasics mains powered USB 3 hub with the 400 Series cameras.  I have one myself and have no problem with it.    They are available with full size USB-A connectors and micro USB-C connectors for connecting the hub to the computer, so if you consider purchasing one, make sure that you get the USB-A type if your computer does not have a USB-C micro USB port on it.

https://support.intelrealsense.com/hc/en-us/community/posts/360046835933/comments/360011483694

Eslamrady42, 2020-06-16:
Ok that’s sounds great, but it’s working with me on 

**realsense-viewer** 

And I tried on another laptop before with lower performance and worked well with me ... 
 I think all I need to solve this problem to know how is RealSense choose the USB camera port automatically and the interface between the camera and Ubuntu. 

Also my application can’t support 36w for only USB hub !!

MartyG-RealSense, 2020-06-16:
My understanding is that there are differences between how the RealSense SDK and the ROS wrapper handle the camera.

You can manually define in the ROS wrapper the USB port that the wrapper looks for a particular camera in using the **usb_port_id** instruction instead of a serial number.  It may not help if the problem is with communication between the port and the camera though.

"usb_port_id: will attach to the device with the given USB port (usb_port_id). i.e 4-1, 4-2 etc.  Default, ignore USB port when choosing a device."

https://github.com/IntelRealSense/realsense-ros#launch-parameters

Eslamrady42, 2020-06-17:
Ok thank you @MartyG-RealSense for your time, you can close this case, I’ll try with the USB hub and I’ll creat new case if there is any errors .

MartyG-RealSense, 2020-06-17:
Thank you very much for your patience during this case @Eslamrady42  - good luck!

closed 2020-06-17 by MartyG-RealSense