realsenseai/librealsense #6452: usb attach error

Long888, 2020-05-25:
[image]

MartyG-RealSense, 2020-05-25:
Please help me to understand what is represented in the image, please.

-  It looks as though two versions of the same script file **DeviceWatcher.java** are being shown side by side, one of them in Librealsense SDK 2.33.1 and the other in SDK 2.34.0.  This file is located in the Android wrapper at this path:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/librealsense/src/main/java/com/intel/realsense/librealsense/DeviceWatcher.java

-  You referred to a USB attach error.  What is the error message that you are receiving, please?

Long888, 2020-05-25:
In SDK 2.34.0, when the USB camera is connected, onDeviceDetach will be called , but not in the old version.

MartyG-RealSense, 2020-05-25:
I am not an Android wrapper programmer so I cannot speak for the intent of the wrapper developers.  Reading the code though, it looks like the **DeviceListener** listens for whether a device is inserted (Attached) or unplugged (Detached) and directs the program to the **invalidateDevices()** function to decide what should happen next.  In that block, it either adds or removes the device depending on the checks that the logic makes about the status of variables.

Does the program still work the same in 2.34.0 as it did in 2.33.1, even if the code has differences?

Long888, 2020-05-25:
No ,it will call different function.

MartyG-RealSense, 2020-05-25:
The code looks okay to me (though I am honestly nor an Android wrapper programming expert).  The listener checks for the insertion status, and takes an action depending on whether an Attach or Detach event is detected.

It is possible for the USB port to mis-identify an insertion as a removal if the connector is inserted in a certain way, such as accidentally pulling the cable backwards a little after pushing it in, or not fully inserting it as far into the port as it will go.  The ideal insertion method is a fast, firm push action instead of a slow insertion. 

If a fast, firm insertion does not fix the problem then I would recommend using Librealsense version 2.33.1 instead of 2.34.0.

kafan1986, 2020-05-25:
@Long888 This indeed looks like a bug. Even if a usb device is plugged and anyone has used the listerner to maybe receive a callback and do to some initialization etc., it will never be triggered. As both insertion and disconnection are will trigger the onDeviceDetach method.

MartyG-RealSense, 2020-05-25:
@kafan1986   Thanks vry much for your input on this case.   Are you experiencing this detach problem with the Android wrapper yourself, please?

kafan1986, 2020-05-25:
@MartyG-RealSense No, I have not tested it personally, so can not guarantee its behaviour. Last I used v2.31.0 SDK and firmware 5.12.1.0.

MartyG-RealSense, 2020-05-26:
@kafan1986   Thanks again for your help.

@Long888  I recommend using 2.33.1 for now and see if your problem is fixed when the next SDK version (probably 2.35.0) is released.

ev-mp, 2020-05-31:
@Long888 , the issue was reproduced and addressed in the attached PR.
To be published in the next release

MartyG-RealSense, 2020-06-14:
@Long888  I can confirm that this bug-fix was included in SDK version **2.35.0** published on 31 May 2020.  Can you test your project with SDK version 3.35.0 or the newer 2.35.2 and let us know if it has resolved your problem, please?   Thanks!

MartyG-RealSense, 2020-06-25:
Hi @Long888  Do you require futher assistance with this case please, or can it be closed?  Thanks!

closed 2020-06-25 by Long888