realsenseai/librealsense #6453: Sync multicamera with an user defined delay

AleBasso80, 2020-05-25:
Camera Model | D435
Firmware Version | 05.12.3
Operating System & Version | Ubuntu 18.04
Platform | PC
SDK Version | 2.33.1
Language | C++
Segment | Multiple cameras

Hello,
I would like to get data from two (or more) D435 cameras defining an acq delay. For instance, I want to get data form cam 1 at t=0, from cam 2 at t=50, from cam 1 at t=150, cam2 t=200 and so on. Looking at the documentation I don't find a way to achive this goal.
Any suggestion?

Thanks!

MartyG-RealSense, 2020-05-25:
I considered your question very carefully.   I ca visualise all the required parts of a solution but do not have a ready-made one.  I feel that the long discussion about individual access of multiple cameras in the link below will probably take care of the multiple camera control aspect of the project, just leaving the matter of setting up a 50 second delay timer between camera accesses instead of accessing all cameras at the same time.

https://github.com/IntelRealSense/librealsense/issues/2219

AleBasso80, 2020-05-26:
Hi @MartyG-RealSense ,
my cameras live in different threads, therefore such synchronization would be quite difficult to implement and wouldn't probably be very accurate. Moreover, I'd like to reduce the camera data stream by means of actual sync trigger signals. I think the only way would be a delay with respect to the sync signal in the camera firmware. What do you think?

MartyG-RealSense, 2020-05-26:
If each of the cameras are on individual streams that are hard to sync, one way to have them capture at specific durations without them having to talk to each other at all might be to give each of them a timer with a different 'reset to zero' time.

Camera 1 counts to 50, captures and resets to 0 and starts counting towards 50 again.
Camera 2 counts to 100, captures and resets to 0 and starts counting towards 100 again.
Camera 3 counts to 150, captures and resets to 0 and starts counting towards 150 again.
Camera 4 counts to 200, captures and resets to 0 and starts counting towards 200 again.

that way, you create a situation where the cameras appear to be capturing 50 seconds apart but are actually following completely independent timing that gives the illusion that it is a single pre-programmed sequence.

timprepscius, 2020-05-26:
I am also interested in this problem, but I am interested if you wanted camera 1 at time 0, camera 2 at time 100 microseconds, camera 3 at time 200 microseconds, camera 4 at time 300 microseconds.

(ie, some time interval shorter than the frame rate of the cameras).

AleBasso80, 2020-05-27:
@MartyG-RealSense : what you propose is certainly possible (and thanks for that BTW), but it wouldn't allow me to reduce the actual data stream on the USB3 bus. A solution trigger based with delay could allow users to play with it.

@timprepscius : no, I'm not. I have a time consuming classifier downstream, therefore I would like to reduce the normal camera stream.

MartyG-RealSense, 2020-06-05:
@AleBasso80   Do you still require assistance with this case please?

AleBasso80, 2020-06-06:
@MartyG-RealSense no, it's fine. Sorry, I forgot to close the issue.

closed 2020-06-06 by AleBasso80