realsenseai/librealsense #6454: D435i: rs2::motion_stream_profile::get_motion_intrinsics() throws exception.

rafaelspring, 2020-05-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.03.00 | 
| Operating System & Version |  Win10 | 
| SDK Version                            | 2.34.0  | 
| Language                            |  C++ | 

### Issue Description
Title says most. Calling `get_motion_intrinsics()` on an `rs2::motion_stream_profile` (e.g. as obtained as one of the Motion Module's streaming profiles) throws an exception.

This only happens with the D435i. Works fine with the L515.

ev-mp, 2020-05-26:
@rafaelspring hello, 
The IMU for D435i is not factory-calibrated, therefore you need to follow the established procedure (and a [white paper](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Depth_D435i_IMU_Calib.pdf)) to generate and save motion intrinsic - see [documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md)

rafaelspring, 2020-05-26:
@ev-mp Thanks.

closed 2020-05-26 by rafaelspring