realsenseai/librealsense #6456: D435i: IMU/depth extrinsics rotation matrix == ID

rafaelspring, 2020-05-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version | Win10 | 
| SDK Version                            | 2.34.0 | 
| Language                            |  C++ | 

### Issue Description
The rotation matrix of IMU vs. depth as reported by rs2::stream_profile::get_extrinsics_to() is equal to the identity matrix, which is very unlikely.

```cpp
accel_stream_profile.get_extrinsics_to(depth_stream_profile); // --> rotation == ID
gyro_stream_profile.get_extrinsics_to(depth_stream_profile); // --> rotation == ID
```

MartyG-RealSense, 2020-05-25:
In a D435i case from March 2019,  @ev-mp said that the rotation matrix should "be diagonal identity matrix".

https://github.com/IntelRealSense/librealsense/issues/3474#issuecomment-478950357

rafaelspring, 2020-05-25:
@MartyG-RealSense Thanks but I think he was referring to depth/color extrinsics, not IMU/depth. But even with depth/color it shouldn't be exactly == ID.

ev-mp, 2020-05-26:
@rafaelspring, 
The D435i IMU extrinsic is precalculated based on CAD. Additionally, the IMU data is aligned in software to Depth Coordinate System, hense the rotational component of that extrinsic is the identity matrix. 
https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#sensor-origin-and-coordinate-system.
Check #5363 for a similar discussion.

P.S. RGB-Depth extrinsic's rotation is **roughly** equivalent to identity as stated.
Theoretically it could be the case for identity. In practice the rotation will always have a certain skew due to mechanical/assembly tolerances.

rafaelspring, 2020-05-26:
@ev-mp Ok thanks.
- So, as a user, should I assume that IMU values from the D435i are w.r.t. the depth coordinate system then?
- What is the situation on the L515? Unfortunately I can't get any data through the SDK due to #6455.

MartyG-RealSense, 2020-06-05:
Tagging @ev-mp for a reminder to please offer @rafaelspring input on the two questions above.  Thanks!

ev-mp, 2020-06-15:
@rafaelspring hello,
Both assumptions are correct. The coordinate systems for all the RealSense Depth cameras (and recursively on all their dependent sensors) are aligned w.r.t to the Depth stream Coordinate System (with exception for T265 that is not a Depth camera).
Note that "aligned" == orientation. Obviously the origin for each sensor is different.

MartyG-RealSense, 2020-06-25:
Hi @rafaelspring  Do you require further assistance with this case, please?  Thanks!

rafaelspring, 2020-06-25:
No, feel free to close the ticket.

Rafael Spring
CTO, Co-Founder
DotProduct LLC / DotProduct GmbH

closed 2020-06-25 by MartyG-RealSense