realsenseai/librealsense #6458: Compile error due to libusb in Ubuntu 19.10

jimruxton, 2020-05-26:
I am having trouble building librealsense on Ubuntu 19.10 . All builds fine till I get to these errors shown below. 
`/usr/bin/ld: ../../librealsense2.so.2.34.0: undefined reference tlibusb_get_port_numbers`

`/usr/bin/ld: ../../librealsense2.so.2.34.0: undefined reference to libusb_handle_events_completed`

My cmake command is 

`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_PYTHON_BINDINGS=true `

Any idea what I need to fix the libusb errors. The tail end of the build is shown below.

`
[ 59%] Building CXX object examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o
cd /home/jim/Downloads/librealsense/build/examples/hello-realsense && ccache /usr/bin/c++  -DBUILD_EASYLOGGINGPP -DBUILD_SHARED_LIBS -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DGLFW_DLL -DHWM_OVER_XU -DRS2_USE_LIBUVC_BACKEND -DUNICODE -I/home/jim/Downloads/librealsense/realsense2 -I/home/jim/Downloads/librealsense/src -I/home/jim/Downloads/librealsense/examples/hello-realsense/rs-hello-realsense -I/home/jim/Downloads/librealsense/build -I/home/jim/Downloads/librealsense/include  -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat -Wformat-security -mssse3 -pthread -O3 -DNDEBUG -fPIE   -std=gnu++11 -o CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o -c /home/jim/Downloads/librealsense/examples/hello-realsense/rs-hello-realsense.cpp
[ 59%] Linking CXX executable rs-hello-realsense
cd /home/jim/Downloads/librealsense/build/examples/hello-realsense && /usr/bin/cmake -E cmake_link_script CMakeFiles/rs-hello-realsense.dir/link.txt --verbose=1
ccache /usr/bin/c++   -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat -Wformat-security -mssse3 -pthread -O3 -DNDEBUG  -rdynamic CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o  -o rs-hello-realsense -Wl,-rpath,/home/jim/Downloads/librealsense/build: ../../librealsense2.so.2.34.0 /usr/lib/x86_64-linux-gnu/libglfw.so.3.3 -lGL -lGLU 
/usr/bin/ld: ../../librealsense2.so.2.34.0: undefined reference to `libusb_get_port_numbers'
/usr/bin/ld: ../../librealsense2.so.2.34.0: undefined reference to `libusb_handle_events_completed'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:88: examples/hello-realsense/rs-hello-realsense] Error 1
make[2]: Leaving directory '/home/jim/Downloads/librealsense/build'
make[1]: *** [CMakeFiles/Makefile2:496: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
make[1]: Leaving directory '/home/jim/Downloads/librealsense/build'
make: *** [Makefile:130: all] Error 2

`

MartyG-RealSense, 2020-05-26:
I have had a case where Librealsense has been successfully built in Ubuntu 19.10 using the FORCE RSUSB method, so it should be possible.  Assuming that you have an internet connection (which is a requirement of this installation method), does the build succeed please if you do not include the **-DBUILD_PYTHON_BINDINGS=true** statement please?

jimruxton, 2020-05-26:
Thanks @MartyG-RealSense .  I tried without  -DBUILD_PYTHON_BINDINGS=true and got the same result unfortunatelel.

MartyG-RealSense, 2020-05-26:
Are you launching the CMake instruction whilst in the Build folder of Librealsense please?   (Librealsense > Build)

If you do not have a Build directory in your Librealsense root folder, it can be created and then navigated to with the instruction ** mkdir build && cd build.**

jimruxton, 2020-05-26:
Thanks @MartyG-RealSense
Yes I am executing the cmake from within the build folder.

`~/Downloads/librealsense/build$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
`

Any other thoughts on this.

MartyG-RealSense, 2020-05-26:
I have never seen these libusb error messages before.  It looks as though it is tripping up on building the **hello-realsense** example program though.  Try leaving out **-DBUILD_EXAMPLES=true** this time please.

jimruxton, 2020-05-26:
It seems that the problem program hello-realsense is still being built despite removing examples from the cmake command. New cmake command is below followed by the tail of the build output. Same errors as before.

` cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release
`

`[ 59%] Building CXX object examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o
cd /home/jim/Downloads/librealsense/build/examples/hello-realsense && ccache /usr/bin/c++  -DBUILD_EASYLOGGINGPP -DBUILD_SHARED_LIBS -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DGLFW_DLL -DHWM_OVER_XU -DRS2_USE_LIBUVC_BACKEND -DUNICODE -I/home/jim/Downloads/librealsense/realsense2 -I/home/jim/Downloads/librealsense/src -I/home/jim/Downloads/librealsense/examples/hello-realsense/rs-hello-realsense -I/home/jim/Downloads/librealsense/build -I/home/jim/Downloads/librealsense/include  -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat -Wformat-security -mssse3 -pthread -O3 -DNDEBUG -fPIE   -std=gnu++11 -o CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o -c /home/jim/Downloads/librealsense/examples/hello-realsense/rs-hello-realsense.cpp
[ 59%] Linking CXX executable rs-hello-realsense
cd /home/jim/Downloads/librealsense/build/examples/hello-realsense && /usr/bin/cmake -E cmake_link_script CMakeFiles/rs-hello-realsense.dir/link.txt --verbose=1
ccache /usr/bin/c++   -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat -Wformat-security -mssse3 -pthread -O3 -DNDEBUG  -rdynamic CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o  -o rs-hello-realsense -Wl,-rpath,/home/jim/Downloads/librealsense/build: ../../librealsense2.so.2.34.0 /usr/lib/x86_64-linux-gnu/libglfw.so.3.3 -lGL -lGLU 
/usr/bin/ld: ../../librealsense2.so.2.34.0: undefined reference to `libusb_get_port_numbers'
/usr/bin/ld: ../../librealsense2.so.2.34.0: undefined reference to `libusb_handle_events_completed'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:88: examples/hello-realsense/rs-hello-realsense] Error 1
make[2]: Leaving directory '/home/jim/Downloads/librealsense/build'
make[1]: *** [CMakeFiles/Makefile2:496: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
make[1]: Leaving directory '/home/jim/Downloads/librealsense/build'
make: *** [Makefile:130: all] Error 2
`

MartyG-RealSense, 2020-05-26:
It looks as though you have stripped the CMake build instruction about as far as you can go.

Did you do any other Librealsense installations on this computer before doing the RSUSB method?  If there are multiple conflicting installations then it may be better to start with a fresh install of Linux.

jimruxton, 2020-05-26:
I did try using a Linux package to install it though it didn't work out so I went this route. I thought I uninstalled all remnants of the package however. Could my error be related to this issue with libusb which I found here #555

MartyG-RealSense, 2020-05-26:
Though that case relates to the current SDK's predecessor Librealsense1, there is a potential for conflicts in any situation where you have files with the same filename and different ages.

jimruxton, 2020-05-26:
I noticed I have a libusb-0.1-4 on my system as well. Maybe that is causing an issue? 
`~$ dpkg -L libusb-0.1-4

/lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4

/usr/share/doc/libusb-0.1-4

/usr/share/doc/libusb-0.1-4/README.Debian

/usr/share/doc/libusb-0.1-4/changelog.Debian.gz

/usr/share/doc/libusb-0.1-4/copyright

/lib/x86_64-linux-gnu/libusb-0.1.so.4

`
Though I tried using this flag with no difference 

`-D USE_SYSTEM_LIBUSB=false
`

MartyG-RealSense, 2020-05-26:
I found an older Librealsense1 case where an old libusb file was also causing the libusb_get_port_numbers error:

https://github.com/IntelRealSense/librealsense/issues/180

jimruxton, 2020-05-26:
Thanks @MartyG-RealSense . I uninstalled libusb-0.1-4 but still no change. such a mystery..

MartyG-RealSense, 2020-05-26:
I'm not sure why it would be having such difficulty over building a simply constructed program such as **rs-hello-realsense.cpp**

If you have another PC available, you could try installing it on that and see if it makes a difference.  Otherwise, I will pick up the case again when I am back on-shift in the morning in 7 hours from the time of writing this.  Good luck!

MartyG-RealSense, 2020-05-27:
I did further research and found other cases where the build fails at 59%.  One user with this problem solved it by explicitly stating **-DBUILD_EXAMPLES=false** in their CMake instruction.

jimruxton, 2020-05-27:
Thank you I will try this later today and get back to you. It would be nice however to be able to build the examples but one step at a time I guess.

jimruxton, 2020-05-27:
Ok @MartyG-RealSense that worked! Thanks! Strangely it also built pyrealsense2 even though it wasn't asked to build the python bindings? It seems as if there is something wrong with the cmake flags. The command I used is 

`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=false`

Now any thoughts on how I can build the examples?

MartyG-RealSense, 2020-05-27:
Great news - congratulations!

Advice on only building the samples is offered here:

https://github.com/IntelRealSense/librealsense/issues/2946#issuecomment-449290237

jimruxton, 2020-05-27:
Yes I tried that . When building only the examples I get the same errors as previously.

MartyG-RealSense, 2020-05-27:
Any luck if you do **make** after CMake?

**cmake .. -DBUILD_EXAMPLES=true && make -J && sudo make install**

jimruxton, 2020-05-27:
No that is what I tried but I get the same errors as before.

MartyG-RealSense, 2020-05-27:
Please try to build it "headless" (no examples that use a graphics interface such as OpenGL) in case the build is breaking due to a graphics related issue:

cmake ../ -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false

jimruxton, 2020-05-27:
Ok I tried that and get the same result which I would expect since it seems to be a libusb issue, not really graphics related from what I can see.

MartyG-RealSense, 2020-05-27:
Would it be possible to use Ubuntu 18.04 instead, since that is more of a known quantity in regard to Librealsense installation.

jimruxton, 2020-05-27:
Sorry no I can't go back to 18.04. I was actually hoping to migrate up to 20.04 soon.  I ran make with the -d option and here is the tail of that in case that is of use..

`   Prerequisite '../include/librealsense2/hpp/rs_device.hpp' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite '../include/librealsense2/hpp/rs_frame.hpp' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite '../include/librealsense2/hpp/rs_options.hpp' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite '../include/librealsense2/hpp/rs_pipeline.hpp' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite '../include/librealsense2/hpp/rs_processing.hpp' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite '../include/librealsense2/hpp/rs_record_playback.hpp' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite '../include/librealsense2/hpp/rs_sensor.hpp' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite '../include/librealsense2/hpp/rs_types.hpp' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite '../include/librealsense2/rs.h' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite '../include/librealsense2/rs.hpp' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite '../examples/hello-realsense/rs-hello-realsense.cpp' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
     Prerequisite 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/flags.make' is older than target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
    No need to remake target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o'.
    Considering target file 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make'.
    File 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make' was considered already.
    Considering target file 'librealsense2.so.2.34.0'.
     Looking for an implicit rule for 'librealsense2.so.2.34.0'.
     Trying pattern rule with stem 'librealsense2.so.2.34.0'.
     Trying implicit prerequisite 'librealsense2.so.2.34.0,v'.
     Trying pattern rule with stem 'librealsense2.so.2.34.0'.
     Trying implicit prerequisite 'RCS/librealsense2.so.2.34.0,v'.
     Trying pattern rule with stem 'librealsense2.so.2.34.0'.
     Trying implicit prerequisite 'RCS/librealsense2.so.2.34.0'.
     Trying pattern rule with stem 'librealsense2.so.2.34.0'.
     Trying implicit prerequisite 's.librealsense2.so.2.34.0'.
     Trying pattern rule with stem 'librealsense2.so.2.34.0'.
     Trying implicit prerequisite 'SCCS/s.librealsense2.so.2.34.0'.
     No implicit rule found for 'librealsense2.so.2.34.0'.
     Finished prerequisites of target file 'librealsense2.so.2.34.0'.
    No need to remake target 'librealsense2.so.2.34.0'.
   Finished prerequisites of target file 'examples/hello-realsense/rs-hello-realsense'.
  Must remake target 'examples/hello-realsense/rs-hello-realsense'.
Putting child 0x55ee9a3355a0 (examples/hello-realsense/rs-hello-realsense) PID 32098 on the chain.
Live child 0x55ee9a3355a0 (examples/hello-realsense/rs-hello-realsense) PID 32098 
[ 85%] Linking CXX executable rs-hello-realsense
Reaping winning child 0x55ee9a3355a0 PID 32098 
Live child 0x55ee9a3355a0 (examples/hello-realsense/rs-hello-realsense) PID 32100 
/usr/bin/ld: ../../librealsense2.so.2.34.0: undefined reference to `libusb_get_port_numbers'
/usr/bin/ld: ../../librealsense2.so.2.34.0: undefined reference to `libusb_handle_events_completed'
collect2: error: ld returned 1 exit status
Reaping losing child 0x55ee9a3355a0 PID 32100 
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:85: examples/hello-realsense/rs-hello-realsense] Error 1
Removing child 0x55ee9a3355a0 PID 32100 from chain.
Reaping losing child 0x55cd9a995160 PID 32097 
make[1]: *** [CMakeFiles/Makefile2:496: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
Removing child 0x55cd9a995160 PID 32097 from chain.
Reaping losing child 0x55983fa64290 PID 32056 
make: *** [Makefile:130: all] Error 2
Removing child 0x55983fa64290 PID 32056 from chain.
 
`

MartyG-RealSense, 2020-05-27:
I'm not sure that there is anything else that can be done to rescue this build.  It has ended up with the same errors that were present at the start of the case (**libusb_handle_events_completed** and **libusb_get_port_numbers**).  It may be better to try again with the upcoming SDK version (probably 2.35.0) or wait until you upgrade to 20.04 and see if the clean Ubuntu install that it will provide removes whatever was preventing the examples from building.

jimruxton, 2020-05-28:
Thanks @MartyG-RealSense I guess I am a bit surprised that none of the developers have a sense from the error messages what is going wrong with the build. I don't really see downloading a new distribution as a great solution. I would prefer to know what is throwing the error message and why.

MartyG-RealSense, 2020-05-28:
I have never seen the 'Trying pattern rule with stem' error before.  It seems to result from CMake rather than Librealsense.  

It is recommended to use CMake version 3.8 or newer for Librealsense building.  Could you check which version you have and update to a newer one from the CMake website if it is old please?

https://cmake.org/download/

jimruxton, 2020-05-28:
My cmake version is much newer that 3.8 ie.
cmake version 3.13.4

MartyG-RealSense, 2020-05-28:
I will refer your case to a support manager for further guidance.  I apologise for the delay in resolving your problem in the meantime.

jimruxton, 2020-05-28:
Thank you. I really appreciate the support.

jimruxton, 2020-05-28:
Ok I figured this out. I rebuilt and installed the latest[ libusb](https://sourceforge.net/projects/libusb/).  This may not be necessary however what was important was to get rid of an older version of libusb-1.0.so  that was in /usr/local/lib  which is where librealsense2.so.2.34 lives. Installing the new libusb put it in /usr/lib . I suppose I could have linked the old libusb to this one as well but the build process was able to find the new libusb anyway.

MartyG-RealSense, 2020-05-28:
Thanks for the update!  Is it all working okay now?

jimruxton, 2020-05-28:
Everything built so that's good. I am still waiting for my sensor to arrive but I think all is good. I believe we can close this issue.

MartyG-RealSense, 2020-05-28:
Awesome news - thanks a lot for your patience during this process.  I will close the case now.  Good luck!

closed 2020-05-28 by MartyG-RealSense