realsenseai/librealsense #6459: Exception Attempted to read or write protected memory

YHabotAetrex, 2020-05-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Win10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.29.0  | 
| Language                            |  C++                          | 

### Issue Description
I've created a C++ dll that exposes a function that does pipe.start() and another method that does pipe.stop(). I'm calling these functions from C# Winforms application that stress tests 4 D415 cameras by doing start()-stop() sequence repeatedly.
The problem is that the sequence start()-stop()-start() throws an exception in C#:
"Attempted to read or write protected memory. This is often an indication that other memory is corrupt."

What is the proper sequence to initialize - dispose - reinitialize the D415 cameras?

MartyG-RealSense, 2020-05-26:
Start-stop-start of the pipeline continuously without pause sounds as though it would get the SDK confused quickly.

If you are able to run C++ programs from C#, I wonder if using the SDK's camera test program **unit-tests** instead would provide a more stable method of camera testing?

https://github.com/IntelRealSense/librealsense/tree/master/unit-tests

YHabotAetrex, 2020-05-26:
Thanks for the prompt replay
> Start-stop-start of the pipeline continuously without pause sounds as though it would get the SDK confused quickly.

It throws the "Attempted to read or write protected memory" even with a pause of a few minutes.

> If you are able to run C++ programs from C#, I wonder if using the SDK's camera test program **unit-tests** instead would provide a more stable method of camera testing?

I looked at https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/unit-tests-live.cpp TEST_CASE("Start-Stop streaming and found the following:
`for (auto s : profiles.first)
            {
                REQUIRE_NOTHROW(s.stop());
                REQUIRE_NOTHROW(s.close());
            }`
I read the documentation about a pipeline class that has start() and stop() methods, but no close(). In the test snippet above, what is the class that implements close()?

MartyG-RealSense, 2020-05-27:
Going through some older cases, I came across one with C++ scripting for start-stop in a loop that I  thought might be a useful reference for you.

https://github.com/IntelRealSense/librealsense/issues/4897

YHabotAetrex, 2020-05-27:
Thank Marty, I read issue #4897 . That's the problem that I am facing. Issue #4897 and unit-tests don't give me any clue on how to resolve this problem. I believe that Intel RealSense needs to provide a proper termination and resources de-allocation on pipe.stop().

I'm still waiting for an answer to my question from May 26:
I looked at https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/unit-tests-live.cpp TEST_CASE("Start-Stop streaming and found the following:
for (auto s : profiles.first) { REQUIRE_NOTHROW(s.stop()); REQUIRE_NOTHROW(s.close()); }
I read the documentation about a pipeline class that has start() and stop() methods, but no close(). In the test snippet above, what is the class that implements close()?

MartyG-RealSense, 2020-05-27:
Your question about **unit-tests-live.cpp** above is outside of my Librealsense programming knowledge unfortunately.  

C# is quite strict about how resources are released:

https://github.com/IntelRealSense/librealsense/issues/2845#issuecomment-444448923

MartyG-RealSense, 2020-06-05:
@YHabotAetrex   I apologise that you did not receive a further response on this case.  Do you still require assistance please, or can the case be closed?   Thanks!

MartyG-RealSense, 2020-06-06:
Thanks for your patience.

It looks as though we have not previously mentioned in this discussion the advice from Dorodnic on use of Dispose() in .NET:

https://github.com/IntelRealSense/librealsense/issues/2845#issuecomment-444448923

It is also worth noting that the latest **5.12.5** firmware version added a fix for the bug of IR frames no longer arriving after several stop / starts.

YHabotAetrex, 2020-06-06:
Thanks Marty. I did see issue 2845 and tried, but it still threw the exception "Attempted to read or write protected memory". I believe that the C++ pipeline does not release the memory.

MartyG-RealSense, 2020-06-06:
I researched your case further very carefully.  The complication of accessing a C++ script from C# means though that a RealSense developer team member may be better able to advise on this subject (I am on the Support team).  

If it were possible to consider doing the open-close with a purely C# script, the link below may provide some useful insights about closing a pipeline in C#.

https://github.com/IntelRealSense/librealsense/issues/5017

Otherwise, if you prefer to stay with the C++ calling from C# method then I will seek input on the problem from within Intel.  

Thanks for your patience!

MartyG-RealSense, 2020-06-14:
@YHabotAetrex   Do you still need assistance on this case, please?

YHabotAetrex, 2020-06-14:
I do need assistance. Is there another contact for RealSense support that can help me with this issue?

MartyG-RealSense, 2020-06-15:
I cannot provide you with a direct contact to another support engineer with the C++ knowledge.  I will make a request for assistance.  Thanks very much for your patience.

RealSenseSupport, 2020-06-24:
Hi @YHabotAetrex,
Could you share more detail of your code?
I can't reproduce this issue in my cpp test app.

BTW, the type of s is rs2::sensor in the reference code in the comment.
https://github.com/IntelRealSense/librealsense/issues/6459#issuecomment-634126058

AetrexJenkins, 2020-06-24:
rs2:sensor.stop() did the trick. Instead of doing pipeline.start - pipeline.stop - pipeline.start we are doing pipeline.start - sensor.stop - pipeline.start.
Thanks for the dedication and help in resolving this issue. You may close it now.

MartyG-RealSense, 2020-06-24:
@AetrexJenkins   Great news that you found a solution - thanks so much for the update!

closed 2020-06-24 by MartyG-RealSense