realsenseai/librealsense #6460: python3.6 & 3.7 - Could not find a version that satisfies the requirement realsense2 (from versions: )

Tomblarom, 2020-05-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Raspbian GNU/Linux 10 (buster)  | 
| Kernel Version (Linux Only)    |  4.19.97-v7l+ | 
| Platform                                 | Raspberry Pi 4 - 4GB  |
| SDK Version                            | - | 
| Language                            |  python3.6 / python3.7                          | 
| Segment			| - | 
 
### Issue Description
I'm currently trying to get [`pyrealsense2`](https://pypi.org/project/pyrealsense2/) installed using pip. I removed every previous python installation using `apt purge ..` and reinstalled python3.7, tested, installed python3.6 as well, all using [this guide](https://installvirtual.com/install-python-3-7-on-raspberry-pi/) and the error still occurs.. I tried all sort if pip commands:
```
pip install realsense2
pip3.6 install realsense2
pip3.7 install realsense2
```

Even with `python3.6 -m pip install realsense2` I'm getting this error:
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting realsense2
  Could not find a version that satisfies the requirement realsense2 (from versions: )
No matching distribution found for realsense2
```

It's officially supported right?
[image]

**Additional information:**
```
$ python -V
Python 3.6.9

$ python3.6 -m pip -V
pip 20.1.1 from /usr/local/lib/python3.6/site-packages/pip (python 3.6)

$ python3.7 -m pip -V
pip 20.1.1 from /usr/local/lib/python3.7/site-packages/pip (python 3.7)
```

MartyG-RealSense, 2020-05-26:
Hi @Tomblarom   I notice that you keep using **install realsense2** above.  The package name is **pyrealsense2**.

Could you try this command please:

**pip3 install pyrealsense2**

Tomblarom, 2020-05-26:
Yes indeed, but it's not fixing my issue here:

```
(venv) $ pip3 install pyrealsense2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2

(venv) $ cat requirements.txt
...
pyrealsense2==2.34.0.1470
...

(venv) $ python3.7 -m pip install -r requirements.txt
...
ERROR: Could not find a version that satisfies the requirement pyrealsense2==2.34.0.1470 (from -r requirements.txt (line 8)) (from versions: none)
ERROR: No matching distribution found for pyrealsense2==2.34.0.1470 (from -r requirements.txt (line 8))

```

MartyG-RealSense, 2020-05-26:
As far as I know there is no need to specify a particular Librealsense SDK version (e.g 2.34.0) when using pip to install the pyrealsense2 wrapper.

Does it work if you launch the pip instruction from the Python 3 folder that pip is installed in?  On Python 3.7 on Ubuntu that is something like **Python37/Scripts** - I am not sure what it would be on Raspbian Buster.

dorodnic, 2020-05-26:
Hi @Tomblarom 
The versions you highlighted are x86_x64, Raspberry Pi is Arm. We don't have pip packages for it, yet.

Tomblarom, 2020-05-27:
This means I need to build it from source? [python - Building From Source](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source)

MartyG-RealSense, 2020-05-27:
@Tomblarom  Building from source would be the recommended option if pip installation is not available, yes.

MartyG-RealSense, 2020-06-05:
Hi @Tomblarom   Do you require further assistance with this case please or can it be closed?   Thanks!

closed 2020-06-14 by MartyG-RealSense