realsenseai/librealsense #6461: D435 Camera Master Sync bit of weirdness after 52 seconds a reset

timprepscius, 2020-05-26:
I'm watching the signal sent from the D435 via the Camera Sync Master pin.

It seems to repeat correctly for around 52 seconds (human timing), and then stop, reset and start again.
It does this at any frame rate.

I can compensate, however, I'm wondering if this is an indication I'm doing something wrong.

Does this ring any bells?

--

Also, while I can detect and use the camera-sync-master signal, somehow I can't see the "Output Trigger Enabled" signal.

Is this trigger a hardware trigger?  Is it on the same pin output as the camera-sync?

Thanks

MartyG-RealSense, 2020-05-27:
If the reset is independent of frame rate, I wonder if it is an electrostatic discharge (ESD) event, where the electrical charge discharges and then starts building up again at the same rate so that it appears like it is occurring after a predictable period of time after the previous discharge.  

If the cameras are connected with sync cabling then they may be vulnerable to an ESD discharge event occurring if the cable has not had ESD protection components built into it.  A discharge resets the camera's frame counter.  This phenomenon and a diagram for building a circuit to protect against ESD is described in Section 1 of the multiple camera white paper.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

Regarding Output Trigger Enabled: there was a RealSense user who said: "in order to get a pulse out of VSYNC pin I have to set [in the RealSense Viewer] the **Stereo Module->Controls->Output Trigger Enabled** and **Stereo Module->Controls->Inter Cam Sync Mode = 1** (i.e. master).  There's no pulse on the VSYNC pin with any other settings combination".

timprepscius, 2020-05-27:
I think you are right.  I just wanted to make sure the 52 second is not a feature rather than my bug.

Is the ESD frame counter reset only via the pin 5 wire?  Or can it occur via other means?

timprepscius, 2020-05-27:
Oh an regarding the:

"in order to get a pulse out of VSYNC pin I have to set [in the RealSense Viewer] the Stereo Module->Controls->Output Trigger Enabled and Stereo Module->Controls->Inter Cam Sync Mode = 1 (i.e. master). There's no pulse on the VSYNC pin with any other settings combination"

I personally, now (I can't remember years ago), D435, almost latest firmware, see a signal when Inter Cam Sync Mode = 1, the Output Trigger Enabled seems to do nothing. (but again, it's confusing why there is a trigger if there is cam_sync_mode, etc, etc)

MartyG-RealSense, 2020-05-27:
I saw a past message during my research of this case that said that the Output Trigger Enabled button did nothing at that point in time, but didn't know if its functionality had been enabled since then in the current-day Viewer (I do not have a hardware sync set up right now to test it).

I would think that the frame counter would be reset if the camera hardware is reset (by doing a Hardware Reset in the RealSense Viewer's "More" option, using hardware_reset() in a script or unplugging / replugging the camera.

timprepscius, 2020-05-27:
I think I may have conquered the ESD problem.  I'm not sure yet though, it's hard to tell, because my environment changes.

Could you verify one thing:
The frame sync pulse seems to occur *after* the shutter has opened and closed.
I'm seeing that the exposure time it counter negative in relation to the sync pulse time.  Is this correct?

MartyG-RealSense, 2020-05-27:
I'm not certain on this detail.  Though a RealSense user said that during their tests on the pins during hardware sync, pin 5 seemed to deliver a 5V "flank" after each depth frame was captured.  This may tally with your own experiences.

MartyG-RealSense, 2020-06-05:
Hi @timprepscius   Do you require further assistance with this case please or can it be closed?   Thanks!

closed 2020-06-18 by MartyG-RealSense