realsenseai/librealsense #6462: Untable to build from vcpkg

thecreativecontrollab, 2020-05-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR305} 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10) /                                      | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 

### Issue Description
I am following these instructions:

git clone https://github.com/Microsoft/vcpkg.git
cd vcpkg
./bootstrap-vcpkg.sh
./vcpkg integrate install
./vcpkg install realsense2

When I reach the final step I receive an error message to ensure that I am using the latest portfiles with .\vcpkg update.  
[image]

thecreativecontrollab, 2020-05-26:
Resolved.  I had to follow the correct instructions for Windows. 

https://github.com/IntelRealSense/librealsense/issues/6131#issue-587968790

MartyG-RealSense, 2020-05-27:
Hi @thecreativecontrollab    I notice that after posting your question, you linked to https://github.com/IntelRealSense/librealsense/issues/6131 -  did it help your problem, please?

danilogr, 2020-05-28:
I am having the same problem here.
```
Error: Building package realsense2:x64-windows failed with: BUILD_FAILED
Please ensure you're using the latest portfiles with `.\vcpkg update`, then
submit an issue at https://github.com/Microsoft/vcpkg/issues including:
  Package: realsense2:x64-windows
  Vcpkg version: 2020.02.04-nohash-external
```

I dug into the log files, and it seems to be related to `alt_sstream_impl.hpp`.
```
Unknown compiler version - please run the configure tests and report the results
C:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\boost\boost/format/alt_sstream_impl.hpp(1): warning C4828: The file contains a character starting at offset 0x1df3 that is illegal in the current source character set (codepage 65001).
ninja: build stopped: subcommand failed.
```

danilogr, 2020-05-28:
I also got a couple of these
```
C:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\src\win\win-helpers.cpp(15): fatal error C1083: Cannot open include file: 'atlstr.h': No such file or directory
```

MartyG-RealSense, 2020-06-05:
Hi @danilogr   Do you require further assistance on this case or can it be closed, please?   Thanks!

danilogr, 2020-06-05:
Yes, I am still unable to build `realsense2` using `vcpkg`

danilogr, 2020-06-06:
```
C:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\src\winusb\device-winusb.h(14): fatal error C1083: Cannot open include file: 'atlstr.h': No such file or directory
```
I updated my `vcpkg` package files once again (still `2020.02.04-nohash`) 
Same problem: Cannot find `atlstr.h`
[install-x64-windows-dbg-out.log](https://github.com/IntelRealSense/librealsense/files/4739160/install-x64-windows-dbg-out.log)

```
[61/131] C:\PROGRA~2\MIB055~1\2019\BUILDT~1\VC\Tools\MSVC\1424~1.283\bin\Hostx64\x64\cl.exe   /TP -DBUILD_EASYLOGGINGPP -DBUILD_SHARED_LIBS -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DENFORCE_METADATA -DHWM_OVER_XU -DNOMINMAX -DRS2_USE_WMF_BACKEND -DUNICODE -D_CRT_SECURE_NO_WARNINGS -D_UNICODE -D__SSSE3__ -Drealsense2_EXPORTS -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\realsense2 -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\src -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\console_bridge\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\cpp_common\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\rosbag_storage\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\roscpp_serialization\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\rostime\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\roscpp_traits\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\roslz4\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\msgs -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\boost -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\lz4\lib -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\include /DWIN32 /D_WINDOWS /W3 /utf-8 /GR /EHsc /MP  /MP /bigobj /wd4819 /D_DEBUG /MDd /Z7 /Ob0 /Od /RTC1 /showIncludes /FoCMakeFiles\realsense2.dir\src\winusb\device-winusb.cpp.obj /FdCMakeFiles\realsense2.dir\ /FS -c C:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\src\winusb\device-winusb.cpp
FAILED: CMakeFiles/realsense2.dir/src/winusb/device-winusb.cpp.obj 
C:\PROGRA~2\MIB055~1\2019\BUILDT~1\VC\Tools\MSVC\1424~1.283\bin\Hostx64\x64\cl.exe   /TP -DBUILD_EASYLOGGINGPP -DBUILD_SHARED_LIBS -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DENFORCE_METADATA -DHWM_OVER_XU -DNOMINMAX -DRS2_USE_WMF_BACKEND -DUNICODE -D_CRT_SECURE_NO_WARNINGS -D_UNICODE -D__SSSE3__ -Drealsense2_EXPORTS -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\realsense2 -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\src -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\console_bridge\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\cpp_common\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\rosbag_storage\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\roscpp_serialization\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\rostime\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\roscpp_traits\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\roslz4\include -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\rosbag\msgs -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\boost -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\third-party\realsense-file\lz4\lib -IC:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\include /DWIN32 /D_WINDOWS /W3 /utf-8 /GR /EHsc /MP  /MP /bigobj /wd4819 /D_DEBUG /MDd /Z7 /Ob0 /Od /RTC1 /showIncludes /FoCMakeFiles\realsense2.dir\src\winusb\device-winusb.cpp.obj /FdCMakeFiles\realsense2.dir\ /FS -c C:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\src\winusb\device-winusb.cpp
Microsoft (R) C/C++ Optimizing Compiler Version 19.24.28315 for x64
Copyright (C) Microsoft Corporation.  All rights reserved.

C:\vcpkg\buildtrees\realsense2\src\93b7ff8597-4347a45ee5\src\winusb\device-winusb.h(14): fatal error C1083: Cannot open include file: 'atlstr.h': No such file or directory```

Any solutions?

MartyG-RealSense, 2020-06-06:
@danilogr  I did further research and found a case of a RealSense user on Microsoft's vcpkg GitHub who had the same problem with being unable to find atlstr.h.  A person in that discussion suggested a method that worked for the RealSense user in the discussion.

https://github.com/microsoft/vcpkg/issues/11374

danilogr, 2020-06-09:
Hey @MartyG-RealSense, thank you for the link. It worked, and I am now able to build `librealsense` using `vcpkg` for Windows 10 x64.

For others that might have a similar problem,  their instructions are for Visual Studio 2017, but I am using Visual Studio 2019. VS2019 has no `Visual C++ ATL Support` component. And I believe that I had all ATL tools installed on my machine. After updating VS2019,  I uninstalled all old C++ tools (< `v142`) and installed all ATL tools for `v142`.

MartyG-RealSense, 2020-06-09:
Great news!   Thanks so much too for sharing your method for VS2019 with the RealSense community.

closed 2020-06-18 by MartyG-RealSense