realsenseai/librealsense #6463: Announcement: RealSense self calibration webinar replay now available

MartyG-RealSense, 2020-05-26:
Hi everyone,

An on-demand recording of the Intel RealSense webinar "**Self-Calibration for Intel RealSense Depth Cameras**" has been added to the RealSense webinar archives.  Please follow the steps below to view the replay.

1.  Visit the webinar registration page at the link below:

https://webinar.intel.com/self-calibration-for-depth-cameras-webinar

2.  Select the tick-box for the 9am or 6pm session and then fill in the registration form on the page and click the 'Register' button at the base of the form.

3.  On the next page that states that you have successfully registered for the webinar, click the 'watch the replay' button.

4.  On the next page, click the 'launch the presentation' button to launch the replay of the webinar.

kafan1986, 2020-05-28:
@MartyG-RealSense @dorodnic What is the minmum SDK version and firmware version requirement for performing self calibration?

MartyG-RealSense, 2020-05-28:
@kafan1986   The self calibration white paper recommends having at least firmware version: **5.12.02.100** and Librealsense version **2.33**.

You can receive feedback on the depth image quality by using the SDK's **Depth Quality Tool**.

malapatiravi, 2020-06-05:
@MartyG-RealSense  
I am using realsense SDK 2.34.1 in ubuntu 16.04 with firmware 5.12.3 connecting two D415 cameras. 
I am using the following compilation flags.
`cmake ../ -DBUILD_EXAMPLES=true -DENFORCE_METADATA=false -DBUILD_PYTHON_BINDINGS=true -DFORCE_LIBUVC=false -DBUILD_WITH_CUDA=true -DIMPORT_DEPTH_CAM_FW=false -DBUILD_WITH_TM2=false -DCMAKE_BUILD_TYPE=Release` 

When i open realsense viewer enable one of the cameras to perform onchip calibration the whole window freezes and turns grey. Any idea?
Later i ran the Depth Quality tool it opened the realsense viewer with less options and i was able to run and get some scores but it works only with first camera i am not sure if this is correct.
Could you please confirm what is the best way to perform onchip-calibration?

MartyG-RealSense, 2020-06-05:
There was a known problem with On-Chip Calibration freezing in SDK version 2.34.0 for some users.  If you are affected, as far as I know the only thing you can do is upgrade to the latest **2.35.0** SDK version or downgrade to **2.33.1**.

malapatiravi, 2020-06-09:
Dowgrading worked for me. Thanks @MartyG-RealSense

MartyG-RealSense, 2020-06-09:
@malapatiravi   You are very welcome.  I'm pleased that it worked for you!

closed 2020-06-22 by RealSenseCustomerSupport