realsenseai/librealsense #6465: stream depth data over network (UDP)

johnny-wang16, 2020-05-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am trying to figure out how to use gstreamer to stream 16bit depth data over the network. It doesn't have to be real-time as I realized the network bandwidth is just not enough for this huge amount of data. It just have to be as quick as it possibly can. I was also advised that saving it to a png image might provide some compression so my plan is first converting 16bit depth image to png and then stream it over UDP. 
I'd like ask if this functionality has been developed and if not, I'd like to get some suggestions on my way of streaming the depth data.

MartyG-RealSense, 2020-05-27:
@TimConnelly recently highlighted a new gstreamer plugin for RealSense that has been tested by its developers with a RealSense D435i camera.

https://github.com/IntelRealSense/librealsense/issues/2503#issuecomment-613653125

There was also a recently published Intel white-paper document on depth image compression by colorization.

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras

Additionally, a new 'Z16H' compressed depth format was introduced to the Librealsense SDK in SDK 2.32.1, though I believe that it is awaiting additional support to be added to the camera firmware driver before it is usable.

https://github.com/IntelRealSense/librealsense/pull/5564

johnny-wang16, 2020-05-28:
Hi @MartyG-RealSense ,  Thanks for your detailed response. I will try out the method mentioned in the white-paper document.

MartyG-RealSense, 2020-05-28:
@johnny-wang16   You're very welcome.   Good luck!

johnny-wang16, 2020-05-30:
For anyone who's looking for a simple solution to stream both RGB and depth data simultaneously over the network, I wrote some scripts that  does exactly this using the approach suggested above. Here's the link to my repo: https://github.com/johnny-wang16/rs-udpstreaming.git

MartyG-RealSense, 2020-05-30:
@johnny-wang16    Thanks so much for sharing your solution!  :)

MartyG-RealSense, 2020-06-05:
Hi again @johnny-wang16   Do you require further assistance with this case please, or can it be closed?  Thanks!

johnny-wang16, 2020-06-05:
I think its ready to be closed. Thank you very much.

closed 2020-06-05 by johnny-wang16

johnny-wang16, 2020-06-11:
Hello @MartyG-RealSense  ,I'd just want to point out that there are some typo in the Intel white-paper document on depth image compression by colorization you mentioned above. All of them are in section 3.2 Depth image recovery from colorized depth images in C++. I'd also like to ask some questions about the code.
I attached the part of code that has some typo here:

```
		unsigned char R = *in++;
		unsigned char G = *in++;
		unsigned char B = *in++;

		unsigned short hue_value = RGBtoD(R, G, B);

		if(out_value > 0)
		{
			if(!is_disparity)
			{
				unsigned short z_value = static_cast<unsigned short>((min_depth + (max_depth - min_depth) * hue_value / 1529.0f) + 0.5f);
				out++ = z_value;
			}
			else
			{
				float disp_value = min_disparity + (max_disparity - min_disparity) * out_value / 1529.0f;
				*out++ = static_cast<unsigned short>((1.0f / disp_value) / depth_units + 0.5f);
			}
		}
		else
		{
			*out++ = 0;
		}
```

Firstly, I highly suspect that the `out_value` in the if statement is actually `hue_value` as `out_value` variable is never initialized in this stand alone script. In addition, the `depth_unitis` variable is also never initialized.

There's also some inconsistency between the code and the formula mentioned in the paper. Mainly, why is it that in both cases of the  `if(!is_disparity)` statement, there's a + `0.5f` ? Is it just simply calibration?  Thank you very much.

MartyG-RealSense, 2020-06-11:
@johnny-wang16  I do not have knowledge of how the depth colorization technique in the white paper works, unfortunately.

@dorodnic  Do you have advice please for @johnny-wang16 regarding the questions in the comment above?

johnny-wang16, 2020-06-14:
Hello there, after I re-read the document , I am pretty sure those are just typos and the way to correct them is what I mentioned above. the 0.5f is just probably calibration.

MartyG-RealSense, 2020-06-14:
@johnny-wang16  Thanks very much for the update.

nicocarl, 2020-06-16:
Were you actually able to recalculate the depth values from the colorized image (hue colorizer)? I am not able to reconstruct the same depth values (also by taking into account the typos mentioned by @johnny-wang16 ). Moreover I had a look at https://github.com/TetsuriSonoda/rs-colorize/tree/development , where the equations look slightly different but still were not successful.

My idea was to colorize the 16 bit grayscale image to RGB image with 8 bit per channel and encode this with the e.g. H264 codec. Afterwards one could reconstruct the depth values by knowing min_distance and max_distance

johnny-wang16, 2020-06-16:
I was able to recalculate the pixel value. its just that im not using it at the moment so I didnt include it. I can update it on my github and  you can test it.

nicocarl, 2020-06-17:
That would be great!

johnny-wang16, 2020-06-21:
@nicocarl   sorry i didnt read your comment clearly. I have the script for reconstructing depth pixel values but I haven't actually tried it. I will upload it later and you can try it out and let me know if it works or not. I'd expect the result wouldn't fit the original depth value perfectly as the depth pixels have been transformed and compressed