realsenseai/librealsense #6466: details on apply_filter and colormap

johnny-wang16, 2020-05-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I'd like to understand what happens when the 16bit depth image gets mapped to color image when `rs_d435.get_depth_frame().apply_filter(color_map)
`
is called. I am guessing the resulting rgb image is the standard 8bit, 3channel image. But how does the 16bit get mapped to the 3 channels? does any compression/post-processing happens in the process? Thank you very much.

dorodnic, 2020-05-27:
Hi @johnny-wang16 
First, color mapping is not 1:1 and does not preserve original depth values. It is intended for visualization purposes only. 
This process is doing several things, you can read the code at [/src/proc/colorizer.cpp](https://github.com/IntelRealSense/librealsense/blob/master/src/proc/colorizer.cpp).
Pseudo-code: 
1. Ignore 0 values
2. (Optional) Calculate 16-bit histogram and perform histogram equalization, for higher contrast visualzation
3. Map 16-bit values to [0, 1] range using min and max values (either set by the user or dynamically calculated) 
4. Query from one of the built-in gradient functions that map float to (r,g,b) values

johnny-wang16, 2020-05-27:
Hi @dorodnic , Thanks for your detailed explanation. I'd like to confirm that the operation done to the depth via `apply_filter(color_map)`  is not the same as the operation described in  [Intel white-paper document on depth image compression by colorization](https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras)

dorodnic, 2020-06-02:
Yes
Please note that the white-paper is using not the default color map. You can modify selected color map in code using `set_option(RS2_OPTION_COLOR_SCHEME, 9.f);`
The paper is suggesting:
```cpp
float min_depth = 0.29f;
float max_depth = 10.0f;

// filter settings
thr_filter.set_option(RS2_OPTION_MIN_DEPTH, min_depth);
thr_filter.set_option(RS2_OPTION_MAX_DEPTH, max_depth);
color_filter.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED, 0);
color_filter.set_option(RS2_OPTION_COLOR_SCHEME, 9.0f);		// Hue colorization
color_filter.set_option(RS2_OPTION_MAX_DEPTH, max_depth);
```

closed 2020-06-16 by johnny-wang16