realsenseai/librealsense #6468: Color Compression and Decompression using JpegCompression

FraDePace, 2020-05-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC |
| Language                            |  {C++ }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I'm trying to develop a simple code to compress/decompress a color frame of a D415 camera using JPEG compression.
I'm using the .ccp and .h files of the "compression" folder 
[https://github.com/IntelRealSense/librealsense/tree/master/src/compression](url)

I have a while loop used to iterate over the frames and I use the following code to get access to the color frame:

_auto frames = pipe.wait_for_frames();
auto color = frames.get_color_frame();
unsigned char* color_ptr = reinterpret_cast<unsigned char*>(const_cast<void*>(color.get_data()));_

Then I compress the color frame, allocating a new char array (compressed_color_ptr ):

_unsigned char* compressed_color_ptr = new unsigned char[color.get_data_size()];
JpegCompression js(width, height, RS2_FORMAT_RGB8, 3);
int compressedHeaderSize = js.compressBuffer(color_ptr, color.get_data_size(),compr_color_ptr);_

Finally, I decompress the compressed array with the following code:

_unsigned char* uncompressed_color_ptr = new unsigned char[color.get_data_size()];
js.decompressBuffer(compressed_color_ptr, compressWithHeaderSize, uncompressed_color_ptr);
delete[] uncompressed_color_ptr;
delete[] compressed_color_ptr;_

The code crashes during the decompression. Specifically, it fails the following test (line 189 of
[https://github.com/IntelRealSense/librealsense/blob/master/src/compression/JpegCompression.cpp](url)) :

_if (jpegHeader != 0xE0FFD8FF)
	{ //check header integrity if = E0FF D8FF - the First 4 bytes jpeg standards.
		std::cerr << "Not a JPEG frame, skipping";
		return -1;
}_

Maybe the compression does not wok properly but I don't know how to fix it.
Any help is really appreciated.
Thanks

FraDePace, 2020-05-28:
Still referring to [https://github.com/IntelRealSense/librealsense/blob/master/src/compression/JpegCompression.cpp](url)
I further checked the _JpegCompression::compressBuffer_ function and, after the compression (basically, the while loop), the size of the compressed stream is firstly copied in abuffer called _t_compressedBuf_, then the compressed stream is copied in the same buffer:

_memcpy(t_compressedBuf, &compressedSize, sizeof(int));
memcpy(t_compressedBuf + sizeof(int), data, compressedSize);_

The problem is that when the _JpegCompression::decompressBuffer_ function is called (I assume the first field _t_buffer_ is the compressed stream), the first four bytes of _t_buffer_ are copied in _jpegHeader_ and then checked with:

_if (jpegHeader != 0xE0FFD8FF)
{   //check header integrity if = E0FF D8FF - the First 4 bytes jpeg standards.
    std::cerr << "Not a JPEG frame, skipping";
    return -1;
}_

Since the value of _jpegHeader_ is actually the size of the compressed stream, this condition will never be satisfied.
Perhaps I misunderstood how the functions should be called and used.
Can someone provide me a working example?

Thank you

MartyG-RealSense, 2020-06-18:
Sincere apologies that you did not receive a timely reply to your case.  It has now been assigned to me.  Do you still require assistance please?

FraDePace, 2020-06-18:
@MartyG-RealSense I handled the compression of the color frames using another library (libjpeg-turbo) but I still don't understand how to compress the rgb frames using the JpegCompression.cpp provided by the Intel repository.
Do you have any working examples?

MartyG-RealSense, 2020-06-18:
I  will enquire within Intel about it.   Thanks again for your patience!

MartyG-RealSense, 2020-06-20:
@FraDePace   I thoroughly researched the subject of **jpegcompression.cpp** but could not find any example of it being used for anything in the SDK, whether for example program or internal SDK functions.  I am glad that you achieved jpeg compression by another means, as **jpegcompression** seems to be a dead-end.  Again, I really appreciate your patience during this case!

FraDePace, 2020-06-21:
@MartyG-RealSense thank you for your support. Although it's still not clear the purpose of **jpegcompression.cpp**, I consider this issue as closed. Thanks!

closed 2020-06-21 by FraDePace