realsenseai/librealsense #6469: Failed to recover from firmware update

Jerichoxiii, 2020-05-27:
Hi everyone ,

I am using the Depth Camera D435i and after a failing attempt to update the firmware using real-sense (2.34.0v ) , It did not recover . I tried in all platforms in different computers by using the fw-update and rs-fw-update software I have the following responses. 

 user@user fw-update % ./rs-fw-update -l                                                

connected devices:
1) Name: Intel RealSense USB2 D4xx Recovery, serial number: unknown, update serial number: 123456789123, firmware version: unknown, USB type: unknown

user@user fw-update % ./rs-fw-update -r -f Signed_Image_UVC_5_12_3_0.bin

update to FW: Signed_Image_UVC_5_12_3_0.bin

recovering device: 
Name: Intel RealSense D4xx Recovery, serial number: unknown, update serial number: 123456789123, firmware version: unknown, USB type: unknown

firmware update started

firmware update progress: 8[%]
failed to recover device

If you need more information please let me know.

Thanks in advance
Georgios

MartyG-RealSense, 2020-05-27:
Are you able to get the camera to be detected in the RealSense Viewer program, please?  If so, if you click on the 'More' icon at the top of its options side-panel then there is an option under its menu to install recommended firmware.  This is a firmware update that does not require you to provide a firmware file.  Can you see if the Viewer can commence the firmware update process using this option please?

[image]

Jerichoxiii, 2020-05-27:
Hi MartyG ,

Thank you very much for your fast response .
The printscreen bellow shows the error using RealSense .

[image]

Thanks

MartyG-RealSense, 2020-05-27:
That suggests to me that firmware 5.12.3.0 is already installed (though may be corrupted) and so it will not update because the current recommended version installed by the Viewer is still 5.12.3.0.

Does the menu provide the option to 'update firmware' and manually provide an older firmware file such as 5.12.2.100?

Jerichoxiii, 2020-05-27:
Hi MartyG,
The camera fully recovered with the version 5.12.2.100 but I can not upgrade it to 5.12.3.0 . See the pictures below

[image]

 
[image]

Thanks 
Georgios

MartyG-RealSense, 2020-05-27:
I haven't personally seen this happen before.  If you are not able to update to 5.12.3.0 by going to 'install recommended firmware' in the More menu again then I recommend leaving it on the currently installed firmware until the next firmware version is released.  I don't think you are missing any major features compared to 5.12.3.0, so it should be fine to wait and just dismiss the alert box about the version in the Viewer if it pops up when the program is launched.

Jerichoxiii, 2020-05-27:
That't fine thank you very much

Jerichoxiii, 2020-05-28:
Hi MartyG ,

The camera works Perfect with the Real-Sense with the version 2.34.1 without Errors and Warnings.

Before the issue mention above ,I was able work  on Jetson nano using a Ros application about slam .When I am start running the application  I see the following message and the application does not even start now .

WARNING [547524432256] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

Since it worked well with 5.12.3.0 version and now I have 5.12.2.100 version may this cause the problems.

Is any possibility any library missing or is not configured properly ?

Kind regards
Georgios

MartyG-RealSense, 2020-05-28:
The librealsense SDK version should be matched with a particular ROS wrapper version recommended  in the version listings for the wrapper.  For the 2.34 librealsense SDK, ROS wrapper version 2.2.13 is the best match.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.13

Jerichoxiii, 2020-05-28:
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
[... 64 lines cut ...]
[ INFO] [1590703523.430707238]: Expected frequency for depth = 30.00000
[ INFO] [1590703523.514563810]: Expected frequency for infra1 = 30.00000
[ INFO] [1590703523.569029515]: Expected frequency for infra2 = 30.00000
[ INFO] [1590703523.622460915]: Expected frequency for color = 30.00000
[ INFO] [1590703523.686588710]: setupStreams...
[ INFO] [1590703523.698911981]: insert Depth to Stereo Module
[ INFO] [1590703523.699102560]: insert Color to RGB Camera
[ INFO] [1590703523.699187199]: insert Infrared to Stereo Module
[ INFO] [1590703523.699266628]: insert Infrared to Stereo Module
[ INFO] [1590703523.699608775]: insert Gyro to Motion Module
[ INFO] [1590703523.699692892]: insert Accel to Motion Module
[ INFO] [1590703523.821027160]: SELECTED BASE:Depth, 0
[ INFO] [1590703523.854330447]: RealSense Node Is Up!
 28/05 23:05:23,948 WARNING [546908234112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/05 23:05:23,998 WARNING [546908234112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/05 23:05:24,049 WARNING [546908234112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/05 23:05:24,100 WARNING [546908234112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 28/05 23:06:11,192 WARNING [546908234112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/05 23:06:39,262 WARNING [546908234112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/05 23:06:45,316 WARNING [546908234112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/05 23:06:45,367 WARNING [546908234112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

Jerichoxiii, 2020-05-28:
Hi MartyG
I had the same version but I downloaded and rebuilt and reinstall it just to be sure but It has the same resaults.

I just noticed that error

28/05 23:05:22,743 WARNING [546941804928] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed.
Error type: No data to return (-21).

Thanks

MartyG-RealSense, 2020-05-28:
The **No data to return (-21)** just means that the D435i's IMU probably needs calibrating.  This is done with a Python script.  It is probably unrelated to the control_transfer returned errors though.

https://dev.intelrealsense.com/docs/depth-camera-d435i-imu-calibration

The control_transfer errors have been a common feature of SDK 2.34 in ROS.   It may be best to wait and see if the upcoming 2.35 SDK version fixes them, and just use Librealsense in the meantime.

Jerichoxiii, 2020-05-28:
All right about that ,

About the firmware version, how it is possible to lock or unlock the version ? 
Is there any script I may used to lock the version?

How can I unlock it ?
Thanks

MartyG-RealSense, 2020-05-28:
You could try a 'gold reset' of the camera to reset it to its as-new factory defaults.

https://forums.intel.com/s/question/0D70P000006EFHESA4/realsense-d435i-not-properly-recognized-by-realsenseviewer?language=en_US

I am off-shift for the day now but will be back in 7 hours from the time of writing this to pick up any new comments that you leave below.   Good luck!

Jerichoxiii, 2020-05-28:
No worries thanks for your support

Jerichoxiii, 2020-05-28:
I did this but the device is still locked.

C:\CalibrationToolAPI\2.6.8.0\bin>Intel.Realsense.CustomRW.exe -g
CustomRW for Intel RealSense D400, Version: 2.6.8.0

  Device PID: 0B3A
  Device name: Intel RealSense D435I
  Serial number: 919122073232
  Firmware version: 05.12.02.100

Calibration on device successfully reset to default gold factory settings.

Jerichoxiii, 2020-05-28:
When I am trying to load this file "Signed_Image_UVC_5_11_1_100.bin" I have the following response despite that that is older than 5.12.3.0 version.

Started Firmware Update process
Backing-up camera flash memory
Backup completed and saved as 'C:\Users\jeric\AppData\Roaming\925223051168.20200528_235146.bin'
Requesting to switch to recovery mode
Recovery device connected, starting update

ERROR: UNKNOWN in rs2_update_firmware_cpp(device:0000029C4590FDA0, fw_image:0000029C4E71F060):
Device: 925223051168 is locked for update.
Use firmware version higher than: 5.12.3.0

MartyG-RealSense, 2020-05-29:
That is unusual.  The camera would not usually prevent downgrades to an older firmware unless the firmware had been downgraded 20 times already.  After 20 downgrades, the camera locks downgrading and only allows upgrades to a newer version than the one currently stored in the camera.

Jerichoxiii, 2020-05-29:
Hi MartyG,

So maybe that is the case. Some times the application crashed during the upgrade process and rich that number. I suspect that I have to wait until the new version is released. Is there any way to reset this counter?

Thanks

MartyG-RealSense, 2020-05-29:
I believe that the downgrade block is permanent once it has been done 20 times.  The policy dates back to the original RealSense cameras in 2015.

closed 2020-05-29 by Jerichoxiii