realsenseai/librealsense #6471: Documentation request for memory leak guidance

cdb0y511, 2020-05-28:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400} | 
| Firmware Version                   | (5.12.03) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc.. |
| SDK Version                            |  { 2.33.1 }                          | 
| Language                            |  {C++/C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Request for memory leak  guidance, such as 
Filter type objects should not create or new during the video loop.
#2115 (comment)
More guidance about memory leak is needed, pls list your memory leak situation during use realsense senors. Then the community can track ，disscuss and finish the documentation.
Thx

MartyG-RealSense, 2020-05-28:
Thanks very much for your documentation feature request.  I have applied the Documentation label to the post so it can be kept open and tracked by the RealSense team members responsible for documentation.

MartyG-RealSense, 2022-05-16:
Hi @cdb0y511  As the documentation change request that you made in 2020 is now 2 years old, can you confim whether you still believe that the requested change should be made, please?  Thanks!

closed 2022-05-25 by cdb0y511