realsenseai/librealsense #6472: Align depth and color images without SDK

ShrutheeshIR, 2020-05-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Operating System & Version |   {Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          |  (NA)
| Language                            |  {python}                          | 

### Issue Description
This is not exactly an issue with the SDK, hence please feel free to close it and delete it if it is inappropriate, however, I could not find the solution anywhere else.

I have images taken from an Intel Realsense D415. In this, the RGB images are of resolution 1920x1080 with an FOV of 69.5, while the depth images are of resolution 1280x720 with a FOV of 85.

I merely have these images that have been given to me by my client, and I have to work with merely  the depth and the RGB images, along with the camera's intrinsics.

Is it possible for me to write a custom code that can align the depth and the color images on my own? I tried looking at the source code, and found it to be very complex to understand. Any pointers on how I can approach this will be appreciated.

Thanks

MartyG-RealSense, 2020-05-28:
I researched your question carefully.  Aligning pre-made color and depth files seems to be more complicated if you avoid using Librealsense than if you do the alignment with Librealsense though.

When trying to align image files with the Librealsense SDK instead of live camera streams, the simplest approach that I could find (which is still somewhat complex) is the one in the link below:

https://github.com/IntelRealSense/librealsense/issues/2534

ShrutheeshIR, 2020-05-28:
Hello @MartyG-RealSense, thanks for the link, it does seem to be somewhat helpful, but I am stuck at one place. How does the SDK deal with different FOVs internally? When aligning depth and color images, does it discard the contents of the depth image that are beyond the FOV of the color image?

MartyG-RealSense, 2020-05-28:
If it were live camera streams, then when aligning depth to color then the depth image is aligned to the color FOV.  On the RealSense 400 Series camera models that have a "wide" IR imager, such as the D435, the color imager has a smaller FOV size.  On the D415 model, the same imager component is used for both the RGB and IR sensors.

kafan1986, 2020-05-30:
@ShrutheeshIR Yes, it is possible and not too difficult I would say. You can take a look at the alignment code. The images that you have can be read and represent as Mat by using library like opencv. Once read you can use the existing alignment code, modify the function and call it easily.

ShrutheeshIR, 2020-06-01:
@kafan1986 
If possible, could you link the exact pieces of code I should be looking at?
Thanks

kafan1986, 2020-06-01:
@ShrutheeshIR Can you share how the depth image was saved as image? As it is 16 bit single channel information and image is usually 8 bit and single channel (grayscale) or 3 channel (color)?

ShrutheeshIR, 2020-06-01:
The depth data is stored as a 16 bit single channel and the RGB images are 8-bit, yes. They are stored them as 16-bit numpy (.npy) files, so they can be accessed as is.

ShrutheeshIR, 2020-06-01:
@kafan1986  @MartyG-RealSense also, after aligning the images, what will be my camera intrinsic matrix? Will it be the same as the depth camera matrix, even if the FOV is changed? Please let me know. Thanks

kafan1986, 2020-06-01:
> @kafan1986 @MartyG-RealSense also, after aligning the images, what will be my camera intrinsic matrix? Will it be the same as the depth camera matrix, even if the FOV is changed? Please let me know. Thanks

Alignment process takes one of the frame and aligns to other. You can either align depth to color or color to depth. Although I am not too sure about my knowledge about camera optics, but I believe the reference image to which the other image is being aligned to, won't have any changes, not sure how it impacts the image which is getting aligned.

But I am not sure why do you need the camera intrinsic post alignment process?

ShrutheeshIR, 2020-06-01:
@kafan1986  From what @MartyG-RealSense indicated, I will have to align the depth image to that of the color image, since the color image has a lower FOV.

I need the camera intrinsic post alignment since I am trying to implement my custom mapping algorithm, along with other related odometry algorithms, for which I would require the camera intrinsics.

kafan1986, 2020-06-01:
> @kafan1986
> If possible, could you link the exact pieces of code I should be looking at?
> Thanks

https://github.com/IntelRealSense/librealsense/blob/9f99fa9a509555f85bffc15ce27531aaa6db6f7e/src/proc/align.cpp#L19

ShrutheeshIR, 2020-06-01:
Thanks @kafan1986

kafan1986, 2020-06-01:
> @kafan1986 From what @MartyG-RealSense indicated, I will have to align the depth image to that of the color image, since the color image has a lower FOV.
>
> I need the camera intrinsic post alignment [...]

Yes, the FOV is different but for D415 the difference very small. During alignment process the reference image is untouched and the image being aligned if it has higher FOV, then it will get cropped and if lower FOV then it will get padded with black band on the edges to compensate. For D435 for example, depth image has near 90 degree field of view whereas the color camera is same as D415 around 65 FOV, the aligned colour image to depth image, looks something like this https://github.com/IntelRealSense/librealsense/issues/5571#issuecomment-570046569

ShrutheeshIR, 2020-06-01:
@kafan1986 apologies for the mistake, I checked the FOV and they seemed to be corresponding to D435, and upon reaching out to the client, the camera was indeed a D435 and not D415. I have edited my question accordingly.

How would my new camera intrinsic be in case of D435. Apologies once again

kafan1986, 2020-06-01:
> @kafan1986 apologies for the mistake, I checked the FOV and they seemed to be corresponding to D435, and upon reaching out to the client, the camera was indeed a D435 and not D415. I have edited my question accordingly.
> 
> How would my new camera intrinsic be in case of D435. Apologies once again

I am not undertstanding your question. The camera intrinsic and extrinsic will change across different model of camera D415, D435; rather due to manufacturing process, even different camera of same model, can have slightly different intrinsic and extrinsic values. 

All the cameras are individually factory calibrated and have the camera intrinsic and extrinsic stored for various supported image resolution. You can get the camera intrinsic once in your code and reuse it for all calculations, during alignment. The camera intrinsic and extrinsic won't change until and unless, user performs a self calibration on the device and permanently writes the values to camera's memory.

But you should not use a global hardcoded camera extrinsic and intrinsic value. As it will lead to incorrect alignment.

ShrutheeshIR, 2020-06-01:
@kafan1986 I will try to rephrase my question. 

From the D435 camera, the color camera has a camera intrinsic matrix, and the depth camera has its own intrinsics matrix. After I align the depth and color images (by aligning the depth **to** color images), will the old camera depth camera intrinsics matrix hold true for the new aligned image? (since I am effectively going from an FOV of 90 to 70).

I am asking this question using a fairly simple analogy. If I resize a given image by half in both dimensions, then the ```cx``` and ```cy``` values of the camera intrinsics are also divided by 2. 

Hence I was wondering, if I have to do something like that, or using the original depth camera intrinsics will hold.

kafan1986, 2020-06-01:
Are you planning to use the cx and cy value somewhere else post alignment?

MartyG-RealSense, 2020-06-18:
@kafan1986   Thanks again for your great contributions!

@ShrutheeshIR   Do you require further advice on this case please, or can it be closed?   Thanks!

ShrutheeshIR, 2020-06-21:
@MartyG-RealSense sorry for the late reply. Though I didnt get around to complete implementation, I more or less have an idea on how to do it. Thanks for the help!

closed 2020-06-21 by ShrutheeshIR

MartyG-RealSense, 2020-06-21:
Thanks so much for the update!

vneseresearcher, 2021-09-02:
> @MartyG-RealSense sorry for the late reply. Though I didnt get around to complete implementation, I more or less have an idea on how to do it. Thanks for the help!

Hi bro,have you done it yet?

emily-stueckelmaier, 2022-08-05:
Hello, do you know if there is a similar code for the align_image() method for python? Or how I can use this method in python?
Thanks in advance!

> > @kafan1986
> > If possible, could you link the exact pieces of code I should be looking at?
> > Thanks
> 
> https://github.com/IntelRealSense/librealsense/blob/9f99fa9a509555f85bffc15ce27531aaa6db6f7e/src/proc/align.cpp#L19

MartyG-RealSense, 2022-08-05:
Hi @emily-stueckelmaier  The RealSense SDK has a Python depth to color alignment example program called **align_depth2color.py** at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

MartyG-RealSense, 2025-05-23:
Thanks so much @mlouielu for sharing your project with the RealSense community!