realsenseai/librealsense #6473: low resolution

SeanDJW, 2020-05-28:
| | |
|---|---|
|**librealsense**|2.34.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D415|
|**Serial Number**|822512060220|
|**Firmware Version**|05.12.03.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|825513023968|
|**Firmware Update Id**|825513023968|

Please provide a description of the problem

SeanDJW, 2020-05-28:
I am using monitor as 4k resuolution, when i open realsense viewer or depth qauality viewer the scence resolution goes to around 800*600 resolution.. I want to solve this errors

MartyG-RealSense, 2020-05-28:
I tried to replicate what you are experiencing on my own monitor.  I found that the Viewer's resolution was most likely to be distorted if it was launched when 'Fullscreen' mode is enabled.  If I used Windowed mode and scaled the window with the arrows on the corners of the window then I did not experience problems.

If the Viewer is filling your entire 4K screen with a distorted 800x600 view (suggesting that it is in Fullscreen mode), please press the F8 key to switch to Windowed mode so that the Viewer window re-scales its display resolution according to the size of the window (you can see what the Viewer resolution is changing to in the log at the bottom of the Viewer window).

If Fullscreen was enabled in the Viewer settings, pressing F8 should set Fullscreen to disabled so that it is in Windowed mode next time you launch the Viewer.

MartyG-RealSense, 2020-06-05:
@SeanDJW   Do you require further assistance please or can this case be closed?   Thanks!

closed 2020-06-18 by MartyG-RealSense