realsenseai/librealsense #6474: "Intel RealSense D400 Advanced Mode " USELESS

mingyangpeng, 2020-05-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D415 } | 
| Firmware Version                   | (05.12.03.00) | 
| Operating System & Version |   {Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC   |
| SDK Version                            |  { legacy / 2.<34>.<0> }                          | 
| Language                            |  {c++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
1. I got a Custom.json file from realsense-viewer , but I can't use it in My C++ application,
and the example https://dev.intelrealsense.com/docs/intel-realsense-d400-advanced-mode didn't work and not show how to use external files *.json.
2.(cmake .. -DBUILD_EXAMPLES=true -DBUILD_RS400_EXTRAS=true) error

3. My code :
I refer to the document (librealsense/doc/rs400/rs400_advanced_mode.md)  
`std::ifstream file(camera_config_path); 
        std::string str((std::istreambuf_iterator<char>(file)), std::istreambuf_iterator<char>());
        auto advance_mode_dev = device.as<rs400::advanced_mode>();
        advance_mode_dev.load_json(str);  `

but my code also useless ,can't control the camera.
Please give me some advices!

MartyG-RealSense, 2020-05-28:
I hope that the simple C++ script for json loading in the link below, and the comments beneath it on the linked page, will provide useful guidance if you have not seen it already.

https://github.com/IntelRealSense/librealsense/issues/3037#issuecomment-452732168

mingyangpeng, 2020-05-29:
Hello, thank you for your reply,I tried the code:
`rs2::pipeline p;
auto prof = p.start();
rs2::device dev = prof.get_device();
auto advanced = dev.as<advanced_mode>();
advanced.load_json(str);`
It doesn't  work, for example, in the json file ,My settings are as follows:
    `    "stream-fps": "30",
    "stream-height": "480",
    "stream-width": "640"`
but, I try to show color image ,it's resolution: 1280*720 (not my want),
besides,the demo code: " str ",what is that ? 
please give me the complete instance code,thank you again!

mingyangpeng, 2020-05-29:
I get the current values of advanced mode controls to a JSON file,via the following code:
   
   rs2_error* e = nullptr;
    std::shared_ptr<rs2_device> _dev;
    _dev = device.get();
    rs2_raw_data_buffer*  _buffer = rs2_serialize_json(_dev.get(), &e); 
    const char* full_file = "advance.json";
    FILE *ft = fopen(full_file , "w"); 
    int _buffer_size = rs2_get_raw_data_size(_buffer, &e);
    const unsigned char* raw_data =  rs2_get_raw_data(_buffer, &e); 
    fwrite( raw_data, sizeof(unsigned char), _buffer_size, ft); 
    fclose(ft);

advance.json file following:
 {
    "aux-param-autoexposure-setpoint": "400",
    "aux-param-colorcorrection1": "0.461914",
    "aux-param-colorcorrection10": "-0.553711",
    "aux-param-colorcorrection11": "-0.553711",
    "aux-param-colorcorrection12": "0.0458984",
    "aux-param-colorcorrection2": "0.540039",
    "aux-param-colorcorrection3": "0.540039",
    "aux-param-colorcorrection4": "0.208008",
    "aux-param-colorcorrection5": "-0.332031",
    "aux-param-colorcorrection6": "-0.212891",
    "aux-param-colorcorrection7": "-0.212891",
    "aux-param-colorcorrection8": "0.68457",
    "aux-param-colorcorrection9": "0.930664",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "False",
    "controls-autoexposure-manual": "33000",
    "controls-color-autoexposure-auto": "False",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-depth-white-balance-auto": "False",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-amplitude-factor": "0",
    "param-autoexposure-setpoint": "400",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
[... 112 lines cut ...]
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "480",
    "stream-width": "640"
}

the both *.json files are different , I want to control format resolution and others via json file,but It didn't work .
Is my description clear？

Hope to get a reply！！

MartyG-RealSense, 2020-05-29:
When RealSense SDK 2.0 first launched in 2018, the FPS and resolution could not be set in a json because the pipeline was overriding the json settings for these values with its own configuration.  I am not aware that this has changed since then.

https://github.com/IntelRealSense/librealsense/issues/1229#issuecomment-368329980
   
If you are writing your own C++ script, you can set the FPS and resolution for each stream by configuring **cfg** (configuration).  This causes the camera to use your settings instead of the camera's default settings.

A good example of defining a cfg configuration is in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md

The cfg instructions should always be placed **before** the pipe start instruction.

Basically, you:

1.  Tell the script that you want to use cfg:

    **rs2::config cfg;**

2.  Define the streams that your program will be using and their resolution, format and FPS speed, like this example for a color stream:

**cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);**

3.  Tell the pipeline to use your cfg configuration instead of the camera's default configuration by putting the word cfg inside the brackets of the pipe start instruction:

 **pipe.start(cfg);**

mingyangpeng, 2020-05-29:
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
I used it before，Thank you for telling me all this！

closed 2020-05-29 by mingyangpeng