realsenseai/librealsense #6475: python flask - RuntimeError: stop() cannot be called before start()

Tomblarom, 2020-05-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Raspbian GNU/Linux 10 (buster)  | 
| Kernel Version (Linux Only)    |  4.19.97-v7l+ | 
| Platform                                 | Raspberry Pi 4 - 4GB  |
| SDK Version                            | - | 
| Language                            |  python3.6 / python3.7                          | 
| Segment			| - | 

### Issue Description
I'm running flask server to stream my video feed (color and depth). Since I'm new to python, I do not really know how to handle `pipeline.start(config)` and `pipeline.stop()` in my case. 

Calling `/rgb` works fine, but calling `/rgbd` or another `/rgb` in a new tab, breaks the running stream and throws `RuntimeError: stop() cannot be called before start()`. How to maintain opened streams?

Is it possible to stream in one resolution and take a momentary picture in a higher resolution?

The script looks like this:

```
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 15)
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 15)

align_to = rs.stream.color
align = rs.align(align_to)

app = Flask(__name__)

def gen_rgb():
    try:
        pipeline.start(config)
        while True:
            frames = pipeline.wait_for_frames()
            color_frame = frames.get_color_frame()
            if not color_frame:
                continue
            color_image = np.asanyarray(color_frame.get_data())
            yield (b"--frame\r\n" b"Content-Type: image/jpeg\r\n\r\n" + cv2.imencode(".jpeg", color_image)[1].tobytes() + b"\r\n")
    finally:
        pipeline.stop()
        
def gen_rgbd():
    try:
        pipeline.start(config)
        while True:
            frames = pipeline.wait_for_frames()
            aligned_frames = align.process(frames)
            depth_frame = aligned_frames.get_depth_frame()
            if not depth_frame:
                continue
            depth_image = np.asanyarray(depth_frame.get_data())
            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.16), cv2.COLORMAP_JET)
            yield (b"--frame\r\n" b"Content-Type: image/jpeg\r\n\r\n" + cv2.imencode(".jpeg", depth_colormap)[1].tobytes() + b"\r\n")
    finally:
        pipeline.stop()

@app.route("/")
def index():
    return redirect("rgb", code=303)

@app.route("/rgb")
def rgb():
    return Response(gen_rgb(), mimetype="multipart/x-mixed-replace; boundary=frame")

@app.route("/rgbd")
def rgbd():
    return Response(gen_rgbd(), mimetype="multipart/x-mixed-replace; boundary=frame")
    
if __name__ == "__main__":
    try:
        app.debug = True
        app.run("0.0.0.0")
    except KeyboardInterrupt:
        pass
    finally:
        pass
```

MartyG-RealSense, 2020-05-29:
Another RealSense user has had problems with using a flask server with Pyrealsense2.  A commentator on that discussion thought that it might be related to a problem with 'pickling':

https://stackoverflow.com/questions/60514606/how-to-send-python-pyrealsense2-object-to-another-server

A potential alternative means of streaming RealSense if the flask server does not work out is a recently created RealSense plugin for **gstreamer**:

https://github.com/WKDSMRT/realsense-gstreamer

soarwing52, 2020-05-29:
https://github.com/soarwing52/Remote-Realsense

i've made my project on Raspberry Pi 4 with D435

and I wrote it with cv2.imencode('.jpg', images)[1].tobytes() and send bytes to Flask

the thing is the two must be multi-processing, therefore I use a mp.Pipe to transport the image array/bytes

and you can write one 
except Exceptions as e:
    print(e)
to check why your rs.pipeline didnt start successfully

my more information can be seen here
https://jaythecheyi.home.blog/2019/11/05/roadmap-real-sense-application/
https://jaythecheyi.home.blog/2019/11/01/raspberry-pi-realsense-usage-of-class-pyqt/

MartyG-RealSense, 2020-05-29:
Great to see you again @soarwing52 - thanks so much for the details about your flask project!

MartyG-RealSense, 2020-06-18:
@Tomblarom  Do you require further assistance please after the kindly provided contribution of @soarwing52 or can the case be closed?   Thanks!

closed 2020-06-25 by MartyG-RealSense