realsenseai/librealsense #6478: the camera is broken?

madtly, 2020-05-30:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

Backend in rs2_create_device(info_list:0B2D3CA8, index:0):
MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070005: "Access is denied."

MartyG-RealSense, 2020-05-30:
If there is an access denial on Windows, please check the camera privacy settings in Windows to make sure that access to the camera hardware and to camera applications is set to "On".  Instructions for how to do this are in the link below.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

madtly, 2020-05-30:
thanks it works

MartyG-RealSense, 2020-05-30:
Great news!  I'm very pleased to hear it.

closed 2020-05-31 by dorodnic