realsenseai/librealsense #6482: Latest Release Fails to Configure

bobdavies2000, 2020-05-31:
On Windows 10 the latest librealsense is failing on the CMake configure.  Here is the log from the failure:

[image]

I will back off to prior version until an update is available.  I tried the download several times to make sure it wasn't some network glitch.

dorodnic, 2020-05-31:
Hi @bobdavies2000 
Yes, we apologise for the issue. There was a problem in the release process, should be addressed soon

ev-mp, 2020-05-31:
@bobdavies2000 hello, it should be fixed by now.
Please check the latest and update

RealSenseSupport, 2020-06-22:
Hi @bobdavies2000 Do you need further help on this issue? If we don't hear from you in 7 days, this issues will be closed.

closed 2020-07-06 by dorodnic