realsenseai/librealsense #6486: Convert frame to cv in NodeJS

Alex-OPTIM, 2020-06-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |  | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | nodejs | 
| Segment			|  Robot | 

### Issue Description
Cannot get image in OpenCV format (cv.Mat). I'm following example on [opencv-wrapper](https://dev.intelrealsense.com/docs/opencv-wrapper). When I'm running the code below I'm getting a black screen.

    const rs2 = require('node-librealsense');
    const cv = require('opencv4nodejs');
    
    const colorizer = new rs2.Colorizer();  // This will make depth image pretty
    const pipeline = new rs2.Pipeline();  // Main work pipeline of RealSense camera
    pipeline.start();  // Start camera
    
    const frameset = pipeline.waitForFrames(); 
    const color = frameset.colorFrame;  // Get RGB image
    
    const matFromArray = new cv.Mat(color.height, color.width, cv.CV_8UC3 , color.getData());
    
    cv.imshow('pic', matFromArray);
    
    cv.waitKey(0);
    rs2.cleanup();

MartyG-RealSense, 2020-06-01:
Hi @Alex-OPTIM   Are you able to confirm that your Node.Js setup is working successfully by having tested it with Node.JS wrapper example programs please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs/examples

Alex-OPTIM, 2020-06-02:
Yes, we have 2 files in the same project: the code above and `nodejs-capture.js` from the examples. Running `nodejs-capture.js` has no issues.

MartyG-RealSense, 2020-06-02:
I apologise for the delay in responding further, as I was researching your case more.  I wonder if you could try changing the CV format from CV_8UC3 to **CV_32FC1**

MartyG-RealSense, 2020-06-02:
I apologise for the delay in responding further, as I was researching your case some more.

It looks as though you are trying to display an RGB image.  Is that correct, please?  If so, I wondered why you need to include a colorizer for depth information.  Or are you planning to add a depth display later once the RGB image is working?  Thanks!

Alex-OPTIM, 2020-06-02:
Thank you for researching our case!

I'm not too sure on question why I need to include a coloriser. Basically we need a colour picture for sending it to Yolo `net` in opencv4nodejs ([similar to this](https://github.com/justadudewhohacks/opencv4nodejs/blob/master/examples/dnnDarknetYOLORealTimeObjectDetection.js)).

Later, we are going to use depth image for detecting distance and geometrical properties of the object. We are have no plans on displaying the depth image.

Alex-OPTIM, 2020-06-02:
Also colourised picture is what I thought we need as per the C++ example:

    //Get each frame
    rs2::frame color_frame = frames.get_color_frame();
    
    // Creating OpenCV Matrix from a color image
    Mat color(Size(640, 480), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);

MartyG-RealSense, 2020-06-02:
My level of Node.js programming knowledge is admittedly limited.  I believe though that a colorizer is for converting a depth image into an RGB image.  So it probably wouldn't be needed if you were just grabbing a color frame from the camera.  The link below explains the concept of a colorizer:

https://github.com/IntelRealSense/librealsense/issues/1223#issuecomment-368219350

Alex-OPTIM, 2020-06-03:
I think I mixed up terms `.colorFrame` with `colorizer` earlier. We don't use `rs2.Colorizer()` at the moment and line 4 in the code snippet above could be deleted:

~~`const colorizer = new rs2.Colorizer();  // This will make depth image pretty`~~

What we do is we get a coloured image from the camera by calling `colorFrame = frameset.colorFrame` and its `.getData()` method returns an array of `Uint8Array`. We pass this array as a `fillValue` into cv.Mat as per [Mat documentation](https://justadudewhohacks.github.io/opencv4nodejs/docs/Mat/). We expect to see an image in Mat object, but it doesn't work.

I guess a question here, why does it work in C++? Are we missing an extra step on `colorFrame` in NodeJS before we can pass it to `cv.Mat()` constructor?

MartyG-RealSense, 2020-06-04:
I note that in the **nodejs-capture** wrapper example, colorFrame is called with **const color = frameset.colorFrame;**

https://github.com/IntelRealSense/librealsense/blob/d04376ad56581b3094d49ef71c35fe231ac8e4e7/wrappers/nodejs/examples/nodejs-capture.js#L32

Alex-OPTIM, 2020-06-05:
This is how we call it too. Line 9 in the code snipped above (first question).

    const color = frameset.colorFrame;  // Get RGB image

and next, we are getting data out of the object by calling `color.getData()`

MartyG-RealSense, 2020-06-05:
Could you try inserting this in the line before rs2.cleanup if you have not done so already, please?

**color.getData();**

So that it looks like this:

cv.waitKey(0);
color.getData();
rs2.cleanup();

Alex-OPTIM, 2020-06-09:
We got it working (eventually) with the following code:

    const rs2 = require('node-librealsense');
    const cv = require('opencv4nodejs');

    const pipeline = new rs2.Pipeline();  // Main work pipeline of RealSense camera
    pipeline.start();  // Start camera

    const frameset = pipeline.waitForFrames();  // Get a set of frames
    const color = frameset.colorFrame;  // Get RGB image

    const arrayBuffer = new ArrayBuffer(color.dataByteLength);
    color.getData(arrayBuffer);

    // converting ArrayBuffer to NodeJS Buffer
    const view = new DataView(arrayBuffer);
    const nodeJsBuffer = Buffer.alloc(arrayBuffer.byteLength)
    for (let i = 0; i < arrayBuffer.byteLength; i++) nodeJsBuffer[i] = view.getUint8(i);

    const mat = new cv.Mat(nodeJsBuffer, color.height, color.width, cv.CV_8UC3);
    cv.imshow('pic', mat);

    cv.waitKey(0);
    rs2.cleanup();

As you can see we had to use `buffer`. However, I don't understand why RealSense library for NodeJs doesn't use NodeJS Buffer class as an argument in `color.getData()`? So we had to convert `ArrayBuffer` to NodeJS `Buffer`.

Also original approach didn't work due to `color.getData()` returns UInt8Array, which isn't typical `Array` of numbers in NodeJS (`Array.isArray(uInt8Array)` returns `false`). When cv.Mat required type is `number[]`. However when we did:

    const uInt8Array = color.getData();
    const mat = new cv.Mat(color.height, color.width, cv.CV_8UC3, Array.from(uInt8Array));

`cv.Mat` constructor throws an error (I don't remember it exactly). So we stay with the buffer solution for now.

closed 2020-06-09 by Alex-OPTIM

MartyG-RealSense, 2020-06-09:
Brilliant, thanks for sharing your solution with the community!