realsenseai/librealsense #6489: Librealsense C# interface

bobdavies2000, 2020-06-01:
I have configured librealsense to include the C# bindings and the wrappers projects for C# are present after CMake.  But every one of the tutorial examples fails with "'realsense2' dll not found" on the first reference to the librealsense context.  This has worked in all the previous releases so what am I doing wrong?

Bob Davies

bobdavies2000, 2020-06-01:
I should have added that this is with the latest version that was just released.

dorodnic, 2020-06-02:
Hello @bobdavies2000 
Please check that `realsense2.dll` is near the executable / working directory / PATH environment variable. 
Since we are using P/Invoke for .NET wrapper, the SDK was always requiring two DLLs - `Intel.RealSense.dll` (.NET assembly) and `realsense2.dll` (the native dependency) 
Perhaps something else has changed in the environment?

bobdavies2000, 2020-06-02:
Sergey - thanks for the note.  The 2.35.0 version of librealsense is working for my Windows 10 application.  I was trying to alert you to the failure of the C# tutorials in the librealsense distribution.  The first C# tutorials are all built in the librealsense/Build/Debug directory along with both libraries - realsense2.dll and intel.realsense.dll.  The current directory is automatically in the path so I am not sure what I could have changed to make this fail.  I researched the dependencies in both dll's and there doesn't appear to be any references that I don't have (there is a "libc" reference in intel.realsense.dll that I don't see on my system but it can't be relevant because my application is working.)

Can you confirm that the C# tutorials run fine on your system?  If so, please let me know and close this issue.  The only change I make to the CMake is to add the C# bindings.

Bob Davies

RealSenseSupport, 2020-07-01:
@bobdavies2000 Do you mean that the examples fails for debug build? I didn't get failure issue for release build on my setup. Could you please try release build? Thanks!

bobdavies2000, 2020-07-01:
Sergey - yes, it is only the debug version that fails.  Release version works fine.  All my tests are run under Windows 10 with the latest available release of librealsense2.  The debug version expects the Realsense2.dll which is not available in the Debug directory - it is called Realsense2d.dll there.  Easy fix - make the Debug version of the C# examples look for Realsensed.dll.  

There is a long-standing tradition of adding a "d" to the DLL for the debug version.  Is it necessary when only the Debug DLL is its own directory?  I think not and this is a good example where the tradition just gets in the way.  Time to give up the tradition.

BTW: L515 has the best data of any Intel camera.  There are still problems - lots of holes, lower resolution, missing IMU - but it is accurate.  The D4xx cameras are just off by a lot.  I am hoping for better things when my D455 arrives later this month.

Bob Davies

RealSenseSupport, 2020-07-03:
@bobdavies2000 Thanks for your suggestion about debug build! Will feedback this to our engineering team.

dorodnic, 2020-07-06:
Hi @bobdavies2000 
Appreciate the feedback, we will make sure to address it

ev-mp, 2020-07-27:
@bobdavies2000 , the "d" suffix for Debug build is incorporated into v2.37.0 (https://github.com/IntelRealSense/librealsense/pull/6872).

RealSenseSupport, 2020-08-13:
@bobdavies2000  Did you get chance to try v2.37.0? Any other questions about this? Looking forward to your update.

bobdavies2000, 2020-08-13:
This is working.  Thanks for your work.  I have been using the latest version since it came out in July.

closed 2020-08-13 by bobdavies2000