realsenseai/librealsense #6490: Save PNG from rs-capture.cpp

CarlSeeTwo, 2020-06-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | 5.12.3.0 | 
| Operating System & Version | Windows 10 | 
| Kernel Version (Linux Only)    |  N/A  | 
| Platform                                 | PC |
| SDK Version                            | 2.0 | 
| Language                            | C++                 | 
| Segment			|                       | 

### Issue Description
I am trying to alter "rs-capture.cpp" to also save a PNG file of the depth information.  I am trying to modify the code from "rs-save-to-disk.cpp" to achieve this, but I am either getting errors, or if-statements are being skipped and I am unsure why.

Here is the current version of the code I have:
`// Capture Example demonstrates how to
// capture depth and color video streams and render them to the screen
int main(int argc, char* argv[]) try
{
    // Declare depth colorizer for pretty visualization of depth data
    rs2::colorizer color_map;

    auto depth_units = rs2::context().query_devices().front()
        .query_sensors().front().get_option(RS2_OPTION_DEPTH_UNITS);

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Start streaming with default recommended configuration
    pipe.start();

    // Capture 30 frames to give autoexposure, etc. a chance to settle
    for (auto i = 0; i < 30; ++i) pipe.wait_for_frames();

 /*
    // Wait for the next set of frames from the camera. Now that autoexposure, etc.
// has settled, we will write these to disk
    for (auto&& frame : pipe.wait_for_frames())
    {   
        // We can only save video frames as pngs, so we skip the rest
        if (auto vf = frame.as<rs2::video_frame>())
        {
            auto stream = frame.get_profile().stream_type();
            // Use the colorizer to get an rgb image for the depth stream
            if (vf.is<rs2::depth_frame>()) { 
                vf = color_map.process(frame);
            }

            // Write images to disk
            std::stringstream png_file;
            png_file << "CMD-" << vf.get_profile().stream_name() << ".png";
            stbi_write_png(png_file.str().c_str(), vf.get_width(), vf.get_height(),
                vf.get_bytes_per_pixel(), vf.get_data(), vf.get_stride_in_bytes());
            std::cout << "Saved " << png_file.str() << std::endl;

        }
    }
*/
    rs2::frame data = pipe.wait_for_frames(); // Wait for next set of frames from the camera

    rs2::frame depth = data.as<rs2::frameset>().get_depth_frame(); // Find and colorize the depth data
    rs2::frame color = data.as<rs2::frameset>().get_color_frame(); // Find the color data
    save_frame_depth_data("depth", depth, depth_units);

    auto vf = data.as<rs2::video_frame>();
    // Write images to disk
    std::stringstream png_file;
    png_file << "CMD-" << vf.get_profile().stream_name() << ".png";
    stbi_write_png(png_file.str().c_str(), vf.get_width(), vf.get_height(),
        vf.get_bytes_per_pixel(), vf.get_data(), vf.get_stride_in_bytes());
    std::cout << "Saved " << png_file.str() << std::endl;

    return EXIT_SUCCESS;
}`

ERROR:
`Exception thrown at 0x00007FFCED429709 in capture.exe: Microsoft C++ exception: std::runtime_error at memory location 0x000000DA8D94E5B0.
Exception thrown at 0x00007FFCED429709 in capture.exe: Microsoft C++ exception: [rethrow] at memory location 0x0000000000000000.
Exception thrown at 0x00007FFCED429709 in capture.exe: Microsoft C++ exception: rs2::error at memory location 0x000000DA8D94E670.`

Can anybody help me understand this better?

MartyG-RealSense, 2020-06-02:
There is a possible long way round of getting the result you are aiming for (saving a PNG with **rs-capture** code).

First, use an adaptation of rs-capture to save the data in **.bin** binary file format.

https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-380851655

Secondly, convert the .bin file into a **.png**

https://github.com/IntelRealSense/librealsense/issues/4740#issuecomment-529819400

MartyG-RealSense, 2020-06-18:
Hi @CarlSeeTwo  Do you require further assistance please, or can this case be closed?   Thanks!

CarlSeeTwo, 2020-06-20:
I suppose it can be closed.  I never really figured out some of the C++ calls, and why creating an rvalue object wouldn't allow me to call some of the functions as when I create an Ivalue object, but I have pivoted away from this regardless and am trying something different.

Thank you for your help.

closed 2020-06-20 by CarlSeeTwo