realsenseai/librealsense #6491: D435i, Viewer V2.33.1 Win10, When Motion Modiule ON, Failure  Prompted

AllanYingYang, 2020-06-01:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|948122073804|
|**Firmware Version**|05.12.03.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|950323050423|
|**Firmware Update Id**|950323050423|

Please provide a description of the problem

MartyG-RealSense, 2020-06-01:
Hi @AllanYingYang   Can you provide the error messages that you see, please?  Thanks!

AllanYingYang, 2020-06-01:
Here is the screen photo. 

发自我的iPhone

AllanYingYang, 2020-06-01:
Hi MartyG-RealSense,

Does D435i support Motion Module ? 

I got error when do the following:

cfg.enable_stream(RS2_STREAM_POSE); 
 pipe.start(cfg);

Thanks.

Allan

MartyG-RealSense, 2020-06-01:
@AllanYingYang   Thanks for the picture but it did not appear.

Yes, D435I does support Motion Module.  It refers to the component of the D435i called the IMU, which contains a gyroscope and accelerometer.

RS2_STREAM_POSE is a function used by the T265 Tracking Camera though.  T265 has the ability to use 6DOF pose data built in, but the D435i does not.

https://github.com/IntelRealSense/librealsense/issues/3070#issuecomment-454689295

There are workarounds for getting pose with D435 though.

https://github.com/IntelRealSense/librealsense/issues/6038

AllanYingYang, 2020-06-01:
Here is the screen shot。

 

 
 
 
#3070 (comment)

AllanYingYang, 2020-06-01:
Here is the screen shot。

MartyG-RealSense, 2020-06-01:
Sorry, the picture is not appearing.  You will probably have to wait until you can use a computer to post it.

AllanYingYang, 2020-06-01:
here is [ www.dynavr.net/falure.jpg](http://www.dynavr.net/falure.jpg)

MartyG-RealSense, 2020-06-01:
Thanks very much!   Yes, this is the same error box I described earlier in this discussion,

https://github.com/IntelRealSense/librealsense/issues/6491#issuecomment-637061047

AllanYingYang, 2020-06-01:
OK, THANKS.
I  close this issue.

closed 2020-06-01 by AllanYingYang