realsenseai/librealsense #6496: New extended cam sync modes

sam598, 2020-06-02:
The latest release has new mode for camera synchronization, but the features do not seem to be fully documented.

What is the difference between `INTERCAM_SYNC_SLAVE` and `INTERCAM_SYNC_FULL_SLAVE`?

How does `INTERCAM_SYNC_MAX` work? If `inter_cam_sync_mode ` is set to 33, does that mean the camera is expecting a sync pulse every 30 frames?

What kind of sync pulse is required for the new modes? Is it still a 100 microsecond pulse at 1.8V?

agrunnet, 2020-06-02:
Sorry, about that. The white paper documentation was awaiting the FW release, and being prepped for publication. 
The new camera synchronization is called "Genlock".  It will allow Global Shutter depth cameras to be slave to external pulses.  
It will be out this week.

sam598, 2020-06-02:
Thanks, looking forward to it!

closed 2020-06-02 by sam598

sam598, 2020-06-08:
Hi @agrunnet is there an update on this documentation?