realsenseai/librealsense #6501: error issue

lg93lggt, 2020-06-02:
| | |
|---|---|
|**librealsense**|2.29.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|912112070809|
|**Firmware Version**|05.11.15.00|
|**Advanced Mode**|NO|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|919423050682|
|**Firmware Update Id**|919423050682|

Please provide a description of the problem

lg93lggt, 2020-06-02:
I used usb3.0 wire(not original), but it shown as usb2.0.

MartyG-RealSense, 2020-06-02:
You are using a USB 3.0 cable that is not the one supplied with the camera but is a longer one that you bought yourself?   Is that correct, please?

lg93lggt, 2020-06-02:
> Hi @lg93lggt You are using a USB 3.0 cable that is not the one supplied with the camera but is a longer one that you bought yourself. And the D435i is showing as a USB 2 connection when it is plugged into a USB 3 port on the computer. Is that correct, please?

Right

MartyG-RealSense, 2020-06-02:
When using your own cable, it is important to choose a high quality one because of the amount of RealSense data that can pass through the cable.  So an inexpensive cable designed for simple functions such as charging up a device's battery may provide reduced performance.  The longer the cable is, the worse the performance will be and the camera may not be able to connect at all in the worst cases.

If you believe that you have a high quality cable and are certain that it is a USB 3 port that you are plugging the camera into and not a USB 2 port, then the next thing that you could try is inserting the camera into the USB port with a quick and firm motion, and not a slow and soft insertion.

lg93lggt, 2020-06-02:
How can I buy another original one? It may be lost.

MartyG-RealSense, 2020-06-02:
High quality USB 3 cables that are not supplied by Intel can work, though it is risky because without recommendations from other RealSense users it is hard to guarantee that a particular cable will work.

My personal recommendation of supplier for cables not supplied by Intel is a company called **Newnex**, as they provide industrial grade cables, have dedicated RealSense cable support and provide video proof that their cables work:

https://www.newnex.com/realsense-3d-camera-connectivity.php
https://www.youtube.com/watch?v=GLQgR1jT04M

In the recent May 2020 edition of the data sheet document for the 400 Series cameras, Newnex is also now officially recommended as a source for USB cables.

[image]

Another supplier that some RealSense users have recommended in the past for 400 Series camera is **OCP**:

https://www.ocp.com/

lg93lggt, 2020-06-03:
> High quality USB 3 cables that are not supplied by Intel can work, though it is risky because without recommendations from other RealSense users it is hard to guarantee that a particular cable will work.
> 
> My personal recommendation of supplier for cables not supplied by Intel is a company called **Newnex**, as they provide industrial grade cables, have dedicated RealSense cable support and provide video proof that their cables work:
> 
> https://www.newnex.com/realsense-3d-camera-connectivity.php
> https://www.youtube.com/watch?v=GLQgR1jT04M
> 
> In the recent May 2020 edition of the data sheet document for the 400 Series cameras, Newnex is also now officially recommended as a source for USB cables.
> 
> [image]
> 
> Another supplier that some RealSense users have recommended in the past for 400 Series cameras is **OCP**:
> 
> https://www.ocp.com/

Thanks!

MartyG-RealSense, 2020-06-18:
Hi @lg93lggt   Do you require further assistance with this case please, or can it be closed?   Thanks!

closed 2020-06-25 by MartyG-RealSense