realsenseai/librealsense #6505: Announcement:  RealSense 400 Series firmware 5.12.5 now available for download

MartyG-RealSense, 2020-06-03:
Hi everyone,

RealSense 400 Series firmware **5.12.5** is now available as an individual download on the firmware release page.

https://dev.intelrealsense.com/docs/firmware-releases

enclarify, 2020-06-04:
This release appears to have bugs that affect at least the D415 cameras. Since the Sunday release my camera has been taken out. I can no longer reflash, hardware reset or any other remediation to make the camera functional again. Errors include:

/dev/video5 is no device Last Error: Resource temporarily unavailable 
xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Invalid or incomplete multibyte or wide character
uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024)

I am running in Ubuntu 18.04. Also tried a brand new D415 and it's useless with the new drivers or flash image. I have tried every remediation ever mentioned in github issue history, mailing lists, etc but no success.

MartyG-RealSense, 2020-06-04:
Hi @enclarify   I sincerely apologise for all the trouble that you have had and have referred your problem to a RealSense team member with knowledge of this situation so that they can provide a response.

RealSenseSupport, 2020-06-04:
Hi @enclarify Could you please send an email to sys_support@intel.com about this case? Let's resolve this through email. Sorry for the inconvenience! And appreciate your cooperation!

enclarify, 2020-06-09:
Hi, I did sent a message to that support address but haven't gotten a response. I used the email address I purchased with, not the one I use on GitHub. 

Even switching to the USB backend doesn't solve my issues. I have been able to flash the camera (with lots of errors) and any time I try to change a settings I get errors.

MartyG-RealSense, 2020-06-09:
I will highlight your message about not receiving a response to the appropriate Intel staff.

enclarify, 2020-06-09:
Nothing in spam

MartyG-RealSense, 2020-06-09:
@enclarify   Okay, thank you very much for checking.  Your message has been passed on.   Your patience is appreciated!

RealSenseSupport, 2020-06-09:
Hello,
I double checked the email box; I did find the email you mentioned. Anyway, I got this email now, let’s go from here.

I apologize for the inconvenience!

At first 15 minutes of v2.35.0 was posted, the FW was a wrong version. So I believed you downloaded it during this time.
15 minutes later, our development team pulled it back and uploaded a correct version.
Unfortunately, the camera cannot be recovered at your side. Please provide your shipping address, I will send a new D435i to you.

Please provide the following info. These are required by Intel shipping department.
Name
Address
Email
Phone number

I will also provide our FedEx account to you. Please ship the bricked camera back to me using this account.

Sorry again for the trouble!

Thank you very much for using RealSense camera!

Appreciate your understanding!

Thanks,
Intel Corporation
Kevin Zhao

closed 2020-06-12 by MartyG-RealSense

enclarify, 2020-06-19:
@MartyG-RealSense I am still having issues and it's been about a week since I have gotten a response from sys_support. Any chance you can see if they can check their support queue?

MartyG-RealSense, 2020-06-19:
@RealSenseSupport will have seen your new comment now that you have posted it.