realsenseai/librealsense #6506: I can not find header file of function: Intel::RealSense::CaptureManager::QueryFrameByTimeStamp

lml0001, 2020-06-03:
how to use this function or namespace
Intel::RealSense::CaptureManager::QueryFrameByTimeStamp

MartyG-RealSense, 2020-06-03:
Hi @lml0001    **QueryFrameByTimeStamp** is a function of the old 2016 RealSense SDK for Windows and not the current RealSense SDK 2.0.  

https://software.intel.com/en-us/node/724699

If you are using the current RealSense SDK 2.0, timestamp querying information can be found in the link below:

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md

The timestamps that can be queried using **metadata** are covered in this section of the above documentation:

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#employing-metadata-attributes-in-demos

MartyG-RealSense, 2020-06-18:
Hi @lml0001  Do you require further assistance with this case please, or can it be closed?   Thanks!

MartyG-RealSense, 2020-06-25:
Case closed due to no further comments received