realsenseai/librealsense #6508: D435i natural white balance value

BarRae, 2020-06-03:
Hello,

I know this is camera question and not really programming related but maybe you guys can help me.

A while back I made a vision system using a d435i camera.
Today while testing with different lighting condition I noticed a blue-ish tint over the complete image.
I've attached a original image and a screenshot of the image the camera takes below:
[image]

[image]

I think this blue-ish tint is possibly caused by two things, the office lighting or the white balance.
Therefore my question is as follows:
What is the white balance value that gives me the most natural image (no blue-ish tint) ((mine is currently set to auto))

Keep in mind: the original image above is a digital version: I printed this image and stuck it to my test-board. The image from the d435i above is the actual image this camera takes.

If you have any other tips/tricks or questions feel free to let me know.
Thanks in advance,
Bart.

MartyG-RealSense, 2020-06-03:
When auto white balance for RGB is active in the RealSense Viewer, a 50% halfway value between minimum and maximum of '4800' is used. 

Below 4800, the more that the white balance is reduced towards zero, the bluer the image seems to become tinted.

BarRae, 2020-06-04:
So 4800 would give the most natural image?
Or would you happen to know what the best way is to capture colors that are the most like the original image?

MartyG-RealSense, 2020-06-04:
Going below 4800 progressively shifted the image towards blue tint and going above 4800 seemed to progressively make the image "warmer" in color (for example, a pale red becoming an intense red like a rosy apple).   

I found that when auto white balance was disabled, the image became warmer at the 4800 setting.  Lowering it to 3500 seemed to produce hues that represented what my human eyes were seeing, and this value was not low enough yet for the blue tint to start emerging.

MartyG-RealSense, 2020-06-18:
Hi @BarRae  Do you require further assistance with this case please?

BarRae, 2020-06-18:
Oh, sorry.

I forgot to close the case.

Thank you for your answers, i've played a bit with the settings and now it seems to be okay.

closed 2020-06-18 by BarRae